robotology/robotology-superbuild #500: Prepare release 2020.11

traversaro, 2020-10-19:
As we are getting used to, it is time to start preparing the 2020.11 release.

traversaro, 2020-10-21:
The release process will follow the workflow documented in the Developer's FAQs : https://github.com/robotology/robotology-superbuild/blob/master/doc/developers-faqs.md#how-to-do-a-new-release .

traversaro, 2020-10-22:
Hi @arrenglover @francodipietro, given that https://github.com/robotology/robotology-superbuild/pull/502 was just merged, it would be nice if you could tag a version of the event-driven so that it can be included in the 2020.11 repo. We typically use the v1.2.3 style for tags, but also any other style is perfectly ok, thanks!

S-Dafarra, 2020-10-27:
I just noticed that after https://github.com/robotology/idyntree/pull/751 gets merged, the ``UnicyclePlanner`` would not compile. Working on a release for it.

traversaro, 2020-10-27:
> 
> 
> I just noticed that after [robotology/idyntree#751](https://github.com/robotology/idyntree/pull/751) gets merged, the `UnicyclePlanner` would not compile. Working on a release for it.

Probably the `UnicyclePlanner`  was using some deprecated iDynTree function? Or there were some problems in https://github.com/robotology/idyntree/pull/751 ?

S-Dafarra, 2020-10-27:
> Probably the `UnicyclePlanner` was using some deprecated iDynTree function? Or there were some problems in [robotology/idyntree#751](https://github.com/robotology/idyntree/pull/751) ?

I forgot some deprecated methods in there :grin:

S-Dafarra, 2020-10-27:
> I just noticed that after [robotology/idyntree#751](https://github.com/robotology/idyntree/pull/751) gets merged, the `UnicyclePlanner` would not compile. Working on a release for it.

Done, #507

traversaro, 2020-11-09:
@prashanthr05 @HosameldinMohamed if you do not see any blocking problem, it would be also useful to merge devel in master and do a new release of https://github.com/robotology/whole-body-estimators .

prashanthr05, 2020-11-09:
> @prashanthr05 @HosameldinMohamed if you do not see any blocking problem, it would be also useful to merge devel in master and do a new release of https://github.com/robotology/whole-body-estimators .

Yes, we shall do that.

prashanthr05, 2020-11-09:
> > @prashanthr05 @HosameldinMohamed if you do not see any blocking problem, it would be also useful to merge devel in master and do a new release of https://github.com/robotology/whole-body-estimators .
> 
> Yes, we shall do that.

A new release of `whole-body-estimators` is available at https://github.com/robotology/whole-body-estimators/releases/tag/v0.3.0

traversaro, 2020-12-04:
The release 2020.11 was finalized.

closed 2020-12-04 by traversaro