robotology/robotology-superbuild #510: Robotology-superbuild fails when ROBOTOLOGY_ENABLE_DYNAMICS and ROBOTOLOGY_USES_PYTHON

Giulero, 2020-11-02:
I upgraded to Ubuntu 20.04 and building the superbuild I have run in strange behavior.
With `ROBOTOLOGY_ENABLES_CORE = ON` and `ROBOTOLOGY_ENABLES_PYTHON = ON` the process fails at:
```
-- Could NOT find PythonLibs (missing: PYTHON_LIBRARIES PYTHON_INCLUDE_DIRS) (Required is exact version "3.6.9")
Traceback (most recent call last):
  File "<string>", line 1, in <module>
ImportError: cannot import name 'sysconfig'
CMake Error at bindings/python/CMakeLists.txt:31 (string):
  string sub-command STRIP requires two arguments.

CMake Error at bindings/python/CMakeLists.txt:32 (install):
  install FILES given no DESTINATION!

CMake Error at bindings/python/CMakeLists.txt:33 (install):
  install TARGETS given no LIBRARY DESTINATION for module target
  "_iDynTree_python".

-- Configuring incomplete, errors occurred!
```
The compilation does not fail if 
`ROBOTOLOGY_ENABLES_CORE = OFF` and `ROBOTOLOGY_ENABLES_PYTHON = ON` 
or
`ROBOTOLOGY_ENABLES_CORE = ON` and `ROBOTOLOGY_ENABLES_PYTHON = OFF` .

Also, if I import `sysconfig` in a python shell I don't have issues.

traversaro, 2020-11-02:
Did you tried  with a clean build?

Giulero, 2020-11-02:
Yes I did. 
I'm trying to clean my configuration, maybe the upgrade to ubuntu 20.04 was somehow harmful.

traversaro, 2020-11-03:
Sorry, the docs for that option are not complete, see https://github.com/robotology/robotology-superbuild/issues/28 . In particular, the package to install are not documented, and from the CI script (see https://github.com/robotology/robotology-superbuild/blob/master/.ci/install_debian.sh#L15) we can see that you should install:
~~~
sudo apt-get install python3-dev python3-numpy
~~~
Can you try to install this packages and try again? Thanks!

Giulero, 2020-11-03:
Hi Silvio! I had these packages also before the building. It seems that, upgrading to 20.04, could be harmful since the default python version changes (in Ubuntu 20.04 it's python 3.8) and some leftovers of the previous python  version remain in the system.
I'm pretty sure that there is a less drastic solution, but eventually I managed to solve the problem removing python and reinstalling it again.

traversaro, 2020-11-03:
Perfect, so I think we can close the issue, right?

Giulero, 2020-11-03:
Sure! Thanks for the help :)

closed 2020-11-03 by traversaro