robotology/robotology-superbuild #515: Error using yarpMEX with MATLAB bindings

lnobile, 2020-11-17:
Hello to everybody,
I recently installed the _robotology-superbuild_  with the intention to use the MATLAB bindings for YARP. 
I enabled:
`ROBOTOLOGY_USES_MATLAB` ON
`ROBOTOLOGY_NOT_USE_SIMULINK` ON
Since I don't need to use Simulink but only MATLAB.
I set the MATLAB path as described and I believe I followed all the listed instructions.
Hovever, when running the test code:
```
yarpVec = yarp.Vector();
yarpVec.fromMatlab([1;2;3]);
yarpVec.toMatlab()
```
 I get the following error:
```
Error using yarpMEX
The error message identifier is invalid.
Error in yarp.Vector/fromMatlab (line 163)
      [varargout{1:nargout}] = yarpMEX(2590, self, varargin{:});
Error in SwigRef/subsref (line 33)
        [varargout{1:nargout}] = builtin('subsref',self,s);
Error in test (line 2)
yarpVec.fromMatlab([1;2;3]);
```

My system config:
MATLAB 2020a Update 2
YARP-3.4
Ubuntu 18.04

traversaro, 2020-11-17:
Can you try to comment the `fromMatlab` version to check if it is a problem just of that? 

I also suggest to enable `ROBOTOLOGY_USES_DYNAMICS` to check if there is the same problem with iDynTree bindings, that also contain a similar `fromMatlab` function.

lnobile, 2020-11-17:
I commented the `fromMatlab` line and it seems to be the only problem.
When I try to compile with `ROBOTOLOGY_USES_DYNAMICS` ON i get the following error:

[ 80%] Building CXX object sources/Simulink/CMakeFiles/Simulink.dir/src/SimulinkBlockInformation.cpp.o
In file included from /home/lucanobile/src/robot/robotology-superbuild/robotology/BlockFactory/sources/Simulink/src/SimulinkBlockInformation.cpp:14:0:
/home/lucanobile/src/robot/robotology-superbuild/robotology/BlockFactory/sources/Simulink/include/BlockFactory/Simulink/Private/SimulinkBlockInformationImpl.h:15:10: fatal error: simstruc.h: No such file or directory
 #include <simstruc.h>

Is this due to the fact that I don't have Simulink?

traversaro, 2020-11-17:
> When I try to compile with `ROBOTOLOGY_USES_DYNAMICS` ON i get the following error:
>
> [ 80%] Building CXX object sources/Simulink/CMakeFiles/Simulink.dir/src/SimulinkBlockInformation.cpp.o
> In file [...]

Thanks, this is indeed a bug in the `ROBOTOLOGY_NOT_USE_SIMULINK`, that is not really used a lot and is tricky to test in CI due to MATLAB/Simulink being proprietary software. This should be fixed by https://github.com/robotology/robotology-superbuild/pull/516 .

traversaro, 2020-11-17:
> I commented the fromMatlab line and it seems to be the only problem.

Interesting, if that is the only problem one workaround is to use the some other methods to fill the vector (if you need this functionality) otherwise you can try to use the functionality of MATLAB/yarp that you wanted to use and see if there are other problems.

lnobile, 2020-11-17:
I will try to use the other functionalities and in case I need to fill the vector I will use other methods, thanks for the advice!

traversaro, 2020-11-17:
Thanks, let's keep this issue open in any case as even if it a problem for which quick workarounds are possible, we eventually need to fix it. Furthermore, if you could provide your feedback on https://github.com/robotology/robotology-superbuild/pull/516 it would be great.

lnobile, 2020-11-17:
> Thanks, let's keep this issue open in any case as even if it a problem for which quick workarounds are possible, we eventually need to fix it. Furthermore, if you could provide your feedback on #516 it would be great.

Just tested and now works, approved ;)

traversaro, 2021-05-11:
Fixes for this issue are provided in:
* https://github.com/robotology/yarp-matlab-bindings/issues/62
* https://github.com/robotology/robotology-superbuild/pull/720

traversaro, 2021-09-07:
The issue was solved by the linked PRs. If the issue is still present, please re-open the issue, thanks!

closed 2021-09-07 by traversaro