robotology/robotology-superbuild #517: Porting bipedal-locomotion-framework in the superbuild

GiulioRomualdi, 2020-11-18:
Discussing with @traversaro we were thinking that it may make sense to start porting the `blf` inside the robotology-superbuild. This will simplify the installation of the framework for the users that want to exploit it (e.g. @isorrentino @diegoferigo)

Since the framework depends on:
- `casADi` 
- `qhull`
- `manif`
- `cppad` (not really required)

we may think to create a new profile.

cc @S-Dafarra @prashanthr05 @DanielePucci

S-Dafarra, 2020-11-18:
I agree. My two cents are:
- I would avoid creating a new profile. ``Dynamics`` seems already a good fit
- Thanks to the dependency mechanisms in blf, we can avoid all those dependencies, setting them optional for the superbuild. If the user decides to install them too, blf will configure accordingly

diegoferigo, 2020-11-18:
Double thumbs up :+1: :+1: When I started using blf, honestly, I thought it was already part of the superbuild :)

traversaro, 2020-11-18:
> I would avoid creating a new profile. Dynamics seems already a good fit

I don't know, I think that forcing the compilation of casadi and qhull on all the Dynamics users not interested in those would increase the compilation time and the possibility of encountering problems in users that could not be interested in bfl, and adding a new component is not really a effort intensive operation, but if you feel strongly about having it in dynamics it is ok for me.

S-Dafarra, 2020-11-18:
> I don't know, I think that forcing the compilation of casadi and qhull on all the Dynamics users not interested in those would increase the compilation time and the possibility of encountering problems in users that could not be interested in bfl

Yeah, that's why I was thinking of dealing with these dependencies separately, maybe with some separate options. Maybe they could go into a separate profile ``BLF_OPTIONAL_DEPENDENCIES`` or something like that, but the nice thing right now is that ``blf`` can compile without them.

traversaro, 2020-11-20:
> Yeah, that's why I was thinking of dealing with these dependencies separately, maybe with some separate options. Maybe they could go into a separate profile `BLF_OPTIONAL_DEPENDENCIES` or something [...]

In that case, I think we can keep core blf also inside Dynamics, and keep blf+optional dependencies inside a separate component (similar to how  YARP and ICUB are part of `CORE`, but if you enable `ROBOTOLOGY_ENABLE_ICUB_HEAD` you get several other options enabled). I just prefer to avoid the creation of options that are not related to profiles or **external** dependencies, because they are tricky to maintain. We can have `ROBOTOLOGY_ENABLE_ADVANCED_WALKING`, what do you think?

S-Dafarra, 2020-11-20:
> In that case, I think we can keep core blf also inside Dynamics, and keep blf+optional dependencies inside a separate component (similar to how YARP and ICUB are part of `CORE`, but if you enable [...]

Ah, I see, makes sense. ``ROBOTOLOGY_ENABLE_ADVANCED_WALKING``? I don't like "advanced walking" as it seems to be related to fancy algorithms :grin:

traversaro, 2020-11-20:
I would avoid `EXPERIMENTAL`, because I know that it will stick around for a long time, and when we start using it in demos is no longer experimental for me.  : )
We can also just have `ROBOTOLOGY_ENABLE_WALKING`, or  `ROBOTOLOGY_ENABLE_WALKING_COMPLETE`, `ROBOTOLOGY_ENABLE_WALKING_FULL`.

traversaro, 2020-11-20:
By the way, the scheme "base stuff with no optional dependencies" in Dynamics, while the advanced options in a separate component (or just not having that) is something that could make sense also for gym_ignition to expose the Scenario API easily to all lab members. However, for that probably we need also to add a `ROBOTOLOGY_USES_IGNITION_GAZEBO` option. fyi @diegoferigo

S-Dafarra, 2020-11-20:
> I would avoid `EXPERIMENTAL`, because I know that it will stick around for a long time, and when we start using it in demos is no longer experimental for me. : )
> We can also just have `ROBOTOLOGY_ENABLE_WALKING`, or `ROBOTOLOGY_ENABLE_WALKING_COMPLETE`, `ROBOTOLOGY_ENABLE_WALKING_FULL`.

> By the way, the scheme "base stuff with no optional dependencies" in Dynamics, while the advanced options in a separate component (or just not having that) is something that could make sense also [...]

Then, why not adding also ``gym_ignition`` to ``DYNAMICS`` and having a separate option for enabling the ``FULL_OPTIONAL`` configuration of both?

traversaro, 2020-11-20:
Something like `ROBOTOLOGY_ENABLE_DYNAMICS_FULL_DEPS`? That is a possibility.

S-Dafarra, 2020-11-20:
> Something like `ROBOTOLOGY_ENABLE_DYNAMICS_FULL_DEPS`? That is a possibility.

Agreed!

traversaro, 2020-11-20:
Perfect. Hopefully over time those dependency will be installed by the standard package managers (or the version installed by the package managers will not be broken, I am looking at you, qhull) and we could get rid of this, but for the time being probably it is the best option.

GiulioRomualdi, 2020-11-20:
This is the current situation with the dependencies

| Dipendency |  Linux | Windows |  macOS |
|:----------:|:------:|:-------:|:------:|
|   `manif`  | GitHub |  GitHub | GitHub |
|   `qhull`  | GitHub |  vcpkg  |  brew  |
|  `catch2`  | GitHub |  vcpkg  |  brew  |
|   `cppad`  | GitHub |  vcpkg  |  brew  |

traversaro, 2020-11-20:
So, do you think ti could make sense for `qhull` and `cppad` (and eventually if we enable tests, `catch2`)  to be built from source just on Linux, while relying on other packages managers on  Windows/macOS? Given that compilation is typically slower and more error-prone on Windows/macOS, this could make sense.

GiulioRomualdi, 2020-11-20:
Hi @traversaro on the bipedal CI I'm installing the dependencies as represented in the table above. 
So I would do the same also with the superbuild

GiulioRomualdi, 2020-11-21:
we can actually propose to add `manif` in vcpkg/brew (Probably we should discuss with some of the mantainers)

traversaro, 2020-11-21:
> we can actually propose to add manif in vcpkg/brew (Probably we should discuss with some of the maintainers).

Yes, that would be easy in vcpkg, more tricky in homebrew (but on the other hand, custom homebrew ports are much easier to have, until you provide bottles). That is the reason why I asked some time ago for the manif release, but @prashanthr05 told me that is blocked by a PR that they need to merge. However, there is no reason for being blocked by this, as we can ship manif with no problem as subproject of the superbuild.

GiulioRomualdi, 2020-11-22:
First experiment: https://github.com/robotology/robotology-superbuild/pull/526

closed 2020-11-26 by traversaro

traversaro, 2021-01-14:
> I don't know, I think that forcing the compilation of casadi and qhull on all the Dynamics users not interested in those would increase the compilation time and the possibility of encountering [...]

cc @CarlottaSartore for the decision of adding bipedal-locomotion-framework to the existing dynamics profile.