robotology/robotology-superbuild #539: Fix matio-cpp repo dependency

kouroshD, 2020-11-27:
I had the following error.  The problem is that, in the superbuild the dependencies of `matio-cpp` is not installed as mentioned in https://github.com/dic-iit/matio-cpp#dependencies with the super-build system dependencies.
I was updating my superbuild with the recent changes but I had the error.

```
CMake Error at /usr/local/Cellar/cmake/3.18.2/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:165 (message):
  Could NOT find MATIO (missing: MATIO_INCLUDE_DIR MATIO_LIBRARY)
Call Stack (most recent call first):
  /usr/local/Cellar/cmake/3.18.2/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:458 (_FPHSA_FAILURE_MESSAGE)
  cmake/FindMATIO.cmake:23 (find_package_handle_standard_args)
  CMakeLists.txt:91 (find_package)
```

traversaro, 2020-11-27:
The dependency on matio for the Dynamics component has been documented for a long time in https://github.com/robotology/robotology-superbuild/blob/master/doc/profiles.md#dynamics . However, especially now that profile docs has been moved somewhere else, probably especially given that libmatio is such as small library, it is more convenient to just move it together with all the other dependencies and mention it in the main readme.

traversaro, 2020-11-27:
Do you agree in moving matio to the main superbuild dependencies?

kouroshD, 2020-11-27:
yes, completely agree. I will do it on Monday if you agree.

traversaro, 2020-11-30:
> 
> 
> yes, completely agree. I will do it on Monday if you agree.

Ok, thanks!

kouroshD, 2020-12-01:
I've added the dependencies for the linx and macos to https://github.com/kouroshD/robotology-superbuild/pull/new/feature/updateDependency.

@traversaro  For the Windows OS, in https://github.com/robotology/robotology-superbuild/blob/master/doc/profiles.md#dependencies-specific-documentation it is referring to the main library dependencies while here https://github.com/dic-iit/matio-cpp#dependencies mentioning that we can use vcpkg. So, I assumed this one https://github.com/robotology/robotology-superbuild/blob/master/doc/profiles.md#dependencies-specific-documentation is outdated. 
When I wanted to add to vcpkg, I saw we have it already there. https://github.com/robotology/robotology-superbuild/blob/master/doc/vcpkg-dependencies.md#install-dependencies-via-a-custom-vcpkg so, no need to be changed for windows.

traversaro, 2020-12-01:
> @traversaro For the Windows OS, in https://github.com/robotology/robotology-superbuild/blob/master/doc/profiles.md#dependencies-specific-documentation it is referring to the main library [...]

That is exact, the matio installation information on Windows was wrong, and the binary archive of vcpkg ports already contains matio.

kouroshD, 2020-12-02:
Since the PR https://github.com/robotology/robotology-superbuild/pull/546 is closed, we can close this issue.

closed 2020-12-02 by kouroshD