robotology/robotology-superbuild #542: Remove icub-gazebo from the robotology-superbuild

traversaro, 2020-11-27:
icub-gazebo was created in November 2013 as "Preliminary files for simulation of the iCub humanoid robot in Gazebo" https://github.com/robotology/icub-gazebo/commit/01c400a8f798a091e7fcf7dd88d56c46511fdb99 . 

After a model with eyes and hands has been added to icub-models in https://github.com/robotology/icub-models/pull/42, I think we can finally think of removing this repo from the superbuild, and eventually moving it in robotology-legacy.

GiulioRomualdi, 2020-11-29:
Rip  `iCubHeidelberg01`

traversaro, 2020-11-29:
For  iCubHeidelberg01 we have the SIMMECHANICS source file, so if necessary we can easily add it to icub-models-generator (or we just add the generated model directly in icub-models-generator).

traversaro, 2021-09-07:
Fixed in https://github.com/robotology/robotology-superbuild/pull/646 .

closed 2021-09-07 by traversaro