robotology/robotology-superbuild #543: Remove icub-gazebo-wholebody?

traversaro, 2020-11-27:
I am not sure if someone is still using the models and worlds in https://github.com/robotology/icub-gazebo-wholebody . If someone does, please comment here. If no one will reply, we should consider removing the repo. 

cc @robotology/iit-dynamic-interaction-control @gabrielenava

gabrielenava, 2020-11-29:
I don't use them anymore, maybe @Yeshasvitvs for the iCub standing up from the chair?

Yeshasvitvs, 2020-11-30:
I am not using then repo models actively. @lrapetti please confirm if you are using this models for pRRI ?

lrapetti, 2020-11-30:
> @lrapetti please confirm if you are using this models for pRRI ?

I am not using those models, so for me we can proceed removing it.

closed 2020-12-15 by traversaro