robotology/robotology-superbuild #550: Casadi fails to compile with robotology-superbuild

prashanthr05, 2020-12-09:
Today @affafjunaidMOMIN was trying to compile the superbuild on a Linux machine and came across with an issue where the casadi compilation was failing due to linking errors to blas and lapack libraries.

This usually means that the `libblas-dev` and `liblapack-dev` is not installed as a prerequisite for installing casadi.

@affafjunaidMOMIN please let us know if the compilation succeeded after installing those from the `apt`.
using 
`sudo apt install libblas-dev liblapack-dev`

cc @GiulioRomualdi

traversaro, 2020-12-09:
Thanks @prashanthr05  for handling this. Just to understand, this is happening after enabling the `ROBOTOLOGY_ENABLE_DYNAMICS_FULL_DEPS` options, right? On which distro are you experiencing this?

traversaro, 2020-12-13:
Friendly ping @affafjunaidMOMIN @prashanthr05 .

prashanthr05, 2020-12-13:
> Just to understand, this is happening after enabling the `ROBOTOLOGY_ENABLE_DYNAMICS_FULL_DEPS` options, right? 

I actually did not pay attention to this variable during that time. So I am really not sure it was enabled or not.

affafjunaidMOMIN, 2020-12-13:
> Thanks @prashanthr05 for handling this. Just to understand, this is happening after enabling the `ROBOTOLOGY_ENABLE_DYNAMICS_FULL_DEPS` options, right? On which distro are you experiencing this?

Yes, I had it enabled before installing

traversaro, 2020-12-13:
That is quite strange. For Debian/Ubuntu distros we test in CI on clean Docker images, and we don't install anything beside the dependencies listed in the README (see https://github.com/robotology/robotology-superbuild/blob/master/.github/workflows/ci.yml#L112). On which distro you had this error?  do you happen to have saved the original error?  Do you think there is anything strange on your setup (previous installations of other software) that you can think of?

affafjunaidMOMIN, 2020-12-14:
@traversaro No. My system was new, so there was nothing installed previously. But I did some mistakes while installing. For example, I selected gazebo and python packages even though I don't have them at the moment. The installation gave an error and I had to switch off these packages and configure again.

traversaro, 2020-12-14:
> @traversaro No. My system was new, so there was nothing installed previously. But I did some mistakes while installing. For example, I selected gazebo and python packages even though I don't have [...]

Thanks for the inputs. Indeed, we do not test all the possible combination of options, and so it is possible that `ROBOTOLOGY_ENABLE_DYNAMICS_FULL_DEPS` set to `ON`  is not working if `ROBOTOLOGY_ENABLE_PYTHON` and `ROBOTOLOGY_ENABLE_GAZEBO`  are set to `OFF`, because `casadi` requires blas and lapack, and those are installed as a transitive dependency of either Gazebo or some Python package, so that could be the problem. I will spin a custom build with just `ROBOTOLOGY_ENABLE_DYNAMICS_FULL_DEPS` without installing Gazebo to understand if that is the problem, thanks!

closed 2021-04-14 by traversaro