robotology/robotology-superbuild #552: Regression in conda-forge Linux CI

traversaro, 2020-12-09:
From https://github.com/robotology/robotology-superbuild/actions . 

Two days ago everything was working fine: 
* https://github.com/robotology/robotology-superbuild/runs/1514709988?check_suite_focus=true 

Yesterday instead something went wrong, and went wrong also in https://github.com/robotology/robotology-superbuild/pull/532 : https://github.com/robotology/robotology-superbuild/runs/1521342051?check_suite_focus=true 

The error is: 
~~~
Looking for: ['ace', 'asio', 'boost', 'eigen', 'gazebo', 'glew', 'glfw', 'gsl', 'ipopt', 'libjpeg-turbo', 'libmatio', 'libode', 'libxml2', 'opencv', 'pkg-config', 'portaudio', 'qt', 'sdl', 'sdl2', 'sqlite', 'tinyxml']

Encountered problems while solving.
Problem: package gazebo-11.0.0-h398b6f7_9 requires qt >=5.12.5,<5.13.0a0, but none of the providers can be installed
~~~

I briefly tried to replicate this locally, and it seems that instead conda is working fine, I spawned https://github.com/robotology/robotology-superbuild/pull/551 to check if this happens also on the CI.

traversaro, 2020-12-10:
Cool, switching to conda as done in https://github.com/robotology/robotology-superbuild/pull/551 fixes the problem on Linux, and it creates a new problem on macOS: https://github.com/robotology/robotology-superbuild/pull/551/checks?check_run_id=1527274310 .

traversaro, 2020-12-10:
Indeed, the problem is in mamba. 
~~~
mamba install -c conda-forge cmake compilers make ninja pkg-config ace asio boost eigen gazebo glew glfw gsl ipopt libjpeg-turbo libmatio libode libxml2 opencv pkg-config portaudio qt sdl sdl2 sqlite tinyxml
~~~
fails with the previous error (see ci job https://github.com/robotology/robotology-superbuild/runs/1529533907), while :

~~~
 conda install -c conda-forge cmake compilers make ninja pkg-config ace asio boost eigen gazebo glew glfw gsl ipopt libjpeg-turbo libmatio libode libxml2 opencv pkg-config portaudio qt sdl sdl2 sqlite tinyxml
~~~
works fine (see ci job https://github.com/robotology/robotology-superbuild/runs/1530469671).

I saw that a new release of mamba was released recently (https://github.com/conda-forge/mamba-feedstock/pull/87), so let's see if the problem is solved in 0.7.5 before reporting the problem in mamba.

wolfv, 2020-12-11:
I just tried 

```
mamba create -c conda-forge cmake compilers make ninja pkg-config ace asio boost eigen gazebo glew glfw gsl ipopt libjpeg-turbo libmatio libode libxml2 opencv pkg-config portaudio qt sdl sdl2 sqlite tinyxml
```

on linux, and that worked fine.

traversaro, 2020-12-11:
> @traversaro unfortunately this release of mamba doesn't address any kind of solving issues :/

Thanks for the hint!

wolfv, 2020-12-11:
Sorry, it's got to be 

```
mamba create cmake compilers make ninja pkg-config ace asio boost eigen gazebo glew glfw gsl ipopt libjpeg-turbo libmatio libode libxml2 opencv pkg-config portaudio qt sdl sdl2 sqlite tinyxml -c conda-forge -n testenv --dry-run
```

traversaro, 2020-12-11:
> Did you also try mamba create ...? If you are installing into the base-environment, there might be an incompatibility somewhere.

In theory the setup-miniconda is not installing in the base-environement, but rather everything in an environment called `test` . Now that I think of, this means that I am not installing `mamba` in the `base`  environment, but just in `test`  one, and I guess that could create problem. I will investigate more.

traversaro, 2020-12-11:
Indeed, the problem will be fixed by https://github.com/robotology/robotology-superbuild/pull/553 that avoiding to manually install mamba, but rather is following the official setup-miniconda docs: https://github.com/conda-incubator/setup-miniconda#example-6-mamba . 

Thanks @wolfv for the help, as usual.

traversaro, 2020-12-13:
Fixed by https://github.com/robotology/robotology-superbuild/pull/553 .

closed 2020-12-13 by traversaro