robotology/robotology-superbuild #602: Compilation failure on Windows

traversaro, 2021-01-25:
~~~
 iCubSkinGui.vcxproj -> C:\src\robotology-superbuild\build\src\ICUB\bin\Release\iCubSkinGui.exe
libifcoremd.lib(for_main.obj) : error LNK2019: unresolved external symbol MAIN__ referenced in function main [C:\src\robotology-superbuild\build\src\ICUB\src\simulators\iCubSimulation\iCub_SIM.vcxproj]
C:\src\robotology-superbuild\build\src\ICUB\bin\Release\iCub_SIM.exe : fatal error LNK1120: 1 unresolved externals [C:\src\robotology-superbuild\build\src\ICUB\src\simulators\iCubSimulation\iCub_SIM.vcxproj]
  iKinCartesianSolver.vcxproj -> C:\src\robotology-superbuild\build\src\ICUB\bin\Release\iKinCartesianSolver.exe
  iKinGazeCtrl.vcxproj -> C:\src\robotology-superbuild\build\src\ICUB\bin\Release\iKinGazeCtrl.exe
  icubmod.vcxproj -> C:\src\robotology-superbuild\build\src\ICUB\lib\Release\icubmod.lib
  imageBlender.vcxproj -> C:\src\robotology-superbuild\build\src\ICUB\bin\Release\imageBlender.exe
  imageCropper.vcxproj -> C:\src\robotology-superbuild\build\src\ICUB\bin\Release\imageCropper.exe
  imageSplitter.vcxproj -> C:\src\robotology-superbuild\build\src\ICUB\bin\Release\imageSplitter.exe
MSVCRT.lib(exe_main.obj) : error LNK2019: unresolved external symbol main referenced in function "int __cdecl __scrt_common_main_seh(void)" (?__scrt_common_main_seh@@YAHXZ) [C:\src\robotology-superbuild\build\src\ICUB\src\tools\joystickCheck\joystickCheck.vcxproj]
C:\src\robotology-superbuild\build\src\ICUB\bin\Release\joystickCheck.exe : fatal error LNK1120: 1 unresolved externals [C:\src\robotology-superbuild\build\src\ICUB\src\tools\joystickCheck\joystickCheck.vcxproj]
MSVCRT.lib(exe_main.obj) : error LNK2019: unresolved external symbol main referenced in function "int __cdecl __scrt_common_main_seh(void)" (?__scrt_common_main_seh@@YAHXZ) [C:\src\robotology-superbuild\build\src\ICUB\src\tools\joystickCtrl\joystickCtrl.vcxproj]
~~~

traversaro, 2021-01-25:
The problem did not affected conda as in that case there is no such strange directory `lib/manual-link`.

S-Dafarra, 2021-01-25:
> . My assumption is that probably power users disseminate the fact that `CMAKE_TOOLCHAIN_FILE` should be used, even if that was not documented in the superbuild README,

It seems documented in https://github.com/robotology/robotology-superbuild-dependencies-vcpkg:

> To install , you just need to download either `vcpkg-robotology.zip` or `vcpkg-robotology-with-gazebo.zip` and extract it in the `C:/` directory.
> The archive contains a `robotology` directory, that will be available in your system as `C:/robotology` .
> 
> To use then vcpkg libraries, you need to configure your CMake projects to use the vcpkg's 
> toolchain as `CMAKE_TOOLCHAIN_FILE`, for example by configuring the projects as:  
> ~~~
> cmake -A x64 -DCMAKE_TOOLCHAIN_FILE=C:/robotology/vcpkg/scripts/buildsystems/vcpkg.cmake ..
> ~~~

I would argue that https://github.com/robotology/robotology-superbuild#system-libraries does not describe in detail how to use the archive, referring then to the documentation of ``robotology-superbuild-dependencies-vcpkg``

> For instructions on how to correctly use this archives, please refer to documentation of the robotology-superbuild-dependencies-vcpkg repo.

traversaro, 2021-01-25:
Ahh, that is a good point. It is also explicitly explain in https://github.com/robotology/robotology-superbuild#superbuild-3 : 
> Once you cloned the repository, you can generate the Visual Studio solution using the CMake GUI, by using as a generator the appropriate Visual Studio version, and the 64 bit as platform, and specifying the vcpkg CMake toolchain as discussed in the previous section. 

This explains why other users did not experience this, thanks! In any case, even if you don't specify the toolchain this will be fixed by https://github.com/robotology/robotology-superbuild-dependencies-vcpkg/pull/40 .

closed 2021-01-25 by traversaro