robotology/robotology-superbuild #608: Env Variables on Windows

Iaxama, 2021-02-03:
When installing using the windows installer on a drive different than C: some environment variables point to C nonetheless.

In my case the installation was on D:\robotology and the environment variables added to the path were:

C:\robotology\robotology\bin
C:\robotology\robotology\lib\rtf

instead of:

D:\robotology\robotology\bin
D:\robotology\robotology\lib\rtf

traversaro, 2021-02-03:
At the moment, the installer do not officially support being installed in prefix different from `C:\robotology` , mainly due to YARP (related issues: https://github.com/robotology/yarp/issues/2445) and Gazebo (https://github.com/osrf/gazebo/issues/2782, https://github.com/osrf/gazebo/issues/2755) not being properly relocatable packages. Unfortunately, I notice that this is not documented in the main robotology readme, while it is documented in https://icub-tech-iit.github.io/documentation/sw_installation/windows_binaries/ . 

Ideally it would be great to avoid that the installer prompts the user to change the install prefix, but I never understood how to do that.

traversaro, 2021-02-03:
I renamed your issue to reflect the underlying problem, so that we have an issue in which to track this improvement.

traversaro, 2021-02-04:
Note that even if at the moment we do not support this explicitly, it should be possible to still use the binaries installed in `D:\robotology` with a few fixes: 
* Do not tick the environmental variables update during the installation process, as in any case you saw that are not valid. If you already changed them, run the `D:/robotology/scripts/removePathsFromUserEnvVariables.ps1` script to cleanup, or cleanup them manually by looking in https://github.com/robotology/robotology-superbuild/blob/master/doc/environment-variables-configuration.md .
* After you cleaned up your environment, go in `D:\robotology\share\robotology-superbuild\addPathsToUserEnvVariables.ps1`and update the script to refer to `D:\` instead of `C:\` 
* If you need to use YARP plugins, to avoid https://github.com/robotology/yarp/issues/2445 manually open the following files and change `C:\` to `D:\`: 
  * `D:/robotology/robotology/share/yarp/plugins/yarp.ini`
  * `D:/robotology/robotology/share/iCub/plugins/iCub.ini` 
  and in general, any file with pattern `D:/robotology/robotology/share/<package>/plugins/<package>.ini`

At this point, you can run again the `D:/robotology/scripts/addPathsToUserEnvVariables.ps1` script to update again the env variables. If you want to use instead the `setup.bat` or `setup.sh` scripts, you will need to update those files instead to refer to `D:` instead of `C:`.

Iaxama, 2021-02-12:
Hi @traversaro sorry for the late response. At the moment I have simply fixed the issue by manually changing the environment variables to have the prefix D: instead of C:  and everything seems to work correctly. 

Maybe just using the installation prefix in the env variable script setup might solve the issue. :man_shrugging:

traversaro, 2021-02-12:
> Maybe just using the installation prefix in the env variable script setup might solve the issue. 🤷‍♂️

Unfortunately that scripts is generate when the installer is generated, and so the installation prefix is not known (as it can be changed by the user). So the solution is either to make the script relocatable (i.e. such that they know their location, and they set all those variables depending on their location, see https://github.com/robotology/robotology-superbuild/issues/264) or to add in the installer a step that fix the scripts. 

However, as we soonish could have conda binaries also on Windows (see https://github.com/robotology/robotology-superbuild/issues/620), I would wait for that stage, and investigate if we can simply generate these installers using [Conda Constructor](https://github.com/conda/constructor), as this should fix all the problems as Conda has automatic support for fixing this kind of non-relocatable scripts (see https://docs.conda.io/projects/conda-build/en/latest/resources/make-relocatable.html).