robotology/robotology-superbuild #614: Enable useful options in CasADi build of the robotology-superbuild

traversaro, 2021-02-08:
See https://github.com/robotology/robotology-superbuild/issues/239#issuecomment-774999559 .

traversaro, 2021-02-08:
Continuing from https://github.com/robotology/robotology-superbuild/issues/239#issuecomment-774999559 : 

@diegoferigo 

> The module could not be imported. I had to apply the following patch: `sed -i "s/class _copyableObject(_object):/class _copyableObject():/g" /usr/local/src/robotology-superbuild/build/install/python/casadi/casadi.py`. I guess it is a Python2 related problem.

Interesting, similar problem (but different sed line) on conda-forge : https://github.com/conda-forge/casadi-feedstock/blob/391139c859e1fba1321818bae27b6ca11fa4081e/recipe/build.sh#L18 . Interestingly, nothing seems to be necessary on Windows, I don't know if a new release came out and fixed that w.r.t. to the fork we are using in the superbuild.

> The Python bindings are installed into <build>/install/python.

I think you can specify that with the `PYTHON_PREFIX` option, see https://github.com/conda-forge/casadi-feedstock/blob/391139c859e1fba1321818bae27b6ca11fa4081e/recipe/build.sh#L13 .

> Yep I agree, we should pass -DWITH_LAPACK=ON WITH_MUMPS=ON WITH_QPOASES=ON.

You have a specific use case for `-DWITH_LAPACK=ON WITH_MUMPS=ON` ? I am afraid those could now work in the vcpkg-based Windows build. My goal is to eventually replace that with the Conda-based Windows build, but probably this will take some time, if you have a specific use case we can enable it just on Linux/macOS, otherwise I would avoid to have cross-platform differences.

diegoferigo, 2021-02-08:
> Interesting, similar problem (but different sed line) on conda-forge : https://github.com/conda-forge/casadi-feedstock/blob/391139c859e1fba1321818bae27b6ca11fa4081e/recipe/build.sh#L18 . [...]

Ow cool, thanks for the reference, I'm glad I was not the only one :) Maybe SWIG on windows generates slightly different code.

> > The Python bindings are installed into /install/python.
> 
> I think you can specify that with the `PYTHON_PREFIX` option, see https://github.com/conda-forge/casadi-feedstock/blob/391139c859e1fba1321818bae27b6ca11fa4081e/recipe/build.sh#L13 .

Yes you can. Btw I noticed yesterday that with a conda-based installation (not sure in a normal Ubuntu system) the `PYTHONPATH` exported by the superbuild is wrong. I start thinking that using an unversioned `<build>/install/lib/python/site-packages` would be more robust.

```
(/conda) root@ed2fd1dfa26f:~# cat /usr/local/src/robotology-superbuild/build/install/share/robotology-superbuild/setup.sh | grep PYTHON
# Add the python bindings directory to the PYTHON_PATH
export PYTHONPATH=${PYTHONPATH:+${PYTHONPATH}:}${ROBOTOLOGY_SUPERBUILD_INSTALL_PREFIX}/lib/python3/dist-packages
export PYTHONPATH=${PYTHONPATH}:${ROBOTOLOGY_SUPERBUILD_INSTALL_PREFIX}/lib/python3/site-packages

(/conda) root@ed2fd1dfa26f:~# ls /usr/local/src/robotology-superbuild/build/install/lib | grep python
python3.8

(/conda) root@ed2fd1dfa26f:~# ls /usr/local/src/robotology-superbuild/build/install/lib/python3.8/   
dist-packages  site-packages
```

> You have a specific use case for `-DWITH_LAPACK=ON WITH_MUMPS=ON` ? I am afraid those could now work in the vcpkg-based Windows build. My goal is to eventually replace that with the Conda-based [...]

One of the two is mandatory for the qpOASES backend, now on the fly I don't recally which one of the two.

traversaro, 2021-05-19:
Fixed in https://github.com/robotology/robotology-superbuild/pull/694 for Python and in https://github.com/robotology/robotology-superbuild/pull/747 for MATLAB.

closed 2021-05-20 by traversaro