robotology/robotology-superbuild #616: Warning when matlab bindings are loaded in matlab

GiulioRomualdi, 2021-02-08:
I install a clean superbuild on my PC, following the i[nstructions](https://github.com/robotology/robotology-superbuild/blob/master/doc/profiles.md#start-matlab-from-the-launcher-or-the-application-menu) I ran `startup_robotology_superbuild.m` and close matlab.

When I open it I get the following warning

```
Warning: Name is nonexistent or not a directory: /home/gromualdi/robot-code/robotology-superbuild/build/install/mex/+wbc/simulink 
```

Indeed `/+wbc` folder doesn't contain `simulink` but just a bunch of files

traversaro, 2021-02-08:
This directory was added in https://github.com/robotology/robotology-superbuild/pull/443 by @gabrielenava , perhaps he can comment on that.

gabrielenava, 2021-02-08:
> This directory was added in #443 by @gabrielenava , perhaps he can comment on that.

The directory exist only in the `devel` branch of `whole-body-controllers`, not in master that's why the warning. It is an harmless warning, but I would like to resolve it.

However I would rather prefer to merge `devel` branch of `whole-body-controllers` in `master` to solve it. I'll try to do it in this week.

traversaro, 2021-02-08:
Just a curiosity, the `simulink` subdirectory does not seem to be in `+wbc` neither in the devel branch, see https://github.com/robotology/whole-body-controllers/tree/devel/library/matlab-wbc/%2Bwbc .

traversaro, 2021-02-08:
Sorry, I got it, is installed in https://github.com/robotology/whole-body-controllers/blob/devel/library/simulink-library/CMakeLists.txt#L10 .

traversaro, 2021-05-25:
I think this was fixed by https://github.com/robotology/whole-body-controllers/pulls?q=is%3Apr+is%3Aclosed and https://github.com/robotology/whole-body-controllers/releases/tag/v2.5.3 , feel free to reopen if this is the case.

closed 2021-05-25 by traversaro