robotology/robotology-superbuild #621: Enable diagnosticdaemon project also for non-Unstable branches?

traversaro, 2021-02-10:
In https://github.com/robotology/robotology-superbuild/pull/436 the `diagnosticdaemon` project was added to the superbuild, but it was only activated for `Unstable` branches, and was also documented in the README. However, after that the  `diagnosticdaemon` was never enabled also for non-Unstable branches even if the features on which it was depending probably now are are available in the Stable and released version of its dependency. 

Could it make sense to: 
* Remove the `(ROBOTOLOGY_PROJECT_TAGS STREQUAL "Unstable")` check in https://github.com/robotology/robotology-superbuild/blob/3dc34724f6f4b41d74ede4c7d1b30021efe2fe20/cmake/RobotologySuperbuildLogic.cmake#L79 
* Do a release of `diagnosticdaemon` so it can be included in releases as all the other projects?

@pattacini @marcoaccame @Nicogene @triccyx

triccyx, 2021-02-10:
For me is ok, but I haven't any competence about "Stable/Unstable". @Nicogene what do you think about this?

Nicogene, 2021-02-10:
Probably it needed `icub_firmware_shared@devel` when it was made public but now master should be ok

About the release, I don't know if it is in a state for a release, maybe @marcoaccame knows it better

traversaro, 2021-02-10:
> About the release, I don't know if it is in a state for a release, maybe @marcoaccame knows it better

Even if it is not ready for a "formal" release, just a simple tag such `v0.0.1` or similar is ok for me.

traversaro, 2021-02-21:
If we want to include this in 2021.02 we probably need to take a decision on this relatively early, if instead that is not problem it would be a good idea to do this later. Furthermore, it probably make sense to define a maintainer for the `diagnosticdaemon` repo.

pattacini, 2021-02-21:
>Furthermore, it probably make sense to define a maintainer for the diagnosticdaemon repo.

Agreed! The official maintainer is now @triccyx as per: https://github.com/robotology/diagnostic-daemon/commit/74accf202297535f2ffe8dc1db834a59457aa5d1

traversaro, 2023-09-11:
Not sure if diagnosticdaemon is actually used, buy anyhow the issue was fixed in https://github.com/robotology/robotology-superbuild/pull/632 .

closed 2023-09-11 by traversaro