robotology/robotology-superbuild #625: iCub_SIM is not recognised

MrHuangWenjie, 2021-02-17:
Hi all,

I have installed Yarp and iCub according to official documents on:

http://yarp.it/git-master/install_yarp_windows.html
https://icub-tech-iit.github.io/documentation/sw_installation/windows_from_sources_superbuild/
https://github.com/robotology/robotology-superbuild#windows-from-source

I did exactly everything documented on them. Also, I have added environment variables by the .ps1 file.

I have successfully run yarpserver, while iCub_SIM is not recognised as a command. Intuitively, iCub_SIM should be a .exe file, so I searched it in the directory, but it turned out not found.

In addition, I found the environment variable YARP_DATA_DIRS was added to the system environment variable, which is different from the one with the same name added when iCub was installed. This confused me a bit and I am wondering that would it cause any conflict?

traversaro, 2021-02-21:
> I did exactly everything documented on them. Also, I have added environment variables by the .ps1 file.

Note that the superbuild installation procedure (documented in https://icub-tech-iit.github.io/documentation/sw_installation/windows_from_sources_superbuild/ and https://github.com/robotology/robotology-superbuild#windows-from-source) is mutually exclusive, as in the superbuild installation procedure YARP is compiled by the superbuild itself, so you should be able to ignore the instructions in http://yarp.it/git-master/install_yarp_windows.html .

> I have successfully run yarpserver, while iCub_SIM is not recognised as a command. Intuitively, iCub_SIM should be a .exe file, so I searched it in the directory, but it turned out not found.

This seems to indicate that there was a compilation problem in icub-main. To double check this, could you navigate with a terminal in the robotology-superbuild build directory, and run the following command:
~~~
cd /path/to/robotology-superbuild/build
cmake --build . --target ICUB --config Release
~~~
this should compile again the ICUB project, that contains and should install iCub_SIM . If this is not happening, probably there is  some problem. I suggest to report any problem that you may find in this issue to help us understanding the problem that you are experiencing. In particular, to get a more detailed error, you can also try to build the ICUB project yourself, by doing:
~~~
cd /path/to/robotology-superbuild/build/src/ICUB
cmake --build . --config Release --target INSTALL
~~~

As a general information, it may be useful that you report some info on your system, such as:
* Visual Studio version
* CMake version
* Version of https://github.com/robotology/robotology-superbuild-dependencies-vcpkg used

MrHuangWenjie, 2021-02-21:
> Note that the superbuild installation procedure (documented in https://icub-tech-iit.github.io/documentation/sw_installation/windows_from_sources_superbuild/ and [...]

Dear Silvio,

Thank you for your help.

Firstly, for clarification, I used VS 2019, CMake 3.19.5 and when I check vcpkg's version it showed this:
_Vcpkg package management program version _2020.11.12--unknownhash_

I have tried to do so:
_cd /path/to/robotology-superbuild/build
_cmake_ --build . --target ICUB --config Release_

but it failed as below:

_D:\YARP\git_source\robotology-superbuild\build>cmake --build . --target ICUB --config Release
_CMake Error: The current CMakeCache.txt directory D:/YARP/git_source/robotology-superbuild/build/CMakeCache.txt is different than the directory d:/YARP/git_source/iCub/robotology-superbuild/build where CMakeCache.txt was created. This may_ result in binaries being created in the wrong place. If you are not sure, reedit the CMakeCache.txt
_CMake_ is re-running because D:/YARP/git_source/iCub/robotology-superbuild/build/CMakeFiles/generate.stamp dependency file is missing.
_CMake_ Error: The current CMakeCache.txt directory D:/YARP/git_source/robotology-superbuild/build/CMakeCache.txt is different than the directory d:/YARP/git_source/iCub/robotology-superbuild/build where CMakeCache.txt was created. This may result in binaries being created in the wrong place. If you are not sure, reedit the CMakeCache.txt
_CMake_ Error: The source directory "D:/YARP/git_source/iCub/robotology-superbuild" does not exist.
Specify --help for usage, or press the help button on the CMake GUI.
_CMake_ Configure step failed.  Build files cannot be regenerated correctly._

In addition, as for manually building ICUB project, there is no src/ICUB under build directory, do I need to create by myself or it should be there but missed?

Best,
Wenjie

traversaro, 2021-02-21:
From the error message, it seems that the build directory full path has been changed since the build was created, and this is not supported by CMake. Can you try to delete your build directory and try again? Thanks!

MrHuangWenjie, 2021-02-21:
> From the error message, it seems that the build directory full path has been changed since the build was created, and this is not supported by CMake. Can you try to delete your build directory and try again? Thanks!

Hi Silvio,

I have uninstalled yarp installed by http://yarp.it/git-master/install_yarp_windows.html and the one remaining from superbuild worked well. However, the iCub_SIM command is still unrecognized.

I again searched it in the build directory and it turned out no such .exe file. Instead, there are iCub_SIM.sln, iCub_SIM.vcxproj.filters, and iCub_SIM.vcxproj.

By the way, I tried to make the build file with Vcpkg which failed with such message:

_Parse error. Expected a command name, got unquoted argument with text "MZ_''

Hence, I just make it with the default tool instead of vcpkg, I hope that was fine.

traversaro, 2021-02-21:
Hi @MrHuangWenjie , sorry, but how is your latest command related to my reply? Did you tried to cancel the build, configure again the robotology-superbuilr project and compile it again? If you did all of this, cmake --build . --target ICUB --config Release should work fine and should compile and install the iCub_SIM.exe in build/install/bin . Can you try this? Thanks!

MrHuangWenjie, 2021-02-21:
> Hi @MrHuangWenjie , sorry, but how is your latest command related to my reply? Did you tried to cancel the build, configure again the robotology-superbuilr project and compile it again? If you did [...]

according to your suggestion, I first delete the build directory, and make build again with default compiler instead of vcpkg as it gave error as mentioned before. Even after this, iCub_SIM was not recognized. Then I tried the command you mentioned to re-compile ICUB, while it reported:

_Error: could not load cache_

Best,
Wenjie

traversaro, 2021-02-21:
> according to your suggestion, I first delete the build directory, and make build again with default compiler instead of vcpkg as it gave error as mentioned before.

I am not sure what you mean by that, the commands for configuring and building are reported in https://github.com/robotology/robotology-superbuild#superbuild-3 and should be:
~~~
cd build
cmake -DCMAKE_TOOLCHAIN_FILE=C:\robotology\vcpkg\scripts\buildsystems\vcpkg.cmake ..
cmake --build . --config Release
~~~

I am not sure what you mean by " build again with default compiler instead of vcpkg". I suggest you to precisely report the command that you are trying and their **complete output** if you like to receive meaningful support.

>  Even after this, iCub_SIM was not recognized. Then I tried the command you mentioned to re-compile ICUB, while it reported:

Please report the complete output, not just the last line. It is extremely difficult to understand what is going on if the complete error is not reported.

MrHuangWenjie, 2021-02-21:
> I am not sure what you mean by that, the commands for configuring and building are reported in https://github.com/robotology/robotology-superbuild#superbuild-3 and should be:
>
> ```
> cd build
> cmake [...]

Dear Silvio,

Thank you so much for your help so far. I think I have fixed it now with the commands you provided.

_> cmake -DCMAKE_TOOLCHAIN_FILE=C:\robotology\vcpkg\scripts\buildsystems\vcpkg.cmake ..
> cmake --build . --config Release_

However, I would like to explain the possible reason why I rushed into this problem. Actually, I was using CMake GUI instead of the command line in Windows, and I was not familiar with vcpkg chain tool. Hence, I failed to make use of vcpkg.cmake (I chose vcpkg.exe file for the chain tool option instead, so there was errors) which was quite stupid.

This seems that we have to use vcpkg.cmake to make the build directory. Otherwise, like what I encountered, the iCub_SIM.exe would not be installed and compiled. Though, I have no idea about why.

Anyway, I am out of that problem now. Thank you so much for your patient help!!

traversaro, 2021-02-22:
Thanks @MrHuangWenjie , this seem to indicate that probably the current documentation is a bit confusing as it suggest to use the GUI tools to perform various steps (including setting the vcpkg toolchain that is definitely not obvious how to do) while specifying the cmd prompt/powershell commands may be easier because they are easier to document and to do exactly as mentioned in the docs. Probably it could make sense to eventually change the doc to do that, I opened https://github.com/robotology/robotology-superbuild/issues/628 to discuss that. If you don't have any other question, feel free to close the issue, thanks!

MrHuangWenjie, 2021-02-22:
Yes. I would also suggest the use of command line. I think the problem maybe the use of vcpkg which I missed at beguning. It would be better if the document could explicitly give the command of configuring vcpkg.cmake.

I will close this issue now. Thank you again for help!

closed 2021-02-22 by MrHuangWenjie