robotology/robotology-superbuild #636: Create a manually dispatched GitHub Action to do a new release

traversaro, 2021-02-25:
Doing a  new release is not a time intensive action (a few minutes every 3 months) but it is quite error prone, as we keep forgetting steps (see https://github.com/robotology/robotology-superbuild/issues/635) and there are some steps that are intrinsically prone to errors (for example, when tagging a new release on the releases/yyyy.mm branch, the default setting of the GitHub UI is to do tag the release on `master` branch, and it is quite easy to forget to switch to releases/yyyy.mm branch). 

For this reason, it may be convenient to create a GitHub Action to do the steps documented in https://github.com/robotology/robotology-superbuild/blob/v2021.02/doc/developers-faqs.md#how-to-do-a-new-release automatically.

traversaro, 2021-05-19:
@Nicogene is preparing something on this, according to verbal communication.

drdanz, 2021-05-19:
FYI YARP stable releases are done 95% using a GH action now:
https://github.com/robotology/yarp/blob/master/.github/workflows/tag.yml

It's still not ready for "feature" releases, and it still needs some operation to be done manually, but it is still saving me lots of time.

traversaro, 2021-05-19:
> 
> 
> FYI YARP stable releases are done 95% using a GH action now:
> https://github.com/robotology/yarp/blob/master/.github/workflows/tag.yml
> 
> It's still not ready for "feature" releases, and it still needs some operation to be done manually, but it is still saving me lots of time.

Super cool!

PR #743 (2021-05-19): Add Release Github Action
This PR add a github action for creating a new `robotology-superbuild` release, it automatize these steps: https://github.com/robotology/robotology-superbuild/blob/master/doc/developers-faqs.md#how-to-do-a-new-release

For doing this I changed the CMakeLists of windows packaging https://github.com/robotology/robotology-superbuild/commit/a2e8999f97d87940dd0c527266e528d711a1b215 for make easier the sed command, the idea is that on master the `INSTALLER_VERSION` is empty(or we can put something like  `master` and then it is versioned in the `releases/yyyy.mm` branch.

Th [... cut]
closed 2021-05-19 by traversaro