robotology/robotology-superbuild #667: Can't find portaudio on conda-forge channel via the miniforge installer

nunoguedelha, 2021-03-24:
Installing the package `portaudio` via the miniforge installer from the conda-forge channel fails:
```
$ conda install portaudio
Collecting package metadata (current_repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Collecting package metadata (repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.

PackagesNotFoundError: The following packages are not available from current channels:

  - portaudio

Current channels:

  - https://conda.anaconda.org/conda-forge/osx-64
  - https://conda.anaconda.org/conda-forge/noarch

To search for alternate channels that may provide the conda package you're
looking for, navigate to

    https://anaconda.org

and use the search bar at the top of the page.
```
Indeed, running the `conda-forge` installer solved the problem:
```
conda install -c conda-forge portaudio
```

nunoguedelha, 2021-03-24:
Probably this only happens with the Mac specific channel `conda-forge/osx-64`?

@traversaro any idea?

traversaro, 2021-03-24:
As a temporary workaround, you can install packages with arguments -c conda-forge -c defaults -c robotology

traversaro, 2021-03-24:
The root cause is https://github.com/conda-forge/portaudio-feedstock/issues/2, fortunately easy to fix on the conda-forge side.

nunoguedelha, 2021-03-25:
> As a temporary workaround, you can install packages with arguments -c conda-forge -c defaults -c robotology

yes, this time I used the ` -c conda-forge` option. I assumed that using this option only for the package `portaudio` wouldn't create conflicts or dependencies issues in general. At least so far, I could build the superbuild.

traversaro, 2021-03-25:
Just to clarify, the issue was due to the fact that portaudio was not available in the conda-forge channel, but just in the defaults one. This should be fixed after https://github.com/conda-forge/portaudio-feedstock/pull/12, probably it just takes a few minutes to gets available. However, the root cause of this issue is that in our instructions we are using miniforge that only has conda-forge enabled, while in our CI we are currently use both conda-forge and defaults . We should ensure that in our CI we do not use the defaults channel, not event accidentally. I opened https://github.com/robotology/robotology-superbuild/issues/669 to track this.

closed 2021-03-25 by nunoguedelha