robotology/robotology-superbuild #668: Build robotology-superbuild with dependencies from Conda fails

nunoguedelha, 2021-03-24:
I'm experiencing a few issues when trying to build robotology-superbuild after installing the dependencies via Conda, following the [conda-forge.md](https://github.com/robotology/robotology-superbuild/blob/master/doc/conda-forge.md) tutorial.

When following the steps in section https://github.com/robotology/robotology-superbuild/blob/master/doc/conda-forge.md#compile-the-robotology-superbuild:

> On Linux or macOS, run:
> ```
> cd robotology-superbuild
> mkdir build
> cmake ..
> cmake --build . --config Release 
> ```

the last command would fail with the error:
```
[ 75%] Performing build step for 'yarp-matlab-bindings'
[ 50%] Building CXX object matlab/CMakeFiles/yarpMEX.dir/autogenerated/yarpMATLAB_wrap.cxx.o
In file included from /Users/nunoguedelha/dev/superbuild-PR-testing/robotology-superbuild/src/yarp-matlab-bindings/matlab/autogenerated/yarpMATLAB_wrap.cxx:2263:
In file included from /Users/nunoguedelha/dev/superbuild-PR-testing/robotology-superbuild/build/install/include/yarp/dev/all.h:18:
In file included from /Users/nunoguedelha/dev/superbuild-PR-testing/robotology-superbuild/build/install/include/yarp/dev/ControlBoardInterfacesImpl.h:26:
In file included from /Users/nunoguedelha/dev/superbuild-PR-testing/robotology-superbuild/build/install/include/yarp/dev/ImplementControlCalibration.h:14:
/Users/nunoguedelha/dev/superbuild-PR-testing/robotology-superbuild/build/install/include/yarp/dev/ControlBoardHelper.h:23:16: error: no template named 'enable_if_t' in namespace 'std'; did you mean 'enable_if'?
          std::enable_if_t<std::is_trivial<T>::value, int> = 0>
          ~~~~~^
/Users/nunoguedelha/miniforge3/envs/robsuperbuild/bin/../include/c++/v1/type_traits:538:63: note: 'enable_if' declared here
template <bool, class _Tp = void> struct _LIBCPP_TEMPLATE_VIS enable_if {};
...
```

nunoguedelha, 2021-03-24:
CC @traversaro

traversaro, 2021-03-24:
This is similar to https://github.com/robotology/robotology-superbuild/pull/644 that should be solved in the latest release of yarp matlab bindings, which version of yarp-matlab-bindings are you using?

nunoguedelha, 2021-03-24:
YARP version: v3.4.3
yarp-matlab-bindings: v3.4.1

But the faulty template definition is in the YARP helpers `src/libYARP_dev/src/yarp/dev/ControlBoardHelper.h`.

nunoguedelha, 2021-03-24:
In `src/libYARP_dev/src/yarp/dev/ControlBoardHelper.h`, we define the template `std::enable_if_t`, while the packages installed through Conda define `std::enable_if`.

nunoguedelha, 2021-03-25:
More precisely, the template `enable_if` is defined in `~/miniforge3/pkgs/libcxx-<version>-<commit>/include/c++/v1/type_traits`. The version I have installed is:
```
$ conda list | grep libcxx
libcxx                    11.1.0               habf9029_0    conda-forge
```

nunoguedelha, 2021-03-25:
Sorry @traversaro , I hadn't read carefully #644, and then https://github.com/robotology/yarp-matlab-bindings/pull/63.

I got it now.

Just as additional information for me, I added the following note...

### Further details on `enable_if` VS `enable_if_t`
http://www.cplusplus.com/forum/general/241535/
> Standard library meta-functions with a _t suffix are just type aliases for the member type named metafunction::type. For instance, std::enable_if_t is something like
> ```
> template <bool Enable, typename T> 
> using std::enable_if_t = typename std::enable_if<Enable, T>::type;
> ```

https://omegaup.com/docs/cpp/en/cpp/types/enable_if.html
[image]

nunoguedelha, 2021-03-25:
Everything builds fine now. I added the conclusion in the description. We can close the issue.

closed 2021-03-25 by nunoguedelha