robotology/robotology-superbuild #678: Wrong name for the Build<package>.cmake for matio-cpp

Nicogene, 2021-04-06:
The cmake name of `matio-cpp` is actually `matioCpp`, then this https://github.com/robotology/robotology-superbuild/blob/master/cmake/Buildmatio-cpp.cmake should be named `BuildmatioCpp.cmake` and `matio-cpp` has to be replaced in other `find_or_build_package` calls.

cc @traversaro @S-Dafarra

S-Dafarra, 2021-04-06:
Is it ok to have the two names different?

traversaro, 2021-04-06:
Good point!

traversaro, 2021-04-06:
> Is it ok to have the two names different?

In the normal workflow that we typically use is not a big difference, having two different names may be problematic if you want to configure the superbuild to use an external matioCpp as find_or_build_package(matio-cpp) will not actually look for the `matioCpp` package.

commit f023d9fe (2021-04-07): matioCpp: fix name, not it is used the "cmake" name

It fixes #678
commit a539e3c5 (2021-04-12): matioCpp: fix name, not it is used the "cmake" name

It fixes #678
closed 2021-04-12 by traversaro