robotology/robotology-superbuild #679: Installation error: Could not find a package configuration file provided by "YARP_robotinterface"

Jingyue21, 2021-04-06:
Dear developers,

I'm installing iCub software following the [instructions](https://github.com/robotology/robotology-superbuild#linux-from-source). 
My system is Ubuntu 18.04.5 LTS (Bionic Beaver) with cmake version 3.18.2. I'm installing Distro 2021.02. When I did superbuild->ccmake, I turned on "ROBOTOLOGY_USES_PYTHON", "ROBOTOLOGY_ENABLE_EVENT_DRIVEN" and "ROBOTOLOGY_USES_GAZEBO" apart from the default configurations.
When I did superbuild->make, I got the error below. Could you kindly help me to solve this error? Many thanks in advance!

[ 51%] Completed 'icub-models'
[ 57%] Built target icub-models
[ 58%] Skipping update step for 'GazeboYARPPlugins'
[ 60%] Performing configure step for 'GazeboYARPPlugins'
Not searching for unused variables given on the command line.
loading initial cache file /<mypath>/robotology-superbuild/build/src/GazeboYARPPlugins/CMakeFiles/YCMTmp/GazeboYARPPlugins-cache-Release.cmake
CMake Error at /usr/local/lib/cmake/YARP/YARPConfig.cmake:162 (find_package):
  Could not find a package configuration file provided by
  "YARP_robotinterface" with any of the following names:

    YARP_robotinterfaceConfig.cmake
    yarp_robotinterface-config.cmake

  Add the installation prefix of "YARP_robotinterface" to CMAKE_PREFIX_PATH
  or set "YARP_robotinterface_DIR" to a directory containing one of the above
  files.  If "YARP_robotinterface" provides a separate development package or
  SDK, be sure it has been installed.
Call Stack (most recent call first):
  CMakeLists.txt:48 (find_package)

-- Configuring incomplete, errors occurred!
See also "/<mypath>/robotology-superbuild/build/src/GazeboYARPPlugins/CMakeFiles/CMakeOutput.log".
CMakeFiles/GazeboYARPPlugins.dir/build.make:120: recipe for target 'src/GazeboYARPPlugins/CMakeFiles/YCMStamp/GazeboYARPPlugins-configure' failed
make[2]: *** [src/GazeboYARPPlugins/CMakeFiles/YCMStamp/GazeboYARPPlugins-configure] Error 1
CMakeFiles/Makefile2:358: recipe for target 'CMakeFiles/GazeboYARPPlugins.dir/all' failed
make[1]: *** [CMakeFiles/GazeboYARPPlugins.dir/all] Error 2
Makefile:113: recipe for target 'all' failed

traversaro, 2021-04-08:
Hi @Jingyue21 ! It seems that you already have a YARP package installed in your system via `sudo make install`, can you try to remove it and try again? If you still have the YARP build from which you called `sudo make install`, you just need to navigate there and call `sudo make uninstall`.

Jingyue21, 2021-04-11:
@traversaro Thanks Silvio! I removed yarp by ```sudo make install```. Then I get the same error mention in this [issue](https://github.com/robotology/robotology-superbuild/issues/272). I'm wondering if there's another way that I don't have to remove anaconda (which is needed by some ongoing projects) from my system. Thanks a lot!
```
[ 93%] Linking CXX executable ../../bin/yarpdataplayer
/usr/lib/x86_64-linux-gnu/libopencv_imgcodecs.so.3.2.0: undefined reference to `TIFFReadDirectory@LIBTIFF_4.0'
//usr/lib/libgdal.so.20: undefined reference to `TIFFLastDirectory@LIBTIFF_4.0'
/usr/lib/x86_64-linux-gnu/libopencv_imgcodecs.so.3.2.0: undefined reference to `TIFFWriteEncodedStrip@LIBTIFF_4.0'
/usr/lib/x86_64-linux-gnu/libopencv_imgcodecs.so.3.2.0: undefined reference to `TIFFIsTiled@LIBTIFF_4.0'
//usr/lib/libgdal.so.20: undefined reference to `TIFFSwabArrayOfShort@LIBTIFF_4.0'
//usr/lib/libgdal.so.20: undefined reference to `TIFFIsByteSwapped@LIBTIFF_4.0'
//usr/lib/libgdal.so.20: undefined reference to `TIFFFlushData@LIBTIFF_4.0'
//usr/lib/libgdal.so.20: undefined reference to `TIFFFreeDirectory@LIBTIFF_4.0'
/usr/lib/x86_64-linux-gnu/libopencv_imgcodecs.so.3.2.0: undefined reference to `TIFFScanlineSize@LIBTIFF_4.0'
//usr/lib/libgdal.so.20: undefined reference to `TIFFWriteEncodedTile@LIBTIFF_4.0'
//usr/lib/libgdal.so.20: undefined reference to `TIFFWriteBufferSetup@LIBTIFF_4.0'
//usr/lib/libgdal.so.20: undefined reference to `TIFFTileSize@LIBTIFF_4.0'
/usr/lib/x86_64-linux-gnu/libopencv_imgcodecs.so.3.2.0: undefined reference to `TIFFRGBAImageOK@LIBTIFF_4.0'
/usr/lib/x86_64-linux-gnu/libopencv_imgcodecs.so.3.2.0: undefined reference to `TIFFClose@LIBTIFF_4.0'
//usr/lib/libgdal.so.20: undefined reference to `TIFFWriteRawStrip@LIBTIFF_4.0'
//usr/lib/libgdal.so.20: undefined reference to `TIFFSetTagExtender@LIBTIFF_4.0'
//usr/lib/libgdal.so.20: undefined reference to `TIFFGetFieldDefaulted@LIBTIFF_4.0'
//usr/lib/libgdal.so.20: undefined reference to `TIFFSwabArrayOfLong@LIBTIFF_4.0'
//usr/lib/libgdal.so.20: undefined reference to `TIFFTileSize64@LIBTIFF_4.0'
//usr/lib/libgdal.so.20: undefined reference to `TIFFReadRGBATileExt@LIBTIFF_4.0'
//usr/lib/libgdal.so.20: undefined reference to `TIFFStripSize@LIBTIFF_4.0'
//usr/lib/libgdal.so.20: undefined reference to `TIFFMergeFieldInfo@LIBTIFF_4.0'
/usr/lib/x86_64-linux-gnu/libopencv_imgcodecs.so.3.2.0: undefined reference to `TIFFSetWarningHandler@LIBTIFF_4.0'
//usr/lib/libgdal.so.20: undefined reference to `TIFFGetConfiguredCODECs@LIBTIFF_4.0'
//usr/lib/x86_64-linux-gnu/libgeotiff.so.2: undefined reference to `_TIFFmalloc@LIBTIFF_4.0'
//usr/lib/x86_64-linux-gnu/libgeotiff.so.2: undefined reference to `_TIFFmemcpy@LIBTIFF_4.0'
//usr/lib/libgdal.so.20: undefined reference to `TIFFClientdata@LIBTIFF_4.0'
//usr/lib/x86_64-linux-gnu/libgeotiff.so.2: undefined reference to `_TIFFrealloc@LIBTIFF_4.0'
//usr/lib/x86_64-linux-gnu/libgeotiff.so.2: undefined reference to `_TIFFmemset@LIBTIFF_4.0'
//usr/lib/libgdal.so.20: undefined reference to `TIFFReadRGBAStripExt@LIBTIFF_4.0'
//usr/lib/libgdal.so.20: undefined reference to `TIFFWriteCheck@LIBTIFF_4.0'
//usr/lib/libgdal.so.20: undefined reference to `TIFFSetWriteOffset@LIBTIFF_4.0'
//usr/lib/libgdal.so.20: undefined reference to `TIFFDefaultStripSize@LIBTIFF_4.0'
//usr/lib/libgdal.so.20: undefined reference to `TIFFScanlineSize64@LIBTIFF_4.0'
//usr/lib/libgdal.so.20: undefined reference to `TIFFIsBigEndian@LIBTIFF_4.0'
//usr/lib/x86_64-linux-gnu/libpoppler.so.73: undefined reference to `TIFFFdOpen@LIBTIFF_4.0'
/usr/lib/x86_64-linux-gnu/libopencv_imgcodecs.so.3.2.0: undefined reference to `TIFFWriteScanline@LIBTIFF_4.0'
//usr/lib/libgdal.so.20: undefined reference to `_TIFFfree@LIBTIFF_4.0'
//usr/lib/libgdal.so.20: undefined reference to `TIFFSwabLong@LIBTIFF_4.0'
/usr/lib/x86_64-linux-gnu/libopencv_imgcodecs.so.3.2.0: undefined reference to `TIFFGetField@LIBTIFF_4.0'
//usr/lib/libgdal.so.20: undefined reference to `TIFFSwabShort@LIBTIFF_4.0'
/usr/lib/x86_64-linux-gnu/libopencv_imgcodecs.so.3.2.0: undefined reference to `TIFFNumberOfStrips@LIBTIFF_4.0'
//usr/lib/libgdal.so.20: undefined reference to `TIFFWriteDirectory@LIBTIFF_4.0'
//usr/lib/x86_64-linux-gnu/libgdcmMSFF.so.2.8: undefined reference to `uuid_generate@UUID_1.0'
/usr/lib/x86_64-linux-gnu/libopencv_imgcodecs.so.3.2.0: undefined reference to `TIFFReadEncodedTile@LIBTIFF_4.0'
/usr/lib/x86_64-linux-gnu/libopencv_imgcodecs.so.3.2.0: undefined reference to `TIFFReadRGBATile@LIBTIFF_4.0'
//usr/lib/x86_64-linux-gnu/libgeotiff.so.2: undefined reference to `TIFFClientOpen@LIBTIFF_4.0'
//usr/lib/libgdal.so.20: undefined reference to `TIFFFlush@LIBTIFF_4.0'
//usr/lib/x86_64-linux-gnu/libgdcmMSFF.so.2.8: undefined reference to `uuid_parse@UUID_1.0'
//usr/lib/libgdal.so.20: undefined reference to `TIFFErrorExt@LIBTIFF_4.0'
//usr/lib/libgdal.so.20: undefined reference to `TIFFNumberOfDirectories@LIBTIFF_4.0'
//usr/lib/libgdal.so.20: undefined reference to `TIFFWriteRawTile@LIBTIFF_4.0'
//usr/lib/libgdal.so.20: undefined reference to `TIFFStripSize64@LIBTIFF_4.0'
//usr/lib/libgdal.so.20: undefined reference to `TIFFCreateDirectory@LIBTIFF_4.0'
/usr/lib/x86_64-linux-gnu/libopencv_imgcodecs.so.3.2.0: undefined reference to `TIFFOpen@LIBTIFF_4.0'
//usr/lib/libgdal.so.20: undefined reference to `TIFFSwabArrayOfDouble@LIBTIFF_4.0'
//usr/lib/libgdal.so.20: undefined reference to `TIFFSetSubDirectory@LIBTIFF_4.0'
[... cut ...]
CMakeFiles/Makefile2:7569: recipe for target 'src/yarpdataplayer/CMakeFiles/yarpdataplayer.dir/all' failed
make[4]: *** [src/yarpdataplayer/CMakeFiles/yarpdataplayer.dir/all] Error 2
Makefile:170: recipe for target 'all' failed
make[3]: *** [all] Error 2
CMakeFiles/YARP.dir/build.make:125: recipe for target 'src/YARP/CMakeFiles/YCMStamp/YARP-build' failed
make[2]: *** [src/YARP/CMakeFiles/YCMStamp/YARP-build] Error 2
CMakeFiles/Makefile2:1341: recipe for target 'CMakeFiles/YARP.dir/all' failed
make[1]: *** [CMakeFiles/YARP.dir/all] Error 2
Makefile:113: recipe for target 'all' failed
```

traversaro, 2021-04-11:
> I'm wondering if there's another way that I don't have to remove anaconda (which is needed by some ongoing projects) from my system.

Interesting, if you are experienced with conda there are two options. 

## If you want to compile robotology-superbuild linking against apt-provided software
In this case, you sure that the problem is exactly a conda problem? `/usr/lib/libgdal.so` and `/usr/lib/x86_64-linux-gnu/libopencv_imgcodecs.so.3.2.0` do not seem to be Anaconda libraries. To avoid conflict between apt-installed software and anaconda software, it should not be necessary to uninstall anaconda, but just to "deactivate" the conda environment, removing the build of robotology-superbuild (otherwise the CMake cache will contain reference to the anaconda libraries), and building again the robotology-superbuild in a terminal in which no conda environment is active. If you want to make sure that terminal by default are opened without any conda environment active, you just need to run the command `conda config --set auto_activate_base false`. 

## If you want to use robotology-superbuild software in conda environments 
However, if you just want to use the robotology software and you are familiar with conda/anaconda, you can just install it via the conda binaries, for which you can find the documentation in https://github.com/robotology/robotology-superbuild/blob/master/doc/conda-forge.md . This documentation is still not linked in the main README because it is quite new, but if you want to try any feedback is welcome. 

In a nutshell, you just need to create a new environments (as our packages are based on [conda-forge](https://conda-forge.org/), it is better to create a new environment to avoid creating conflicts with packages from the `defaults` anaconda channel):
~~~
conda create -n robotologyenv
conda activate robotologyenv
conda install -c robotology ycm-cmake-modules yarp icub-main icub-models gazebo-yarp-plugins event-driven
~~~

Note that however support for the `ROBOTOLOGY_USES_PYTHON` option is still a bit a work in progress (see https://github.com/robotology/robotology-superbuild/issues/641).

traversaro, 2021-04-11:
@Jingyue21 if you are not sure about which option is better for you, feel free to explain your use case.

traversaro, 2021-04-11:
Given that you are interested in event-driven, probably also @marco-monforte or @ChiaraBartolozzi can also provide some info on your use case, and how do you need to use the robotology software you are interested in.

Jingyue21, 2021-04-12:
@traversaro Thanks a lot for the quick response!
In my case, I'd like to interface iCubSim with another neuromorphic platform that uses Python APIs. So I think I would need to set ```ROBOTOLOGY_USES_PYTHON``` on, right? Then I tried the first option: "deactivate" the conda environment following your instructions. The compiling succeeds if I turn off ```ROBOTOLOGY_USES_GAZEBO```. If not, I get this error:

[ 70%] Performing configure step for 'GazeboYARPPlugins'
Not searching for unused variables given on the command line.
loading initial cache file /home/jingyue/aa_projects/high_level_ctrl/robotology-superbuild/build/src/GazeboYARPPlugins/CMakeFiles/YCMTmp/GazeboYARPPlugins-cache-Release.cmake
-- The C compiler identification is GNU 9.3.0
-- The CXX compiler identification is GNU 9.3.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found OpenCV: /usr (found version "3.2.0") 
-- Found YARP: /home/jingyue/aa_projects/high_level_ctrl/robotology-superbuild/build/install/lib/cmake/YARP (found version "3.4.3")
CMake Error at CMakeLists.txt:49 (find_package):
  By not providing "FindGazebo.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "Gazebo", but
  CMake did not find one.

  Could not find a package configuration file provided by "Gazebo" with any
  of the following names:

    GazeboConfig.cmake
    gazebo-config.cmake

  Add the installation prefix of "Gazebo" to CMAKE_PREFIX_PATH or set
  "Gazebo_DIR" to a directory containing one of the above files.  If "Gazebo"
  provides a separate development package or SDK, be sure it has been
  installed.

-- Configuring incomplete, errors occurred!
See also "/home/jingyue/aa_projects/high_level_ctrl/robotology-superbuild/build/src/GazeboYARPPlugins/CMakeFiles/CMakeOutput.log".
CMakeFiles/GazeboYARPPlugins.dir/build.make:121: recipe for target 'src/GazeboYARPPlugins/CMakeFiles/YCMStamp/GazeboYARPPlugins-configure' failed
make[2]: *** [src/GazeboYARPPlugins/CMakeFiles/YCMStamp/GazeboYARPPlugins-configure] Error 1
CMakeFiles/Makefile2:308: recipe for target 'CMakeFiles/GazeboYARPPlugins.dir/all' failed
make[1]: *** [CMakeFiles/GazeboYARPPlugins.dir/all] Error 2
Makefile:113: recipe for target 'all' failed
make: *** [all] Error 2

traversaro, 2021-04-12:
> In my case, I'd like to interface iCubSim with another neuromorphic platform that uses Python APIs. So I think I would need to set `ROBOTOLOGY_USES_PYTHON` on, right?

If you need to use the legacy (non-Gazebo based) `iCub_SIM`, you do not even need the `ROBOTOLOGY_USES_GAZEBO` option. However, I think it is critical to understand which Python libraries you would use on the robotology side (YARP Python bindings?) and on which Python interpreter this code will need to run. If you want to run the Python code on the Python interpreter provided by anaconda, I am not sure if the Python bindings compile against apt-dependencies will work, as there could be other conflicts between the apt-installed dependencies and the conda-installed ones. For that, you may want to install instead the robotology-superbuild via conda itself (we just merged support for `ROBOTOLOGY_USES_PYTHON` when building from source on conda with https://github.com/robotology/robotology-superbuild/pull/663).

> If not, I get this error:

As wrote before I am not sure if you actually need `ROBOTOLOGY_USES_GAZEBO`, but in any case the error seems due to the fact that you did not installed Gazebo via apt on your system. See the docs in https://github.com/robotology/robotology-superbuild/blob/master/doc/profiles.md#gazebo on how to do so (and do not forget to install the `libgazebo*-dev` packages.

Jingyue21, 2021-04-12:
For Python libraries, yes, I think it's mainly YARP Python bindings for the communication between ```iCub_SIM``` (A) and the neuromorphic platform (B), e.g. to send some robot states from A to B and some config from B to A periodically (every 10 msec?). I'm currently using Python 3.8.3 in standalone mode, not with ipython. Surprisingly, I can run YARP and iCub [see here](https://icub-tech-iit.github.io/documentation/sw_installation/check_your_installation/) without conda, with conda base or whatever conda environment I have. To me, it seems like the conflicts only exist when I do ```make```. After ```make``` is done, the execution so far is conflict-free :)

OK I see, I'll have look and see if I need ```ROBOTOLOGY_USES_GAZEBO```. Thanks again for the kind help!

traversaro, 2021-04-12:
If that works fine for you, great! However, note that if you run apt-based software in a conda-environment, you can also have **runtime** issues, depending on which specific software did you installed in the conda environment, or depending on which Python interpreter you have in conda and in the system, so it is possible that that would break in the future.

If you think you solved your problems, please close the issue, thanks!

Jingyue21, 2021-04-12:
Thanks for the information! I'll come back to option2 if I need to in the future. Yes, my problems are solved. Thank you!

closed 2021-04-12 by Jingyue21