robotology/robotology-superbuild #703: Conda/Linux/Unstable with ROBOTOLOGY_USES_IGNITION is failing

traversaro, 2021-04-28:
The error is:
~~~
2021-04-28T03:06:58.6227577Z FAILED: cpp/scenario/plugins/Physics/CMakeFiles/PhysicsSystem.dir/Physics.cc.o 
2021-04-28T03:06:58.6240050Z /usr/share/miniconda/envs/test/bin/x86_64-conda-linux-gnu-c++ -DPhysicsSystem_EXPORTS -DQT_CORE_LIB -DQT_GUI_LIB -DQT_NETWORK_LIB -DQT_NO_DEBUG -DQT_QML_LIB -DQT_QUICKCONTROLS2_LIB -DQT_QUICK_LIB -DQT_WIDGETS_LIB -I/home/runner/work/robotology-superbuild/robotology-superbuild/src/gym-ignition/cpp/scenario/plugins/Physics -I/home/runner/work/robotology-superbuild/robotology-superbuild/src/gym-ignition/cpp/scenario/gazebo/include -I/home/runner/work/robotology-superbuild/robotology-superbuild/src/gym-ignition/cpp/scenario/core/include -isystem /usr/share/miniconda/envs/test/include/ignition/gazebo4 -isystem /usr/share/miniconda/envs/test/include/ignition/math6 -isystem /usr/share/miniconda/envs/test/include/ignition/plugin1 -isystem /usr/share/miniconda/envs/test/include/ignition/cmake2 -isystem /usr/share/miniconda/envs/test/include/ignition/common3 -isystem /usr/share/miniconda/envs/test/include/uuid -isystem /usr/share/miniconda/envs/test/include/ignition/fuel_tools5 -isystem /usr/share/miniconda/envs/test/include/ignition/msgs6 -isystem /usr/share/miniconda/envs/test/include/ignition/gui4 -isystem /usr/share/miniconda/envs/test/include/eigen3 -isystem /usr/share/miniconda/envs/test/include/ignition/transport9 -isystem /usr/share/miniconda/envs/test/include/qt -isystem /usr/share/miniconda/envs/test/include/qt/QtCore -isystem /usr/share/miniconda/envs/test/./mkspecs/linux-g++ -isystem /usr/share/miniconda/envs/test/include/qt/QtQml -isystem /usr/share/miniconda/envs/test/include/qt/QtNetwork -isystem /usr/share/miniconda/envs/test/include/qt/QtQuick -isystem /usr/share/miniconda/envs/test/include/qt/QtGui -isystem /usr/share/miniconda/envs/test/include/qt/QtQuickControls2 -isystem /usr/share/miniconda/envs/test/include/qt/QtWidgets -isystem /usr/share/miniconda/envs/test/include/sdformat-10.3/sdf/.. -isystem /usr/share/miniconda/envs/test/include/ignition/physics3 -fvisibility-inlines-hidden -std=c++17 -fmessage-length=0 -march=nocona -mtune=haswell -ftree-vectorize -fPIC -fstack-protector-strong -fno-plt -O2 -ffunction-sections -pipe -isystem /usr/share/miniconda/envs/test/include -O3 -DNDEBUG -fPIC -I/usr/share/miniconda/envs/test/include/uuid -I/usr/share/miniconda/envs/test/include -fPIC -std=c++17 -MD -MT cpp/scenario/plugins/Physics/CMakeFiles/PhysicsSystem.dir/Physics.cc.o -MF cpp/scenario/plugins/Physics/CMakeFiles/PhysicsSystem.dir/Physics.cc.o.d -o cpp/scenario/plugins/Physics/CMakeFiles/PhysicsSystem.dir/Physics.cc.o -c /home/runner/work/robotology-superbuild/robotology-superbuild/src/gym-ignition/cpp/scenario/plugins/Physics/Physics.cc
2021-04-28T03:06:58.6251830Z In file included from /home/runner/work/robotology-superbuild/robotology-superbuild/src/gym-ignition/cpp/scenario/plugins/Physics/Physics.cc:19:
2021-04-28T03:06:58.6253751Z /home/runner/work/robotology-superbuild/robotology-superbuild/src/gym-ignition/cpp/scenario/plugins/Physics/Physics.hh:29:10: fatal error: ignition/physics/CapsuleShape.hh: No such file or directory
2021-04-28T03:06:58.6254970Z    29 | #include <ignition/physics/CapsuleShape.hh>
2021-04-28T03:06:58.6255454Z       |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
2021-04-28T03:06:58.6255860Z compilation terminated.
~~~

Any idea @diegoferigo ?

diegoferigo, 2021-04-28:
Yes, the `Unstable` tags is cloning the [`devel` branch](https://github.com/robotology/robotology-superbuild/blob/ee06604d1d1e69f3e34243156a42d7e72800e3ca/cmake/ProjectsTagsUnstable.cmake#L33) that requires Edifice, but the superbuild installs Dome.

A quick workaround is using `master` also in `Unstable`, since in any case Edifice is not yet available in conda-forge.

traversaro, 2021-04-28:
> A quick workaround is using `master` also in `Unstable`, since in any case Edifice is not yet available in conda-forge.

Yes, I think this is a  good idea.

diegoferigo, 2021-04-28:
As a reference, the missing feature was introduced in https://github.com/ignitionrobotics/ign-physics/pull/202.

> > A quick workaround is using `master` also in `Unstable`, since in any case Edifice is not yet available in conda-forge.
> 
> Yes, I think this is a good idea.

Done in https://github.com/robotology/robotology-superbuild/pull/705.

closed 2021-04-28 by traversaro