robotology/robotology-superbuild #707: MATLAB

nunoguedelha, 2021-05-03:
### Short description

The Simulink models compilation is failing after installing the robotology-superbuild based on conda-forge:
```
<img width="656" alt="image" src="https://user-images.githubusercontent.com/6848872/116880009-be356280-ac21-11eb-80f3-1ebbbf22cf07.png">
```

### Platform environment

Hardware: MacBook Pro
OS: MacOS Catalina 10.15.7
robotolgy-superbuild: commit 8c45dd3aea5a4f1f79c043111cf84ee140e0ebb7, source installation + dependencies installed with conda-forge (https://github.com/robotology/robotology-superbuild/blob/master/doc/conda-forge.md#source-installation)
MATLAB/Simulink: R2020b/10.2
Simulink model: [floating-base-balancing-torque-control](https://github.com/robotology/whole-body-controllers/tree/devel/controllers/floating-base-balancing-torque-control)

### Procedure to reproduce the issue

Procedure to reproduce the issue running the example model [floating-base-balancing-torque-control](https://github.com/robotology/whole-body-controllers/tree/devel/controllers/floating-base-balancing-torque-control):

1. Install the superbuild following https://github.com/robotology/robotology-superbuild/blob/master/doc/conda-forge.md#source-installation.
2. Run MATLAB from `~/Documents/MATLAB`.
3. Set working directory to `[getenv('ROBOTOLOGY_SUPERBUILD_SOURCE_DIR'),'/src/whole-body-controllers/controllers/floating-base-balancing-torque-control']`.
4. 
4. Open the model `torqueControlBalancing.mdl`
5. Compile (⌘+D on MacOS, CTRL+D on Linux and Windows I guess).

### Workaround

Before compiling:
1. close the model.
2. run in the command window
```
>> clear all
>> path(pathdef)
```
3. open the model again.
4. Compile.

nunoguedelha, 2021-05-03:
It doesn't seem to be related to whole-body-controllers repo.

traversaro, 2021-06-05:
@DanielePucci experienced the same problem following the one-line installation, for him the error was:
~~~
Warning: Unsupported robot model. Supported models are listed below:
- iCubGazeboV2_5 [2 similar]
kinDynListDimensions =
  15×1 cell array
    {1×2 double}
    {1×2 double}
    {1×2 double}
    {1×2 double}
    {1×2 double}
    {1×2 double}
    {1×2 double}
    {1×2 double}
    {1×2 double}
    {1×2 double}
    {1×2 double}
    {1×2 double}
    {1×2 double}
    {1×2 double}
    {1×2 double}
Error evaluating 'InitFcn' callback of SubSystem block (mask) 'torqueControlBalancingWithSimu/Robot Simulator/RobotDynWithContacts'. 
Callback string is '%% initialize the simulation
mwbs.RobotDynamicsWithContacts.initRobotDynamicsWithContactsCB;'
Caused by:
Invalid MEX-file '/Users/danielepucci/src/matlab/robotology-matlab/mex/yarpMEX.mexmaci64': dlopen(/Users/danielepucci/src/matlab/robotology-matlab/mex/yarpMEX.mexmaci64, 6): Library not loaded: @rpath/libpng16.16.dylib
  Referenced from: /Users/danielepucci/src/matlab/robotology-matlab/lib/libYARP_sig.3.4.5.dylib
  Reason: Incompatible library version: libYARP_sig.3.dylib requires version 54.0.0 or later, but libpng16.16.dylib provides version 52.0.0 
Component:Simulink | Category:Model error
Error evaluating 'InitFcn' callback of mwbs.Visualizers.robotVisualizer.RobotVisualizer block (mask) 'torqueControlBalancingWithSimu/Robot Visualizer/RobotVisualizer'. 
Callback string is 'mwbs.Visualizers.robotVisualizer.initRobotVisualizer'
Caused by:
Invalid MEX-file '/Users/danielepucci/src/matlab/robotology-matlab/mex/yarpMEX.mexmaci64': dlopen(/Users/danielepucci/src/matlab/robotology-matlab/mex/yarpMEX.mexmaci64, 6): Library not loaded: @rpath/libpng16.16.dylib
  Referenced from: /Users/danielepucci/src/matlab/robotology-matlab/lib/libYARP_sig.3.4.5.dylib
  Reason: Incompatible library version: libYARP_sig.3.dylib requires version 54.0.0 or later, but libpng16.16.dylib provides version 52.0.0 
Component:Simulink | Category:Model error
~~~

traversaro, 2021-06-05:
@DanielePucci I did not remember that @nunoguedelha had found a workaround, if you like you can test it.

DanielePucci, 2021-06-07:
I tried the workaround, and what follows is what I did

1. I opened Matlab
2. I launched the command `floatingBaseBalancingTorqueControlWithSimulator` and the output was
~~~
Unable to resolve the name floatingBaseBalancingTorqueControlWithSimulator.torqueControlBalancingWithSimu.
~~~ 
3. I was not sure what to do (I felt that it was supposed to work, but probably I was wrong) so I launched again `robotology_setup` and **there was no output**
4. Then, I launched `floatingBaseBalancingTorqueControlWithSimulator.torqueControlBalancingWithSimu` that now worked and opened the model
5. I clicked on *Play* on the Simulink model with the output
~~~
torqueControlBalancingWithSimu 
 Simulation  1  2  2 
10:58 AM Elapsed: 7 sec
Warning: Unsupported robot model. Supported models are listed below:
- iCubGazeboV2_5 [2 similar]
kinDynListDimensions =
  15×1 cell array
    {1×2 double}
    {1×2 double}
    {1×2 double}
    {1×2 double}
    {1×2 double}
    {1×2 double}
    {1×2 double}
    {1×2 double}
    {1×2 double}
    {1×2 double}
    {1×2 double}
    {1×2 double}
    {1×2 double}
    {1×2 double}
    {1×2 double}
Error evaluating 'InitFcn' callback of SubSystem block (mask) 'torqueControlBalancingWithSimu/Robot Simulator/RobotDynWithContacts'. 
Callback string is '%% initialize the simulation
mwbs.RobotDynamicsWithContacts.initRobotDynamicsWithContactsCB;'
Caused by:
Invalid MEX-file '/Users/danielepucci/src/matlab/robotology-matlab/mex/yarpMEX.mexmaci64': dlopen(/Users/danielepucci/src/matlab/robotology-matlab/mex/yarpMEX.mexmaci64, 6): Library not loaded: @rpath/libpng16.16.dylib
  Referenced from: /Users/danielepucci/src/matlab/robotology-matlab/lib/libYARP_sig.3.4.5.dylib
  Reason: Incompatible library version: libYARP_sig.3.dylib requires version 54.0.0 or later, but libpng16.16.dylib provides version 52.0.0 
Component:Simulink | Category:Model error
Error evaluating 'InitFcn' callback of mwbs.Visualizers.robotVisualizer.RobotVisualizer block (mask) 'torqueControlBalancingWithSimu/Robot Visualizer/RobotVisualizer'. 
Callback string is 'mwbs.Visualizers.robotVisualizer.initRobotVisualizer'
Caused by:
Invalid MEX-file '/Users/danielepucci/src/matlab/robotology-matlab/mex/yarpMEX.mexmaci64': dlopen(/Users/danielepucci/src/matlab/robotology-matlab/mex/yarpMEX.mexmaci64, 6): Library not loaded: @rpath/libpng16.16.dylib
  Referenced from: /Users/danielepucci/src/matlab/robotology-matlab/lib/libYARP_sig.3.4.5.dylib
  Reason: Incompatible library version: libYARP_sig.3.dylib requires version 54.0.0 or later, but libpng16.16.dylib provides version 52.0.0 
Component:Simulink | Category:Model error
~~~
6. Then I followed the workaround in [the first comment](https://github.com/robotology/robotology-superbuild/issues/707#issue-874587819), but there was another action I had to take (not save the file), and the overall output is 
    * I closed the model, then to the question *Save "torqueControlBalancingWithSimu" before closing?* I clicked on *No*
    * In the command window I ran: 
~~~
>> clear all
>> path(pathdef)
~~~
    * Then I opened the model again re-launching `floatingBaseBalancingTorqueControlWithSimulator.torqueControlBalancingWithSimu` but I got the output
~~~
Unable to resolve the name floatingBaseBalancingTorqueControlWithSimulator.torqueControlBalancingWithSimu.
~~~
    * So, I launched again `robotology_setup` and then `floatingBaseBalancingTorqueControlWithSimulator.torqueControlBalancingWithSimu`. No it worked 
[image]
    * I have to say that it was awesome to see the Yoga running in Matlab with *our simulation infrastructure*. I did not know if it was supposed to finish, or how long the simulation was supposed to run. So, when the robot arrived back to two-fee standing, then I sopped. Probably, if we have to send this demo to someone, it would be better to fix a simulation time and show the percentage of the demo

traversaro, 2021-06-07:
The fact that `clear all` makes the problem goes aways may indicate that there is some shared library that was loaded by MATLAB and that is unloaded when your run `clear all`.

traversaro, 2021-06-07:
Things that it would be nice to check (eventually remainder for myself): 
* Try if the workaround works just with `clear mex`
* Print [`inmem('-completenames')`](https://www.mathworks.com/help/matlab/ref/inmem.html) in the situation after the compilation of the model failed, and after the compilation of the model was successful

nunoguedelha, 2021-06-11:
Hi @DanielePucci , thanks again for your feedback. Thanks @traversaro for the great support. I'm answering here to some of Dani's observations during his tests..

> 2. I launched the command `floatingBaseBalancingTorqueControlWithSimulator` and the output was
> 
> ```
> Unable to resolve the name floatingBaseBalancingTorqueControlWithSimulator.torqueControlBalancingWithSimu.
> ```

Most probably, you already know by now why this wasn't working, but just to be sure, I try to explain here:
If I understood correctly, this is a new MATLAB session, so the MATLAB Path has been reset to the default value and unless you saved the MATLAB path to a `pathdef.m` file, the default value does not contain the paths:
```
<robotology_install_prefix>/mex
<robotology_install_prefix>/mex/+wbc/simulink
<robotology_install_prefix>/mex/+wbc/examples
<robotology_install_prefix>/share/WBToolbox
<robotology_install_prefix>/share/WBToolbox/images
```
where `<robotology_install_prefix>/mex/+wbc/examples` would be the path containing the model to test. That's why MATLAB couldn't resolve the model name.

> 6.3 Then I opened the model again re-launching `floatingBaseBalancingTorqueControlWithSimulator.torqueControlBalancingWithSimu` but I got the output
> 
> ```
> Unable to resolve the name floatingBaseBalancingTorqueControlWithSimulator.torqueControlBalancingWithSimu.
> ```

This is due to similar reasons. At startup, or when you run `path(pathdef)`, the MATLAB path is initialized from a `pathdef.m` if it exists, and the environment variable `MATLABPATH`. This variable is not set in script `robotology_setup` as we add the paths directly through the `addpath` command. For that reason you have to run `robotology_setup` again after running `path(pathdef)`. 

**Note:** When following the [Binary Installation from the Conda Robotology Channel](https://github.com/dic-iit/matlab-whole-body-simulator#binary-installation-from-the-conda-robotology-channel) or the [Source Installation](https://github.com/dic-iit/matlab-whole-body-simulator#floppy_disk-source-installation), the environment variable `MATLABPATH` is set then applied at MATLAB startup.

> Probably, if we have to send this demo to someone, it would be better to fix a simulation time and show the percentage of the demo.

good point, will do it ASAP.