robotology/robotology-superbuild #725: QT5 question

spotanjo3, 2021-05-12:
Hello. I am a macOS user. How do I set mine for cmake -DCMAKE_BUILD_TYPE=Release -DQt5_DIR=/usr/local/opt/qt/lib/cmake/Qt5 ?

traversaro, 2021-05-12:
Hi @spotanjo3 . What do you mean by "How do I set mine"? Is this question related to the robotology-superbuild?

traversaro, 2021-05-12:
Did you read the macOS documentation on Homebrew and Qt5 in https://github.com/robotology/robotology-superbuild/blob/5f75faf046783da124dcfb5be7ee2e5f45003c6f/README.md#system-dependencies-1 ?

spotanjo3, 2021-05-12:
Sorry for the confused. When I compiled the building.. cmake -DCMAKE_BUILD_TYPE=Release -DQt5_DIR=/usr/local/opt/qt/lib/cmake/Qt5

CMake Warning:
  No source or binary directory provided.  Both will be assumed to be the
  same as the current working directory, but note that this warning will
  become a fatal error in future CMake releases.

CMake Error: The source directory "/Users/myname/build-release" does not appear to contain CMakeLists.txt.

traversaro, 2021-05-12:
It seems that if you command is:
~~~
cmake -DCMAKE_BUILD_TYPE=Release -DQt5_DIR=/usr/local/opt/qt/lib/cmake/Qt5
~~~

you are not specifying anywhere which CMake source directory you want to configure. In the case of the superbuild, this should be the directory in which you cloned the `robotology-superbuild` . In the official instructions https://github.com/robotology/robotology-superbuild#superbuild-2 , this directory is specified as `..` .

traversaro, 2021-05-12:
Hi @spotanjo3, if you think your issue is solved, feel free to close the issue, thanks!

closed 2021-05-12 by spotanjo3