robotology/robotology-superbuild #779: Test robotology-superbuild on macOS with zsh

traversaro, 2021-06-01:
The new default shell in macOS is zsh  since 10.15 (see https://scriptingosx.com/2019/06/moving-to-zsh/). Apparently, most users of the robotology-superbuild still do not use zsh, but we should nevertheless test it if it works fine, so we are not surprised when users try to use it.

Things to test:
* [ ] Build everything from source, and source the `<install_prefix>/share/robotology-superbuild/setup.sh`  script of the superbuild, and check if everything is correct or if that script contains some bash-ism 
* [ ] Try to install binary packages via conda following the instructions in https://github.com/robotology/robotology-superbuild/blob/master/doc/conda-forge.md . In theory for our packages we already generate setup scripts for zsh via [multisheller](https://github.com/mamba-org/multisheller), see https://github.com/robotology/robotology-superbuild/blob/545ec06c6915e0581dfaa2a09cc435f37ad5d614/conda/python/generate_conda_recipes_from_metametadata.py#L80 , but in practice that is completed untested.

nunoguedelha, 2021-06-02:
Ciao @traversaro , I guess that for testing you will first do:
- Replace the shebang `#!/bin/bash` by `#!/bin/zsh` or `#!/bin/sh`.
- Change the symbolic link stored in `/var/select/sh` to point the zsh shell (probably done by `chsh -s /bin/zsh`).
- Translate the ~/.bash_profile config to the zsh configuration files for zsh.

Please let me know if you need help for this.

traversaro, 2021-06-02:
Even without switching the default terminal of your system to `zsh`, I guess an easy check given an existing installation is to open a zsh terminal, and just try to source the `setup.sh` and check if the program start as usual or not. If you have already a robotology-superbuild installation available, that should be easy to do. By the way, I just realized this should be relatively easy do do even on Linux, just by installing zsh.

traversaro, 2021-06-05:
I tested on a Ubuntu 18.04 docker image zsh (`sudo apt install zsh`) and the setup.sh is correctly sourced, including features as checking if a file exits with `-f` and the syntax to append a value only of the variable is not defined.

lrapetti, 2021-06-08:
> * Build everything from source, and source the `<install_prefix>/share/robotology-superbuild/setup.sh`  script of the superbuild, and check if everything is correct or if that script contains some bash-ism

I did build everything from source again, and I have blindly copied my `.bash_profile` into the `.zshrc`. The `setup.sh` seems to be working fine. 

Only problem I have encounter is in using the yarp autocompletion, the instruction in http://wiki.icub.org/wiki/Homebrew#Bash_completion are no longer working.
In particular, the following is thrown
```
/usr/local/etc/bash_completion:59: command not found: shopt
/usr/local/etc/bash_completion:65: command not found: complete
/usr/local/etc/bash_completion:72: command not found: complete
/usr/local/etc/bash_completion:73: command not found: complete
/usr/local/etc/bash_completion:74: command not found: complete
/usr/local/etc/bash_completion:76: command not found: complete
/usr/local/etc/bash_completion:77: command not found: complete
/usr/local/etc/bash_completion:79: command not found: complete
/usr/local/etc/bash_completion:80: command not found: complete
/usr/local/etc/bash_completion:81: command not found: complete
/usr/local/etc/bash_completion:82: command not found: complete
/usr/local/etc/bash_completion:83: command not found: complete
/usr/local/etc/bash_completion:84: command not found: complete
/usr/local/etc/bash_completion:85: command not found: complete
/usr/local/etc/bash_completion:86: command not found: complete
/usr/local/etc/bash_completion:87: command not found: complete
/usr/local/etc/bash_completion:88: command not found: complete
/usr/local/etc/bash_completion:89: command not found: complete
/usr/local/etc/bash_completion:90: command not found: complete
/usr/local/etc/bash_completion:91: command not found: complete
/usr/local/etc/bash_completion:92: command not found: complete
/usr/local/etc/bash_completion:93: command not found: complete
/usr/local/etc/bash_completion:94: command not found: complete
/usr/local/etc/bash_completion:95: command not found: complete
/usr/local/etc/bash_completion:96: command not found: complete
/usr/local/etc/bash_completion:97: command not found: complete
/usr/local/etc/bash_completion:98: command not found: complete
/usr/local/etc/bash_completion:99: command not found: complete
/usr/local/etc/bash_completion:100: command not found: complete
/usr/local/etc/bash_completion:101: command not found: complete
/usr/local/etc/bash_completion:102: command not found: complete
/usr/local/etc/bash_completion:103: command not found: complete
/usr/local/etc/bash_completion:104: command not found: complete
/usr/local/etc/bash_completion:105: command not found: complete
/usr/local/etc/bash_completion:106: command not found: complete
/usr/local/etc/bash_completion:107: command not found: complete
/usr/local/etc/bash_completion:108: command not found: complete
/usr/local/etc/bash_completion:110: command not found: complete
/usr/local/etc/bash_completion:111: command not found: complete
/usr/local/etc/bash_completion:112: command not found: complete
/usr/local/etc/bash_completion:113: command not found: complete
/usr/local/etc/bash_completion:114: command not found: complete
/usr/local/etc/bash_completion:115: command not found: complete
/usr/local/etc/bash_completion:116: command not found: complete
/usr/local/etc/bash_completion:117: command not found: complete
/usr/local/etc/bash_completion:118: command not found: complete
/usr/local/etc/bash_completion:119: command not found: complete
/usr/local/etc/bash_completion:120: command not found: complete
/usr/local/etc/bash_completion:121: command not found: complete
/usr/local/etc/bash_completion:122: command not found: complete
/usr/local/etc/bash_completion:123: command not found: complete
/usr/local/etc/bash_completion:124: command not found: complete
/usr/local/etc/bash_completion:125: command not found: complete
/usr/local/etc/bash_completion:126: command not found: complete
/usr/local/etc/bash_completion:132: command not found: complete
/usr/local/etc/bash_completion:135: command not found: complete
/usr/local/etc/bash_completion:138: command not found: complete
/usr/local/etc/bash_completion:141: command not found: complete
/usr/local/etc/bash_completion:144: command not found: complete
/usr/local/etc/bash_completion:147: command not found: complete
/usr/local/etc/bash_completion:150: command not found: complete
/usr/local/etc/bash_completion:153: command not found: complete
/usr/local/etc/bash_completion:156: command not found: complete
/usr/local/etc/bash_completion:159: command not found: complete
/usr/local/etc/bash_completion:162: command not found: complete
/usr/local/etc/bash_completion:1192: command not found: shopt
/usr/local/etc/bash_completion:1367: command not found: complete
/usr/local/etc/bash_completion:1420: command not found: shopt
/usr/local/etc/bash_completion:1423: command not found: complete
/usr/local/etc/bash_completion:1523: command not found: complete
/usr/local/etc/bash_completion:1532: command not found: complete
/usr/local/etc/bash_completion:1577: command not found: complete
```

lrapetti, 2021-06-08:
> * Try to install binary packages via conda following the instructions in https://github.com/robotology/robotology-superbuild/blob/master/doc/conda-forge.md . In theory for our packages we already generate setup scripts for zsh via [multisheller](https://github.com/mamba-org/multisheller), see

- instruction in https://github.com/robotology/robotology-superbuild/blob/master/doc/install-miniforge.md#install-1 refer to `bash ./Miniforge3.sh ` instead of `zsh ./Miniforge3.sh`

- I was able to install the packages with conda (I just tried `conda install -c conda-forge -c robotology yarp icub-main`), and I tried to run the yarpserver. Let me know if you would need further testing on this.

traversaro, 2021-06-08:
>     * I was able to install the packages with conda (I just tried `conda install -c conda-forge -c robotology yarp icub-main`), and I tried to run the yarpserver. Let me know if you would need further testing on this.

Just a curiosity, can you try `yarp` command autocompletion?

lrapetti, 2021-06-08:
> > ```
> > * I was able to install the packages with conda (I just tried `conda install -c conda-forge -c robotology yarp icub-main`), and I tried to run the yarpserver. Let me know if you would need further testing on this.
> > ```
> 
> 
> Just a curiosity, can you try `yarp` command autocompletion?

It doesn't seem to be working (e.g. `yarp name list` no autocomplete).

Moreover, I have to other comments:
- Now I am getting the following message:
  ```
   $conda activate robotologyenv
  pathappend:3: bad substitution 
  pathappend:3: bad substitution
  pathappend:3: bad substitution 
  pathappend:3: bad substitution
  pathappend:3: bad substitution
  ```
  that was not there when I have activated it the first time. Anyway, basic functionality of yarp are still working.
- I tried to run few more applications that seems not to be working
  - yarpmanager
  ```
   $yarpmanager
  [ERROR] |yarp.os.Property| cannot read from ymanager.ini
  ```
  nothing is opened
  - yarpdataplayer
  ```
  $yarpdataplayer
  [ERROR] |yarp.os.Property| cannot read from config.ini
  [INFO] |yarp.os.Port| Port /yarpdataplayer/rpc:i active at tcp://10.240.9.60:10002/
  QObject::connect: No such signal MainWindow::internalStep(Bottle*)
  QObject::connect:  (sender name:   'MainWindow')
  QObject::connect:  (receiver name: 'MainWindow')
  QShortcut: No widget parent defined
  QObject::connect: Incompatible sender/receiver arguments
          QShortcut::activated() --> MainWindow::onInternalLoad(QString)
  QShortcut: No widget parent defined
  ```
  nothing is opened.

Note that I have commented out the sourcing to the `setup.sh` of the robotlogy-superbuild compiled from source

traversaro, 2021-06-12:
I renamed the issue as in https://github.com/dic-iit/training-conda-robotics/issues/2 @Giulero was able to reproduce problems similar to the one of @lrapetti also on Linux, and I was also able to reproduce them.

traversaro, 2021-06-12:
I tested a bit on Linux the `yarp_activate.zsh` script, but it is not working as intended. In particular, it seems that also on `zsh` the `yarp_activate.sh` is executed, while the `yarp_activate.zsh` is completely ignored. At this point, I supposed that the `.sh` script was meant for being compatible with both `bash` and `zsh`, so I tried to rename `yarp_activate.sh` in `yarp_activate.bash`, but the only difference seems that now both in bash and zsh `yarp_activate.bash` and `yarp_activate.zsh` are ignored.

traversaro, 2021-06-12:
I opened https://github.com/mamba-org/multisheller/issues/11 to understand how to tackle this issue.

traversaro, 2021-06-28:
Renaming `$CONDA_PREFIX/etc/conda/yarp_activate.sh` to `$CONDA_PREFIX/etc/conda/yarp_activate.bash`  and saving as `$CONDA_PREFIX/etc/conda/yarp_activate.sh` : 
~~~
if [ ! -z "$ZSH_VERSION" ]; then
  SCRIPT_DIR=${0:a:h}
  source ${SCRIPT_DIR}/yarp_activate.zsh
else
  SCRIPT_DIR="$( cd "$( dirname "${BASH_SOURCE[0]}" )" &> /dev/null && pwd )"
  source ${SCRIPT_DIR}/yarp_activate.bash
fi
~~~

Indeed after `conda activate yarp` in zsh the correct script is executed. However, once that is executed, `yarpdev --list` still does not work as expected. By inspecting the env variables, it is possible to see:
~~~
(yarp) \[\e]0;\u@\h: \w\a\]${debian_chroot:+($debian_chroot)}\[\033[01;32m\]\u@\h\[\033[00m\]:\[\033[01;34m\]\w\[\033[00m\]\$ env | grep YARP
YARP_DATA_DIRS=:$CONDA_PREFIX/share/yarp
~~~

The problem seems to be that `$CONDA_PREFIX` is reference verbatim in the env variable, while it should be expanded. Indeed by inspecting `yarp_activate.zsh` it is possible to see:
~~~
if [[ ! -z $COMSPEC ]];
then
    listappend "\$CONDA_PREFIX/Library\\share\\yarp" YARP_DATA_DIRS
    listappend "\$CONDA_PREFIX/Library\\share" XDG_DATA_DIRS
else
    listappend "\$CONDA_PREFIX/share/yarp" YARP_DATA_DIRS
    listappend "\$CONDA_PREFIX/share" XDG_DATA_DIRS
fi;
~~~

While the correct version should be:
~~~
if [[ ! -z $COMSPEC ]];
then
    listappend "$CONDA_PREFIX/Library\\share\\yarp" YARP_DATA_DIRS
    listappend "$CONDA_PREFIX/Library\\share" XDG_DATA_DIRS
else
    listappend "$CONDA_PREFIX/share/yarp" YARP_DATA_DIRS
    listappend "$CONDA_PREFIX/share" XDG_DATA_DIRS
fi;
~~~

Probably the `ensure_quotes` in https://github.com/mamba-org/multisheller/blob/8da3d64cf618879c9f021c40d9d770855fb0fc27/multisheller/backend/zsh.py#L5 is inserting the `\` members.

traversaro, 2021-06-29:
> Probably the `ensure_quotes` in https://github.com/mamba-org/multisheller/blob/8da3d64cf618879c9f021c40d9d770855fb0fc27/multisheller/backend/zsh.py#L5 is inserting the `\` members.

This was fixed by https://github.com/mamba-org/multisheller/pull/13 .

closed 2021-07-06 by traversaro

traversaro, 2021-07-30:
I just tested with zsh on Ubuntu 20.04, and everything seems to be working as expected.