robotology/robotology-superbuild #780: install_robotology_packages MATLAB helper: test in path with spaces

traversaro, 2021-06-01:
We should test the https://github.com/robotology/robotology-superbuild/blob/master/doc/matlab-one-line-install.md in a path with spaces, and verify that it prints a clear message for a user. 

If I remember correctly, the `miniforge`  installer does not support at the moment to install in path with spaces (and that is the reason why we have a separated instructions for MATLAB online: https://github.com/robotology/robotology-superbuild/blob/master/doc/matlab-one-line-install.md#installation-on-matlab-online).

Even if installing in path with spaces is not supported and is not a problem that it is easy to fix being in upstream miniforge/mambaforge, we should clearly document it and print a clear error for the user in case this happens.

traversaro, 2021-06-01:
To give an example, this is the error that @DanielePucci experienced:
~~~
Installing robotology MATLAB/Simulink binaries in /Users/danielepucci/Dropbox/My Mac (Danieles-MBP.homenet.telecomitalia.it)/Documents/MATLAB/robotology-matlab
Downloading mambaforge installer 
Download of mambaforge installer completed
Installing mambaforge
zsh:1: no matches found: (Danieles-MBP.homenet.telecomitalia.it)/Documents/MATLAB/robotology-matlab
Installation of mambaforge completed
Installation in /Users/danielepucci/Dropbox/My Mac (Danieles-MBP.homenet.telecomitalia.it)/Documents/MATLAB/robotology-matlab failed for unknown reason, please open an issue at https://github.com/robotology/robotology-superbuild/issues/new
~~~

and this is what I get on MATLAB online (that by default opens the MATLAB interpreter in `/MATLAB Drive` , with a space:
~~~
>> install_robotology_packages
Installing robotology MATLAB/Simulink binaries in /MATLAB Drive/robotology-matlab
Downloading mambaforge installer 
Download of mambaforge installer completed
Installing mambaforge
ERROR: did not recognize option 'Drive/robotology-matlab', please try -h
Installation of mambaforge completed
Installation in /MATLAB Drive/robotology-matlab failed for unknown reason, please open an issue at https://github.com/robotology/robotology-superbuild/issues/new
~~~

traversaro, 2021-06-01:
Indeed, I try to see if I could fix the invocation of the command in when `install_path` has a space, but even by fixing the problem in MATLAB, I then get the conda error:
~~~
>> system('sh Mambaforge-Linux-x86_64.sh -b -p "/MATLAB Drive/robotology-matlab"')
ERROR: did not recognize option 'Drive/robotology-matlab', please try -h

ans =

     1

>> system('sh Mambaforge-Linux-x86_64.sh -b -p "/MATLAB\ Drive/robotology-matlab"')
ERROR: Cannot install into directories with spaces

ans =

     1

>> 
~~~

traversaro, 2021-06-01:
Related issue: https://github.com/ContinuumIO/anaconda-issues/issues/716 . Apparently a recent PR was opened to fix this: https://github.com/conda/constructor/pull/449 .

traversaro, 2021-06-01:
Until https://github.com/conda/constructor/pull/449 is merged and a new release of constructor tagged, we can temporary add an error if a user try to install in a path with directory. The alternative is to build our own `mambaforge`-like installers using the patched constructor, but to be honest I would prefer to avoid that unless it is strictly necessary. 

@DanielePucci do you think that for your (and more importantly, for users like you) it is doable/easy to just run the `install_robotology_packages` in a directory without spaces, or it is complicated/a dealbreaker?

DanielePucci, 2021-06-02:
@traversaro 
> @DanielePucci do you think that for your (and more importantly, for users like you) it is doable/easy to just run the install_robotology_packages in a directory without spaces, or it is complicated/a dealbreaker?

I think it's ok, dumb users like me could still make it if it is properly documented with a bug message

traversaro, 2021-06-02:
> dumb users like me 

Actually I meant [executive management users](https://en.wikipedia.org/wiki/Senior_management),  that seldom interacts with code anymore even if they used to do in their previous role. Such users typically work on Windows/macOS (hence the focus on multi-platform compatibility). In particular, I was not afraid of the technical difficulty of creating a directory without a space, but rather the technical convenience of doing so. In your case for example, it seems that you were creating the directory inside your Dropbox personal folder that had a space in the path, perhaps that is a common workflow and it is tricky for whatever reason to do something outside that folder. Another thing I would be afraid of is users that have space in their username, for them creating the folder in a path without a space may be quite complicated.

DanielePucci, 2021-06-02:
@traversaro 
>  In your case for example, it seems that you were creating the directory inside your Dropbox personal folder that had a space in the path, perhaps that is a common workflow and it is tricky for whatever reason to do something outside that folder. 

Yeah that's true, since it is a sort of handy to have the new folders inside automatic backup systems like Dropbox, but I sincerely do not know if this is a common habit among the [executive management users](https://youtu.be/GENEBQ0SvfE)

> Another thing I would be afraid of is users that have space in their username, for them creating the folder in a path without a space may be quite complicated.

That's true, in that case I see it to be much more complex to make folders without spaces

nunoguedelha, 2021-06-02:
> That's true, in that case I see it to be much more complex to make folders without spaces

I haven't tried this personnaly, but maybe creating a symbolic link, without spaces, to the home folder would be a viable workaround. It would require an additional line in the installation steps, but it would still be acceptable...

traversaro, 2021-06-02:
I just tested on Windows with `C:\src\test with space` and the good news is that on Windows there is no such problem, so the problem is just for Linux and macOS .

traversaro, 2021-06-02:
> I haven't tried this personnaly, but maybe creating a symbolic link, without spaces, to the home folder would be a viable workaround. It would require an additional line in the installation steps, [...]

Which additional command do you have in mind? The big problem is that if there is a writable directory somewhere without any space where to create the symbolic link, perhaps it is just easier for the user to move there rather then create a symbolic link.

traversaro, 2021-06-02:
I fixed the installation in path with spaces on Windows and added a clear error message in https://github.com/robotology/robotology-superbuild/pull/781 . Once https://github.com/conda/constructor/pull/449 is merged and a new miniforge release is done, we can remove the check and everything will work.

nunoguedelha, 2021-06-02:
> Which additional command do you have in mind?

a command to create the link to the home folder, like `ln -s` on MacOS, but the best would be a script to do it in a more portable way.

> perhaps it is just easier for the user to move there rather then create a symbolic link.

I was thinking about this use case:

> Yeah that's true, since it is a sort of handy to have the new folders inside automatic backup systems like Dropbox, but I sincerely do not know if this is a common habit among the executive management users

closed 2021-06-07 by traversaro