robotology/robotology-superbuild #785: Conda\Linux CI failure

traversaro, 2021-06-05:
Since Today, the Conda\Linux CI is failing with error:
~~~
 Looking for: ['libignition-gazebo4']

Encountered problems while solving:
  - package python-3.6.12-4_73_pypy requires python_abi 3.6 *_pypy36_pp73, but none of the providers can be installed
~~~

Example of failing job: https://github.com/robotology/robotology-superbuild/pull/784/checks?check_run_id=2753897981 .

traversaro, 2021-06-05:
Installing `libignition-gazebo4` on an empty environment works fine with both conda and mamba (tested in a local install), so probably this is a problem related to some other package installed earlier by the CI.

traversaro, 2021-06-06:
By inspecting the logs, it seems that for some reason the step that install regular dependencies:
~~~
mamba install ace asio boost eigen gazebo glew glfw gsl ipopt irrlicht libjpeg-turbo libmatio libode libxml2 nlohmann_json opencv pkg-config portaudio qt sdl sdl2 sqlite tinyxml spdlog
~~~
started to install the Python package provided by pypy 3.7 instead of regular python 3.9 .

traversaro, 2021-06-06:
I just tried with a local mamba install, and:
~~~
mamba install python
~~~
wants to install Python 3.7 as we would expect.

On the other hand, also locally
~~~
mamba install ace asio boost eigen gazebo glew glfw gsl ipopt irrlicht libjpeg-turbo libmatio libode libxml2 nlohmann_json opencv pkg-config portaudio qt sdl sdl2 sqlite tinyxml spdlog
~~~
results in
~~~
  _libgcc_mutex                       0.1  conda_forge          conda-forge/linux-64     Cached
  _openmp_mutex                       4.5  1_gnu                conda-forge/linux-64     Cached
  ace                               7.0.2  h9c3ff4c_0           conda-forge/linux-64       7 MB
  alsa-lib                          1.2.3  h516909a_0           conda-forge/linux-64     560 KB
  ampl-mp                           3.1.0  h616b090_1004        conda-forge/linux-64       1 MB
  asio                             1.14.1  h9c3ff4c_1           conda-forge/linux-64     333 KB
  boost                            1.74.0  py36hb4f4ce5_3       conda-forge/linux-64     364 KB
  boost-cpp                        1.74.0  hc6e9bd1_3           conda-forge/linux-64      16 MB
  bzip2                             1.0.8  h7f98852_4           conda-forge/linux-64     Cached
  c-ares                           1.17.1  h7f98852_1           conda-forge/linux-64     Cached
  ca-certificates               2021.5.30  ha878542_0           conda-forge/linux-64     Cached
  cairo                            1.16.0  h6cf1ce9_1008        conda-forge/linux-64       1 MB
  certifi                       2021.5.30  py36hd000896_0       conda-forge/linux-64     141 KB
  console_bridge                    1.0.1  hc9558a2_0           conda-forge/linux-64     Cached
  cppzmq                            4.7.1  hf7cf922_2           conda-forge/linux-64      25 KB
  curl                             7.77.0  hea6ffbf_0           conda-forge/linux-64     149 KB
  dbus                             1.13.6  h48d8840_2           conda-forge/linux-64     572 KB
  eigen                             3.3.9  h4bd325d_1           conda-forge/linux-64     970 KB
  expat                             2.4.1  h9c3ff4c_0           conda-forge/linux-64     Cached
  ffmpeg                            4.3.1  h03821db_2           conda-forge/linux-64      92 MB
  fontconfig                       2.13.1  hba837de_1005        conda-forge/linux-64     357 KB
  freeimage                        3.18.0  hadfbfc9_7           conda-forge/linux-64     477 KB
  freetype                         2.10.4  h0708190_1           conda-forge/linux-64     890 KB
  gazebo                           11.5.0  h5fc56bc_1           conda-forge/linux-64      63 MB
  gdbm                               1.18  h0a1914f_2           conda-forge/linux-64     190 KB
  gettext                        0.19.8.1  h0b5b191_1005        conda-forge/linux-64       4 MB
  glew                              2.1.0  h9c3ff4c_2           conda-forge/linux-64     647 KB
  glfw                              3.3.4  h7f98852_0           conda-forge/linux-64     126 KB
  glib                             2.68.2  h9c3ff4c_2           conda-forge/linux-64     450 KB
  glib-tools                       2.68.2  h9c3ff4c_2           conda-forge/linux-64      86 KB
  gmp                               6.2.1  h58526e2_0           conda-forge/linux-64     806 KB
  gnutls                           3.6.13  h85f3911_1           conda-forge/linux-64       2 MB
  graphite2                        1.3.13  he1b5a44_1001        conda-forge/linux-64     102 KB
  gsl                                 2.6  he838d99_2           conda-forge/linux-64       3 MB
  gst-plugins-base                 1.18.4  hf529b03_2           conda-forge/linux-64       3 MB
  gstreamer                        1.18.4  h76c114f_2           conda-forge/linux-64       2 MB
  gts                               0.7.6  h1fc3fae_2           conda-forge/linux-64     388 KB
  harfbuzz                          2.8.1  h83ec7ef_0           conda-forge/linux-64       2 MB
  hdf5                             1.10.6  nompi_h6a2412b_1114  conda-forge/linux-64       3 MB
  icu                                68.1  h58526e2_0           conda-forge/linux-64     Cached
  ilmbase                           2.5.5  h780b84a_0           conda-forge/linux-64     343 KB
  ipopt                            3.13.4  h7ede334_0           conda-forge/linux-64       1 MB
  irrlicht                          1.8.4  hdaeae82_1           conda-forge/linux-64       2 MB
  jasper                          1.900.1  h07fcdf6_1006        conda-forge/linux-64     286 KB
  jpeg                                 9d  h516909a_0           conda-forge/linux-64     Cached
  jsoncpp                           1.9.4  h4bd325d_2           conda-forge/linux-64     310 KB
  jxrlib                              1.1  h516909a_2           conda-forge/linux-64     236 KB
  krb5                             1.19.1  hcc1bbae_0           conda-forge/linux-64     Cached
  lame                              3.100  h14c3975_1001        conda-forge/linux-64     498 KB
  lcms2                              2.12  hddcbb42_0           conda-forge/linux-64     443 KB
  libblas                           3.9.0  9_openblas           conda-forge/linux-64      11 KB
  libcblas                          3.9.0  9_openblas           conda-forge/linux-64      11 KB
  libccd                              2.1  he1b5a44_1           conda-forge/linux-64      33 KB
[... cut ...]
  xz                                5.2.5  h516909a_1           conda-forge/linux-64     Cached
  yaml                              0.2.5  h516909a_0           conda-forge/linux-64     Cached
  zeromq                            4.3.4  h9c3ff4c_0           conda-forge/linux-64     Cached
  zlib                             1.2.11  h516909a_1010        conda-forge/linux-64     Cached
  zstd                              1.4.9  ha95c52a_0           conda-forge/linux-64     431 KB
  zziplib                         0.13.69  hed695b0_1           conda-forge/linux-64     Cached
~~~

traversaro, 2021-06-06:
The problem seems to occur only with mamba, as with conda CPython 3.9 is correctly installed:
~~~
  _libgcc_mutex      conda-forge/linux-64::_libgcc_mutex-0.1-conda_forge
  _openmp_mutex      conda-forge/linux-64::_openmp_mutex-4.5-1_gnu
  ace                conda-forge/linux-64::ace-7.0.2-h9c3ff4c_0
  alsa-lib           conda-forge/linux-64::alsa-lib-1.2.3-h516909a_0
  ampl-mp            conda-forge/linux-64::ampl-mp-3.1.0-h616b090_1004
  asio               conda-forge/linux-64::asio-1.14.1-h9c3ff4c_1
  assimp             conda-forge/linux-64::assimp-5.0.1-hedfc422_5
  atk-1.0            conda-forge/linux-64::atk-1.0-2.36.0-h3371d22_4
  boost              conda-forge/linux-64::boost-1.74.0-py39h5472131_3
  boost-cpp          conda-forge/linux-64::boost-cpp-1.74.0-hc6e9bd1_3
  bullet-cpp         conda-forge/linux-64::bullet-cpp-3.09-hde0f152_2
  bzip2              conda-forge/linux-64::bzip2-1.0.8-h7f98852_4
  c-ares             conda-forge/linux-64::c-ares-1.17.1-h7f98852_1
  ca-certificates    conda-forge/linux-64::ca-certificates-2021.5.30-ha878542_0
  cairo              conda-forge/linux-64::cairo-1.16.0-h6cf1ce9_1008
  certifi            conda-forge/linux-64::certifi-2021.5.30-py39hf3d152e_0
  cfitsio            conda-forge/linux-64::cfitsio-3.470-hb418390_7
  console_bridge     conda-forge/linux-64::console_bridge-1.0.1-h4bd325d_0
  cppzmq             conda-forge/linux-64::cppzmq-4.7.1-hf7cf922_2
  curl               conda-forge/linux-64::curl-7.77.0-hea6ffbf_0
  dartsim            conda-forge/linux-64::dartsim-6.10.1-h53b96f2_0
  dbus               conda-forge/linux-64::dbus-1.13.6-h48d8840_2
  eigen              conda-forge/linux-64::eigen-3.3.9-h4bd325d_1
  expat              conda-forge/linux-64::expat-2.4.1-h9c3ff4c_0
  fcl                conda-forge/linux-64::fcl-0.6.1-h2cbc392_3
  ffmpeg             conda-forge/linux-64::ffmpeg-4.3.1-hca11adc_2
  flann              conda-forge/linux-64::flann-1.9.1-h2e58136_1008
  font-ttf-dejavu-s~ conda-forge/noarch::font-ttf-dejavu-sans-mono-2.37-hab24e00_0
  font-ttf-inconsol~ conda-forge/noarch::font-ttf-inconsolata-3.000-h77eed37_0
  font-ttf-source-c~ conda-forge/noarch::font-ttf-source-code-pro-2.038-h77eed37_0
  font-ttf-ubuntu    conda-forge/noarch::font-ttf-ubuntu-0.83-hab24e00_0
  fontconfig         conda-forge/linux-64::fontconfig-2.13.1-hba837de_1005
  fonts-conda-ecosy~ conda-forge/noarch::fonts-conda-ecosystem-1-0
  fonts-conda-forge  conda-forge/noarch::fonts-conda-forge-1-0
  freeglut           conda-forge/linux-64::freeglut-3.2.1-h9c3ff4c_2
  freeimage          conda-forge/linux-64::freeimage-3.18.0-h88c329d_7
  freetype           conda-forge/linux-64::freetype-2.10.4-h0708190_1
  freexl             conda-forge/linux-64::freexl-1.0.6-h7f98852_0
  fribidi            conda-forge/linux-64::fribidi-1.0.10-h36c2ea0_0
  gazebo             conda-forge/linux-64::gazebo-11.5.1-haa37ec6_2
  gdbm               conda-forge/linux-64::gdbm-1.18-h0a1914f_2
  gdk-pixbuf         conda-forge/linux-64::gdk-pixbuf-2.42.6-h04a7f16_0
  geos               conda-forge/linux-64::geos-3.9.1-h9c3ff4c_2
  geotiff            conda-forge/linux-64::geotiff-1.6.0-hcf90da6_5
  gettext            conda-forge/linux-64::gettext-0.19.8.1-h0b5b191_1005
  giflib             conda-forge/linux-64::giflib-5.2.1-h36c2ea0_2
  glew               conda-forge/linux-64::glew-2.1.0-h9c3ff4c_2
  glfw               conda-forge/linux-64::glfw-3.3.4-h7f98852_0
  glib               conda-forge/linux-64::glib-2.68.2-h9c3ff4c_2
  glib-tools         conda-forge/linux-64::glib-tools-2.68.2-h9c3ff4c_2
  gmp                conda-forge/linux-64::gmp-6.2.1-h58526e2_0
  gnutls             conda-forge/linux-64::gnutls-3.6.13-h85f3911_1
  graphite2          conda-forge/linux-64::graphite2-1.3.13-h58526e2_1001
  graphviz           conda-forge/linux-64::graphviz-2.47.2-h85b4f2f_0
  gsl                conda-forge/linux-64::gsl-2.6-he838d99_2
  gst-plugins-base   conda-forge/linux-64::gst-plugins-base-1.18.4-hf529b03_2
  gstreamer          conda-forge/linux-64::gstreamer-1.18.4-h76c114f_2
  gtk2               conda-forge/linux-64::gtk2-2.24.33-h539f30e_1
  gts                conda-forge/linux-64::gts-0.7.6-h64030ff_2
  harfbuzz           conda-forge/linux-64::harfbuzz-2.8.1-h83ec7ef_0
  hdf4               conda-forge/linux-64::hdf4-4.2.15-h10796ff_3
  hdf5               conda-forge/linux-64::hdf5-1.10.6-nompi_h6a2412b_1114
  icu                conda-forge/linux-64::icu-68.1-h58526e2_0
  ilmbase            conda-forge/linux-64::ilmbase-2.5.5-h780b84a_0
  ipopt              conda-forge/linux-64::ipopt-3.13.4-h7ede334_0
  irrlicht           conda-forge/linux-64::irrlicht-1.8.4-hdaeae82_1
  jasper             conda-forge/linux-64::jasper-1.900.1-h07fcdf6_1006
  jpeg               conda-forge/linux-64::jpeg-9d-h36c2ea0_0
[... 138 lines cut ...]
  xorg-libx11        conda-forge/linux-64::xorg-libx11-1.6.12-h36c2ea0_0
  xorg-libxau        conda-forge/linux-64::xorg-libxau-1.0.9-h7f98852_0
  xorg-libxaw        conda-forge/linux-64::xorg-libxaw-1.0.14-h7f98852_0
  xorg-libxcursor    conda-forge/linux-64::xorg-libxcursor-1.2.0-h516909a_0
  xorg-libxdmcp      conda-forge/linux-64::xorg-libxdmcp-1.1.3-h7f98852_0
  xorg-libxext       conda-forge/linux-64::xorg-libxext-1.3.4-h516909a_0
  xorg-libxfixes     conda-forge/linux-64::xorg-libxfixes-5.0.3-h516909a_1004
  xorg-libxi         conda-forge/linux-64::xorg-libxi-1.7.10-h516909a_0
  xorg-libxinerama   conda-forge/linux-64::xorg-libxinerama-1.1.4-h9c3ff4c_1001
[... cut ...]
  xorg-xextproto     conda-forge/linux-64::xorg-xextproto-7.3.0-h7f98852_1002
  xorg-xproto        conda-forge/linux-64::xorg-xproto-7.0.31-h7f98852_1007
  xz                 conda-forge/linux-64::xz-5.2.5-h516909a_1
  yaml               conda-forge/linux-64::yaml-0.2.5-h516909a_0
  zeromq             conda-forge/linux-64::zeromq-4.3.4-h9c3ff4c_0
  zlib               conda-forge/linux-64::zlib-1.2.11-h516909a_1010
  zstd               conda-forge/linux-64::zstd-1.4.9-ha95c52a_0
  zziplib            conda-forge/linux-64::zziplib-0.13.69-h27826a3_1
~~~

traversaro, 2021-06-06:
I narrowed down the problem to mamba when it is trying to install boost and gazebo, in that case we have:
~~~
(test) root@4442114f7f41:/# mamba install boost gazebo

                  __    __    __    __
                 /  \  /  \  /  \  /  \
                /    \/    \/    \/    \
███████████████/  /██/  /██/  /██/  /████████████████████████
              /  / \   / \   / \   / \  \____
             /  /   \_/   \_/   \_/   \    o \__,
            / _/                       \_____/  `
            |/
        ███╗   ███╗ █████╗ ███╗   ███╗██████╗  █████╗
        ████╗ ████║██╔══██╗████╗ ████║██╔══██╗██╔══██╗
        ██╔████╔██║███████║██╔████╔██║██████╔╝███████║
        ██║╚██╔╝██║██╔══██║██║╚██╔╝██║██╔══██╗██╔══██║
        ██║ ╚═╝ ██║██║  ██║██║ ╚═╝ ██║██████╔╝██║  ██║
        ╚═╝     ╚═╝╚═╝  ╚═╝╚═╝     ╚═╝╚═════╝ ╚═╝  ╚═╝

        mamba (0.13.0) supported by @QuantStack

        GitHub:  https://github.com/mamba-org/mamba
        Twitter: https://twitter.com/QuantStack

█████████████████████████████████████████████████████████████

Looking for: ['boost', 'gazebo']

conda-forge/linux-64     [====================] (00m:00s) No change
conda-forge/noarch       [====================] (00m:03s) Done
Transaction

  Prefix: /root/miniforge3/envs/test

  Updating specs:

   - boost
   - gazebo

  Package                         Version  Build                Channel                    Size
─────────────────────────────────────────────────────────────────────────────────────────────────
  Install:
─────────────────────────────────────────────────────────────────────────────────────────────────

  _libgcc_mutex                       0.1  conda_forge          conda-forge/linux-64     Cached
  _openmp_mutex                       4.5  1_gnu                conda-forge/linux-64     Cached
  alsa-lib                          1.2.3  h516909a_0           conda-forge/linux-64     Cached
  boost                            1.74.0  py36hb4f4ce5_3       conda-forge/linux-64     364 KB
  boost-cpp                        1.74.0  hc6e9bd1_3           conda-forge/linux-64     Cached
  bzip2                             1.0.8  h7f98852_4           conda-forge/linux-64     Cached
  c-ares                           1.17.1  h7f98852_1           conda-forge/linux-64     Cached
  ca-certificates               2021.5.30  ha878542_0           conda-forge/linux-64     Cached
  certifi                       2021.5.30  py36hd000896_0       conda-forge/linux-64     141 KB
  console_bridge                    1.0.1  hc9558a2_0           conda-forge/linux-64     Cached
  cppzmq                            4.7.1  hf7cf922_2           conda-forge/linux-64     Cached
  curl                             7.77.0  hea6ffbf_0           conda-forge/linux-64     Cached
  dbus                             1.13.6  h48d8840_2           conda-forge/linux-64     Cached
  eigen                             3.3.9  h4bd325d_1           conda-forge/linux-64     Cached
  expat                             2.4.1  h9c3ff4c_0           conda-forge/linux-64     Cached
  ffmpeg                            4.3.1  h03821db_2           conda-forge/linux-64      92 MB
  fontconfig                       2.13.1  hba837de_1005        conda-forge/linux-64     Cached
  freeimage                        3.18.0  hadfbfc9_7           conda-forge/linux-64     477 KB
  freetype                         2.10.4  h0708190_1           conda-forge/linux-64     Cached
  gazebo                           11.5.0  h5fc56bc_1           conda-forge/linux-64      63 MB
  gdbm                               1.18  h0a1914f_2           conda-forge/linux-64     Cached
  gettext                        0.19.8.1  h0b5b191_1005        conda-forge/linux-64     Cached
  glib                             2.68.2  h9c3ff4c_2           conda-forge/linux-64     Cached
  glib-tools                       2.68.2  h9c3ff4c_2           conda-forge/linux-64     Cached
  gmp                               6.2.1  h58526e2_0           conda-forge/linux-64     Cached
  gnutls                           3.6.13  h85f3911_1           conda-forge/linux-64     Cached
[... 337 lines cut ...]
  xorg-kbproto       conda-forge/linux-64::xorg-kbproto-1.0.7-h7f98852_1002
  xorg-libice        conda-forge/linux-64::xorg-libice-1.0.10-h7f98852_0
  xorg-libsm         conda-forge/linux-64::xorg-libsm-1.2.3-hd9c2040_1000
  xorg-libx11        conda-forge/linux-64::xorg-libx11-1.6.12-h36c2ea0_0
  xorg-libxau        conda-forge/linux-64::xorg-libxau-1.0.9-h7f98852_0
  xorg-libxaw        conda-forge/linux-64::xorg-libxaw-1.0.14-h7f98852_0
  xorg-libxdmcp      conda-forge/linux-64::xorg-libxdmcp-1.1.3-h7f98852_0
  xorg-libxext       conda-forge/linux-64::xorg-libxext-1.3.4-h516909a_0
  xorg-libxfixes     conda-forge/linux-64::xorg-libxfixes-5.0.3-h516909a_1004
  xorg-libxi         conda-forge/linux-64::xorg-libxi-1.7.10-h516909a_0
  xorg-libxmu        conda-forge/linux-64::xorg-libxmu-1.1.3-h516909a_0
  xorg-libxpm        conda-forge/linux-64::xorg-libxpm-3.5.13-h516909a_0
  xorg-libxrender    conda-forge/linux-64::xorg-libxrender-0.9.10-h516909a_1002
  xorg-libxt         conda-forge/linux-64::xorg-libxt-1.1.5-h516909a_1003
  xorg-renderproto   conda-forge/linux-64::xorg-renderproto-0.11.1-h7f98852_1002
  xorg-xextproto     conda-forge/linux-64::xorg-xextproto-7.3.0-h7f98852_1002
  xorg-xproto        conda-forge/linux-64::xorg-xproto-7.0.31-h7f98852_1007
  xz                 conda-forge/linux-64::xz-5.2.5-h516909a_1
  yaml               conda-forge/linux-64::yaml-0.2.5-h516909a_0
  zeromq             conda-forge/linux-64::zeromq-4.3.4-h9c3ff4c_0
  zlib               conda-forge/linux-64::zlib-1.2.11-h516909a_1010
  zstd               conda-forge/linux-64::zstd-1.4.9-ha95c52a_0
  zziplib            conda-forge/linux-64::zziplib-0.13.69-h27826a3_1

~~~

traversaro, 2021-06-06:
I opened an issue for this in https://github.com/mamba-org/mamba/issues/986 .

wolfv, 2021-06-06:
One quick fix is to require cpython explicitly (python=3.8*=*cpython) but I'll have a closer look tomorrow on what's going on. Looks like my fix on libsolv wasn't sufficient :/

traversaro, 2021-06-06:
> 
> 
> One quick fix is to require cpython explicitly (python=3.8*=*cpython) but I'll have a closer look tomorrow on what's going on. Looks like my fix on libsolv wasn't sufficient :/

Thanks! I noticed that we actually did not needed full boost, so for now I added https://github.com/robotology/robotology-superbuild/pull/786 , let's see if that helps. As additional data point, note that this CI runs nightly, and until 2 days ago (see https://github.com/robotology/robotology-superbuild/actions/runs/908249889) everything was fine, while it started having this problem just last night: https://github.com/robotology/robotology-superbuild/actions/runs/908249889 .

traversaro, 2021-07-20:
Fixed in https://github.com/mamba-org/mamba/issues/986 and verified in https://github.com/robotology/robotology-superbuild/pull/833 .

closed 2021-07-20 by traversaro