robotology/robotology-superbuild #792: Document how to add Windows Terminal profile for a given Conda environment

traversaro, 2021-06-09:
Similar to https://dev.to/voodu/windows-terminal-conda-d3e, but now Windows Terminal permit to add new profile from GUI:

[image]

traversaro, 2021-09-03:
Interestingly, @sgiraz did not have the "Add new" section. I just installed Windows Terminal from Windows Store, and I have it:
[image]

sgiraz, 2021-09-03:
Yeah, I noticed that too. In my case, the last time, this option wasn't visible due to the small size of the window.

Said that, now I'm trying the new [Windows Terminal Preview](https://www.microsoft.com/en-us/p/windows-terminal-preview/9n8g5rfz9xk3?activetab=pivot:overviewtab) which contains the latest features as they are developed.

For example, an useful option I found is the possibility to split the tab using context menu maintaining the current profile. Of course there have been introduced more other [interesting features](https://devblogs.microsoft.com/commandline/windows-terminal-preview-1-11-release/).

traversaro, 2021-09-03:
Another useful trick to document is to define a profile that both activate a conda environent, *and* also calls the custom `setup.bat` scripts that needs to be called when the superbuild is compiled from source (see https://github.com/robotology/robotology-superbuild/blob/master/doc/conda-forge.md#run-software-installed). Probably we can steal the same trick used in ROSOnWindows docs (see http://wiki.ros.org/Installation/Windows). The trick is to concatenate commands with `&&` .

sgiraz, 2021-09-06:
Yep. Let me paste here the screen of the profile just for reference:

[image]

In general, the command has the following pattern:
`cmd.exe /k %HOME%/AppData/Local/mambaforge/Scripts/activate.bat <your-conda-env> && call <your robotology-superbuild-installation-path>/build/install/share/robotology-superbuild/setup.bat`

traversaro, 2021-11-11:
According to @S-Dafarra this was something he was missing during one of the first actual tests of the conda-based Windows robotology-superbuild installation. In particular, for actual demos/deployments it would be also useful to also have specific desktop shortcuts that just open a specific Windows Terminal profile. For that the info in https://github.com/microsoft/terminal/issues/6366 could be useful.

S-Dafarra, 2021-11-11:
Nice to know! I think we can start with the custom profile for the time being

S-Dafarra, 2021-11-11:
Thanks @traversaro @sgiraz, I have added a custom profile in a Windows machine in our setup (``icub-virtualizer``). In our case the installation of ``mamba`` was in a different folder. Here the command I used inside the profile:
```
cmd.exe /k C:\mambaforge\Scripts\activate.bat robsub && call C:\robotology-superbuild\build\install\share\robotology-superbuild\setup.bat
```

I have also found a *yarpino* icon in ``YARP`` and used it as an icon.
[image]

 Here how it looks when opening ``Windows Terminal``:
[image]

cc @isorrentino @RiccardoGrieco @kouroshD @lrapetti

G-Cervettini, 2022-09-12:
@traversaro I'm struggling with an error while trying to install Windows Terminal on the new ALIENWARE IITICUBLAP290.
Do you know what could be going on here?
[image]