robotology/robotology-superbuild #794: Unstable projects tags CI failure 2021/06/12

traversaro, 2021-06-12:
Yesterday's nightly job shown that `icub-main` devel does not compile anymore with `yarp` master branch:
~~~
2021-06-12T02:33:33.3075776Z FAILED: src/libraries/icubmod/canBusMotionControl/CMakeFiles/canmotioncontrol.dir/CanBusMotionControl.cpp.o 
2021-06-12T02:33:33.3084803Z /usr/bin/c++  -DICUB_CANMASKS_STRICT_FILTER -DICUB_CANPROTOCOL_STRICT -D_REENTRANT -D__ACE_INLINE__ -Dcanmotioncontrol_EXPORTS -I/__w/robotology-superbuild/robotology-superbuild/src/ICUB/src/libraries/icubmod/canBusMotionControl -I/__w/robotology-superbuild/robotology-superbuild/src/ICUB/src/libraries/icubmod/canBusMotionControl/../motionControlLib -I/__w/robotology-superbuild/robotology-superbuild/src/ICUB/src/libraries/iCubDev/include -isystem /__w/robotology-superbuild/robotology-superbuild/build/install/include -isystem /__w/robotology-superbuild/robotology-superbuild/build/install/include/icub_firmware_shared/canProtocolLib -g -fPIC   -std=c++14 -MD -MT src/libraries/icubmod/canBusMotionControl/CMakeFiles/canmotioncontrol.dir/CanBusMotionControl.cpp.o -MF src/libraries/icubmod/canBusMotionControl/CMakeFiles/canmotioncontrol.dir/CanBusMotionControl.cpp.o.d -o src/libraries/icubmod/canBusMotionControl/CMakeFiles/canmotioncontrol.dir/CanBusMotionControl.cpp.o -c /__w/robotology-superbuild/robotology-superbuild/src/ICUB/src/libraries/icubmod/canBusMotionControl/CanBusMotionControl.cpp
2021-06-12T02:33:33.3092711Z In file included from /__w/robotology-superbuild/robotology-superbuild/src/ICUB/src/libraries/icubmod/canBusMotionControl/CanBusMotionControl.cpp:40:
2021-06-12T02:33:33.3095215Z /__w/robotology-superbuild/robotology-superbuild/src/ICUB/src/libraries/icubmod/canBusMotionControl/CanBusMotionControl.h:667:44: error: expected class-name before ',' token
2021-06-12T02:33:33.3096803Z   667 |             public ImplementPositionControl,
2021-06-12T02:33:33.3097463Z       |                                            ^
2021-06-12T02:33:33.3099109Z /__w/robotology-superbuild/robotology-superbuild/src/ICUB/src/libraries/icubmod/canBusMotionControl/CanBusMotionControl.h:669:44: error: expected class-name before ',' token
2021-06-12T02:33:33.3100691Z   669 |             public ImplementVelocityControl,
2021-06-12T02:33:33.3101337Z       |                                            ^
2021-06-12T02:33:33.3103582Z /__w/robotology-superbuild/robotology-superbuild/src/ICUB/src/libraries/icubmod/canBusMotionControl/CanBusMotionControl.cpp: In constructor 'yarp::dev::CanBusMotionControl::CanBusMotionControl()':
2021-06-12T02:33:33.3107292Z /__w/robotology-superbuild/robotology-superbuild/src/ICUB/src/libraries/icubmod/canBusMotionControl/CanBusMotionControl.cpp:2393:1: error: class 'yarp::dev::CanBusMotionControl' does not have any field named 'ImplementPositionControl'
2021-06-12T02:33:33.3109234Z  2393 | ImplementPositionControl(this),
2021-06-12T02:33:33.3109842Z       | ^~~~~~~~~~~~~~~~~~~~~~~~
2021-06-12T02:33:33.3112204Z /__w/robotology-superbuild/robotology-superbuild/src/ICUB/src/libraries/icubmod/canBusMotionControl/CanBusMotionControl.cpp:2394:1: error: class 'yarp::dev::CanBusMotionControl' does not have any field named 'ImplementVelocityControl'
2021-06-12T02:33:33.3114141Z  2394 | ImplementVelocityControl(this),
2021-06-12T02:33:33.3114864Z       | ^~~~~~~~~~~~~~~~~~~~~~~~
2021-06-12T02:33:33.3117078Z /__w/robotology-superbuild/robotology-superbuild/src/ICUB/src/libraries/icubmod/canBusMotionControl/CanBusMotionControl.cpp: In member function 'virtual bool yarp::dev::CanBusMotionControl::open(yarp::os::Searchable&)':
2021-06-12T02:33:33.3120081Z /__w/robotology-superbuild/robotology-superbuild/src/ICUB/src/libraries/icubmod/canBusMotionControl/CanBusMotionControl.cpp:2507:5: error: 'ImplementPositionControl' has not been declared
2021-06-12T02:33:33.3122110Z  2507 |     ImplementPositionControl::initialize(p._njoints, p._axisMap, p._angleToEncoder, p._zeros);
2021-06-12T02:33:33.3122991Z       |     ^~~~~~~~~~~~~~~~~~~~~~~~
2021-06-12T02:33:33.3124823Z /__w/robotology-superbuild/robotology-superbuild/src/ICUB/src/libraries/icubmod/canBusMotionControl/CanBusMotionControl.cpp:2508:5: error: 'ImplementVelocityControl' has not been declared
2021-06-12T02:33:33.3126850Z  2508 |     ImplementVelocityControl::initialize(p._njoints, p._axisMap, p._angleToEncoder, p._zeros);
2021-06-12T02:33:33.3127803Z       |     ^~~~~~~~~~~~~~~~~~~~~~~~
2021-06-12T02:33:33.3129696Z /__w/robotology-superbuild/robotology-superbuild/src/ICUB/src/libraries/icubmod/canBusMotionControl/CanBusMotionControl.cpp:2606:9: error: 'setRefSpeed' was not declared in this scope; did you mean 'setRefSpeedRaw'?
2021-06-12T02:33:33.3131284Z  2606 |         setRefSpeed(j, 10.0);
2021-06-12T02:33:33.3131732Z       |         ^~~~~~~~~~~
2021-06-12T02:33:33.3132177Z       |         setRefSpeedRaw
2021-06-12T02:33:33.3134373Z /__w/robotology-superbuild/robotology-superbuild/src/ICUB/src/libraries/icubmod/canBusMotionControl/CanBusMotionControl.cpp: In member function 'virtual bool yarp::dev::CanBusMotionControl::close()':
2021-06-12T02:33:33.3137322Z /__w/robotology-superbuild/robotology-superbuild/src/ICUB/src/libraries/icubmod/canBusMotionControl/CanBusMotionControl.cpp:2996:9: error: 'ImplementPositionControl2' has not been declared
2021-06-12T02:33:33.3139157Z  2996 |         ImplementPositionControl2::uninitialize();
2021-06-12T02:33:33.3139861Z       |         ^~~~~~~~~~~~~~~~~~~~~~~~~
2021-06-12T02:33:33.3141687Z /__w/robotology-superbuild/robotology-superbuild/src/ICUB/src/libraries/icubmod/canBusMotionControl/CanBusMotionControl.cpp:2997:9: error: 'ImplementVelocityControl2' has not been declared
2021-06-12T02:33:33.3143523Z  2997 |         ImplementVelocityControl2::uninitialize();
2021-06-12T02:33:33.3144220Z       |         ^~~~~~~~~~~~~~~~~~~~~~~~~
2021-06-12T02:33:33.3186677Z /__w/robotology-superbuild/robotology-superbuild/src/ICUB/src/libraries/icubmod/canBusMotionControl/CanBusMotionControl.cpp:3009:9: error: 'ImplementControlMode2' has not been declared
2021-06-12T02:33:33.3188434Z  3009 |         ImplementControlMode2::uninitialize();
2021-06-12T02:33:33.3189055Z       |         ^~~~~~~~~~~~~~~~~~~~~
2021-06-12T02:33:33.3191368Z /__w/robotology-superbuild/robotology-superbuild/src/ICUB/src/libraries/icubmod/canBusMotionControl/CanBusMotionControl.cpp: In member function 'unsigned char yarp::dev::CanBusMotionControl::from_modevocab_to_modeint(int)':
2021-06-12T02:33:33.3194919Z /__w/robotology-superbuild/robotology-superbuild/src/ICUB/src/libraries/icubmod/canBusMotionControl/CanBusMotionControl.cpp:3934:17: warning: unsigned conversion from 'yarp::conf::vocab32_t' {aka 'int'} to 'unsigned char' changes value from '2003529333' to '117' [-Woverflow]
2021-06-12T02:33:33.3196604Z  3934 |         return  VOCAB_CM_UNKNOWN;
2021-06-12T02:33:33.3197080Z       |                 ^~~~~~~~~~~~~~~~
2021-06-12T02:33:33.3199424Z /__w/robotology-superbuild/robotology-superbuild/src/ICUB/src/libraries/icubmod/canBusMotionControl/CanBusMotionControl.cpp:3943:16: warning: unsigned conversion from 'yarp::conf::vocab32_t' {aka 'int'} to 'unsigned char' changes value from '2003529333' to '117' [-Woverflow]
2021-06-12T02:33:33.3201100Z  3943 |         return VOCAB_CM_UNKNOWN;
2021-06-12T02:33:33.3201561Z       |                ^~~~~~~~~~~~~~~~
2021-06-12T02:33:33.3203817Z /__w/robotology-superbuild/robotology-superbuild/src/ICUB/src/libraries/icubmod/canBusMotionControl/CanBusMotionControl.cpp: In member function 'virtual void yarp::dev::CanBusMotionControl::run()':
2021-06-12T02:33:33.3206881Z /__w/robotology-superbuild/robotology-superbuild/src/ICUB/src/libraries/icubmod/canBusMotionControl/CanBusMotionControl.cpp:3614:49: warning: '%s' directive writing up to 254 bytes into a region of size between 1 and 255 [-Wformat-overflow=]
2021-06-12T02:33:33.3208598Z  3614 |                             sprintf(message, "%s%s", message, tmp);
2021-06-12T02:33:33.3209152Z       |                                                 ^~            ~~~
2021-06-12T02:33:33.3210975Z /__w/robotology-superbuild/robotology-superbuild/src/ICUB/src/libraries/icubmod/canBusMotionControl/CanBusMotionControl.cpp:3614:36: note: 'sprintf' output between 1 and 509 bytes into a destination of size 255
2021-06-12T02:33:33.3212559Z  3614 |                             sprintf(message, "%s%s", message, tmp);
2021-06-12T02:33:33.3213198Z       |                             ~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[... cut ...] destination of size 255
2021-06-12T02:33:33.3220813Z  3628 |                     sprintf(message, "%s%s", message, tmp);
~~~

At first I tought this was an intentional breaking change in YARP 3.5 that still needed to be  accounted for in icub-main, but after a quick inspection the only possible relevant PR is https://github.com/robotology/yarp/pull/2596, and that does not seems to break anything. I wonder if there was still some deprecated include in `canBusMotionControl` that we can easily fix.

traversaro, 2021-06-12:
This two PRs should solve the problem:
* https://github.com/robotology/yarp/pull/2605
* https://github.com/robotology/icub-main/pull/755

The first one is a regression of https://github.com/robotology/yarp/pull/2596, while  https://github.com/robotology/icub-main/pull/755 is a leftover problem that was still present in icub-main that  https://github.com/robotology/yarp/pull/2596 only make clear.

pattacini, 2021-06-12:
I've spotted build errors also for what regards iCub_SIM

<details>
<summary></summary>

```
C:\dev\icub-main\src\simulators\iCubSimulation\wrapper\iCubSimulationControl.h(81,36): error C2504: 'ImplementPositionControl': base class undefined [C:\dev\icub-main\build\src\simulators\iCubSimulation\iCub_SIM.vcxproj]
C:\dev\icub-main\src\simulators\iCubSimulation\wrapper\iCubSimulationControl.h(83,36): error C2504: 'ImplementVelocityControl': base class undefined [C:\dev\icub-main\build\src\simulators\iCubSimulation\iCub_SIM.vcxproj]
C:\dev\icub-main\src\simulators\iCubSimulation\wrapper\iCubSimulationControl.h(81,36): error C2504: 'ImplementPositionControl': base class undefined [C:\dev\icub-main\build\src\simulators\iCubSimulation\iCub_SIM.vcxproj]
C:\dev\icub-main\src\simulators\iCubSimulation\wrapper\iCubSimulationControl.h(83,36): error C2504: 'ImplementVelocityControl': base class undefined [C:\dev\icub-main\build\src\simulators\iCubSimulation\iCub_SIM.vcxproj]
C:\dev\icub-main\src\simulators\iCubSimulation\wrapper\iCubSimulationControl.cpp(57,5): error C2614: 'yarp::dev::iCubSimulationControl': illegal member initialization: 'ImplementPositionControl' is not a base or member [C:\dev\icub-main\build\src\simulators\iCubSimulation\iCub_SIM.vcxproj]
C:\dev\icub-main\src\simulators\iCubSimulation\wrapper\iCubSimulationControl.cpp(58,5): error C2614: 'yarp::dev::iCubSimulationControl': illegal member initialization: 'ImplementVelocityControl' is not a base or member [C:\dev\icub-main\build\src\simulators\iCubSimulation\iCub_SIM.vcxproj]
C:\dev\icub-main\src\simulators\iCubSimulation\wrapper\iCubSimulationControl.cpp(306,5): error C2653: 'ImplementPositionControl2': is not a class or namespace name [C:\dev\icub-main\build\src\simulators\iCubSimulation\iCub_SIM.vcxproj]
C:\dev\icub-main\src\simulators\iCubSimulation\wrapper\iCubSimulationControl.cpp(306,42): error C2385: ambiguous access of 'initialize' [C:\dev\icub-main\build\src\simulators\iCubSimulation\iCub_SIM.vcxproj]
C:\dev\icub-main\src\simulators\iCubSimulation\wrapper\iCubSimulationControl.cpp(306,42): message : could be the 'initialize' in base 'yarp::dev::ImplementTorqueControl' [C:\dev\icub-main\build\src\simulators\iCubSimulation\iCub_SIM.vcxproj]
C:\dev\icub-main\src\simulators\iCubSimulation\wrapper\iCubSimulationControl.cpp(306,42): message : or could be the 'initialize' in base 'yarp::dev::ImplementAmplifierControl' [C:\dev\icub-main\build\src\simulators\iCubSimulation\iCub_SIM.vcxproj]
C:\dev\icub-main\src\simulators\iCubSimulation\wrapper\iCubSimulationControl.cpp(306,42): message : or could be the 'initialize' in base 'yarp::dev::ImplementControlCalibration' [C:\dev\icub-main\build\src\simulators\iCubSimulation\iCub_SIM.vcxproj]
C:\dev\icub-main\src\simulators\iCubSimulation\wrapper\iCubSimulationControl.cpp(306,42): message : or could be the 'initialize' in base 'yarp::dev::ImplementControlLimits' [C:\dev\icub-main\build\src\simulators\iCubSimulation\iCub_SIM.vcxproj]
C:\dev\icub-main\src\simulators\iCubSimulation\wrapper\iCubSimulationControl.cpp(306,42): message : or could be the 'initialize' in base 'yarp::dev::ImplementControlMode' [C:\dev\icub-main\build\src\simulators\iCubSimulation\iCub_SIM.vcxproj]
C:\dev\icub-main\src\simulators\iCubSimulation\wrapper\iCubSimulationControl.cpp(306,42): message : or could be the 'initialize' in base 'yarp::dev::ImplementInteractionMode' [C:\dev\icub-main\build\src\simulators\iCubSimulation\iCub_SIM.vcxproj]
C:\dev\icub-main\src\simulators\iCubSimulation\wrapper\iCubSimulationControl.cpp(306,42): message : or could be the 'initialize' in base 'yarp::dev::ImplementPidControl' [C:\dev\icub-main\build\src\simulators\iCubSimulation\iCub_SIM.vcxproj]
C:\dev\icub-main\src\simulators\iCubSimulation\wrapper\iCubSimulationControl.cpp(306,42): message : or could be the 'initialize' in base 'yarp::dev::ImplementEncodersTimed' [C:\dev\icub-main\build\src\simulators\iCubSimulation\iCub_SIM.vcxproj]
C:\dev\icub-main\src\simulators\iCubSimulation\wrapper\iCubSimulationControl.cpp(306,42): message : or could be the 'initialize' in base 'yarp::dev::ImplementMotorEncoders' [C:\dev\icub-main\build\src\simulators\iCubSimulation\iCub_SIM.vcxproj]
C:\dev\icub-main\src\simulators\iCubSimulation\wrapper\iCubSimulationControl.cpp(306,42): message : or could be the 'initialize' in base 'yarp::dev::ImplementMotor' [C:\dev\icub-main\build\src\simulators\iCubSimulation\iCub_SIM.vcxproj]
C:\dev\icub-main\src\simulators\iCubSimulation\wrapper\iCubSimulationControl.cpp(306,42): message : or could be the 'initialize' in base 'yarp::dev::ImplementPositionDirect' [C:\dev\icub-main\build\src\simulators\iCubSimulation\iCub_SIM.vcxproj]
C:\dev\icub-main\src\simulators\iCubSimulation\wrapper\iCubSimulationControl.cpp(306,42): message : or could be the 'initialize' in base 'yarp::dev::ImplementRemoteVariables' [C:\dev\icub-main\build\src\simulators\iCubSimulation\iCub_SIM.vcxproj]
C:\dev\icub-main\src\simulators\iCubSimulation\wrapper\iCubSimulationControl.cpp(306,42): message : or could be the 'initialize' in base 'yarp::dev::ImplementAxisInfo' [C:\dev\icub-main\build\src\simulators\iCubSimulation\iCub_SIM.vcxproj]
C:\dev\icub-main\src\simulators\iCubSimulation\wrapper\iCubSimulationControl.cpp(306,42): message : or could be the 'initialize' in base 'yarp::dev::ImplementPWMControl' [C:\dev\icub-main\build\src\simulators\iCubSimulation\iCub_SIM.vcxproj]
[... cut ...]
C:\dev\icub-main\src\simulators\iCubSimulation\wrapper\iCubSimulationControl.h(83,36): error C2504: 'ImplementVelocityControl': base class undefined [C:\dev\icub-main\build\src\simulators\iCubSimulation\iCub_SIM.vcxproj]
C:\dev\icub-main\src\simulators\iCubSimulation\wrapper\iCubSimulationControl.h(81,36): error C2504: 'ImplementPositionControl': base class undefined [C:\dev\icub-main\build\src\simulators\iCubSimulation\iCub_SIM.vcxproj]
C:\dev\icub-main\src\simulators\iCubSimulation\wrapper\iCubSimulationControl.h(83,36): error C2504: 'ImplementVelocityControl': base class undefined [C:\dev\icub-main\build\src\simulators\iCubSimulation\iCub_SIM.vcxproj]

```

</details>

traversaro, 2021-06-12:
> I've spotted build errors also for what regards iCub_SIM:
> 
> 🔘 Click to show the log

Those should be fixed by https://github.com/robotology/yarp/pull/2605 .

traversaro, 2021-06-21:
> 
> 
> This two PRs should solve the problem:
> 
>     * [robotology/yarp#2605](https://github.com/robotology/yarp/pull/2605)
> 
>     * [robotology/icub-main#755](https://github.com/robotology/icub-main/pull/755)
> 
> 
> The first one is a regression of [robotology/yarp#2596](https://github.com/robotology/yarp/pull/2596), while [robotology/icub-main#755](https://github.com/robotology/icub-main/pull/755) is a leftover problem that was still present in icub-main that [robotology/yarp#2596](https://github.com/robotology/yarp/pull/2596) only make clear.

These two PR were merged.

closed 2021-06-21 by traversaro