robotology/robotology-superbuild #799: Enable manifpy compilation when ROBOTOLOGY_USES_PYTHON is enabled

traversaro, 2021-06-15:


traversaro, 2021-06-15:
Sister issue: https://github.com/conda-forge/manif-feedstock/issues/3 .

diegoferigo, 2021-06-15:
I'm not really an expert of packaging for conda. If I understood correctly, instead of getting files from the CMake install tree as we do for other projects, we have to pass through setuptools that will install the Python files in the enabled site-packages, and then package this folder with conda. Correct?

If this is ok in the conda-side, it seems to be the easiest solution at the moment. Note that I recently opened https://github.com/artivis/manif/issues/228 that, depending if and how it will get implemented, could involve a new CMake machinery similar to what we are are used to have in our hybrid projects.

traversaro, 2021-06-15:
To clarify, the issue on manifpy packaging in conda is https://github.com/conda-forge/manif-feedstock/issues/3 . This one is to install manifpy as part of the robotology-superbuild.

traversaro, 2021-06-16:
Calling a setup.py from the [`Buildmanif.cmake` file](https://github.com/robotology/robotology-superbuild/blob/master/cmake/Buildmanif.cmake) or even creating a `Buildmanifpy.cmake`  seems non obvious, as we would need to use features of `ycm_ep_helper` that I do not really know if they have been tested a lot in the past. Probably just modifying `manifpy`  to support installing its Python bindings is the best option.

traversaro, 2021-06-16:
> Maybe we can wait [artivis/manif#233](https://github.com/artivis/manif/pull/233)? However, this means that we cannot support the newly tagged release.

Cool, I did not saw that! This is not a big problem, if the PR is not merged soon I can quickly create a internal release on a fork and use that one, the proper release was just necessary for conda-forge.

closed 2021-06-25 by traversaro