robotology/robotology-superbuild #811: Problem in activating environment in Powershell in Windows when powershell profile is stored in OneDrive

traversaro, 2021-06-29:
@ste93 is having problems in activating arbitrary conda environment in powershell, while in command prompt everything is working. In particular `conda activate <nameenv>`  does not seem to do anything.

This is the output of his `conda init`:
~~~
PS C:\Users\sbernagozzi> conda init
no change     C:\Users\sbernagozzi\AppData\Local\miniforge3\Scripts\conda.exe
no change     C:\Users\sbernagozzi\AppData\Local\miniforge3\Scripts\conda-env.exe
no change     C:\Users\sbernagozzi\AppData\Local\miniforge3\Scripts\conda-script.py
no change     C:\Users\sbernagozzi\AppData\Local\miniforge3\Scripts\conda-env-script.py
no change     C:\Users\sbernagozzi\AppData\Local\miniforge3\condabin\conda.bat
no change     C:\Users\sbernagozzi\AppData\Local\miniforge3\Library\bin\conda.bat
no change     C:\Users\sbernagozzi\AppData\Local\miniforge3\condabin\_conda_activate.bat
no change     C:\Users\sbernagozzi\AppData\Local\miniforge3\condabin\rename_tmp.bat
no change     C:\Users\sbernagozzi\AppData\Local\miniforge3\condabin\conda_auto_activate.bat
no change     C:\Users\sbernagozzi\AppData\Local\miniforge3\condabin\conda_hook.bat
no change     C:\Users\sbernagozzi\AppData\Local\miniforge3\Scripts\activate.bat
no change     C:\Users\sbernagozzi\AppData\Local\miniforge3\condabin\activate.bat
no change     C:\Users\sbernagozzi\AppData\Local\miniforge3\condabin\deactivate.bat
no change     C:\Users\sbernagozzi\AppData\Local\miniforge3\Scripts\activate
no change     C:\Users\sbernagozzi\AppData\Local\miniforge3\Scripts\deactivate
no change     C:\Users\sbernagozzi\AppData\Local\miniforge3\etc\profile.d\conda.sh
no change     C:\Users\sbernagozzi\AppData\Local\miniforge3\etc\fish\conf.d\conda.fish
no change     C:\Users\sbernagozzi\AppData\Local\miniforge3\shell\condabin\Conda.psm1
no change     C:\Users\sbernagozzi\AppData\Local\miniforge3\shell\condabin\conda-hook.ps1
no change     C:\Users\sbernagozzi\AppData\Local\miniforge3\Lib\site-packages\xontrib\conda.xsh
no change     C:\Users\sbernagozzi\AppData\Local\miniforge3\etc\profile.d\conda.csh
modified      C:\Users\sbernagozzi\OneDrive - Fondazione Istituto Italiano Tecnologia\Documents\WindowsPowerShell\profile.ps1
modified      HKEY_CURRENT_USER\Software\Microsoft\Command Processor\AutoRun
~~~

While in my case is:
~~~
(new) PS C:\Users\STraversaro> conda init
no change     C:\Users\STraversaro\AppData\Local\miniforge3\Scripts\conda.exe
no change     C:\Users\STraversaro\AppData\Local\miniforge3\Scripts\conda-env.exe
no change     C:\Users\STraversaro\AppData\Local\miniforge3\Scripts\conda-script.py
no change     C:\Users\STraversaro\AppData\Local\miniforge3\Scripts\conda-env-script.py
no change     C:\Users\STraversaro\AppData\Local\miniforge3\condabin\conda.bat
no change     C:\Users\STraversaro\AppData\Local\miniforge3\Library\bin\conda.bat
no change     C:\Users\STraversaro\AppData\Local\miniforge3\condabin\_conda_activate.bat
no change     C:\Users\STraversaro\AppData\Local\miniforge3\condabin\rename_tmp.bat
no change     C:\Users\STraversaro\AppData\Local\miniforge3\condabin\conda_auto_activate.bat
no change     C:\Users\STraversaro\AppData\Local\miniforge3\condabin\conda_hook.bat
no change     C:\Users\STraversaro\AppData\Local\miniforge3\Scripts\activate.bat
no change     C:\Users\STraversaro\AppData\Local\miniforge3\condabin\activate.bat
no change     C:\Users\STraversaro\AppData\Local\miniforge3\condabin\deactivate.bat
no change     C:\Users\STraversaro\AppData\Local\miniforge3\Scripts\activate
no change     C:\Users\STraversaro\AppData\Local\miniforge3\Scripts\deactivate
no change     C:\Users\STraversaro\AppData\Local\miniforge3\etc\profile.d\conda.sh
no change     C:\Users\STraversaro\AppData\Local\miniforge3\etc\fish\conf.d\conda.fish
no change     C:\Users\STraversaro\AppData\Local\miniforge3\shell\condabin\Conda.psm1
no change     C:\Users\STraversaro\AppData\Local\miniforge3\shell\condabin\conda-hook.ps1
no change     C:\Users\STraversaro\AppData\Local\miniforge3\Lib\site-packages\xontrib\conda.xsh
no change     C:\Users\STraversaro\AppData\Local\miniforge3\etc\profile.d\conda.csh
modified      C:\Users\STraversaro\Documents\WindowsPowerShell\profile.ps1
modified      HKEY_CURRENT_USER\Software\Microsoft\Command Processor\AutoRun
~~~

We do not know if this is actually the problem, but the biggest difference seems that in my case my `profile.ps1` is `C:\Users\STraversaro\Documents\WindowsPowerShell\profile.ps1`, while in @ste93 it is `C:\Users\sbernagozzi\OneDrive - Fondazione Istituto Italiano Tecnologia\Documents\WindowsPowerShell\profile.ps1`, so it is possible that the difference is related to that.

traversaro, 2021-09-09:
I installed the new computer and I also selected the option to have the `Documents` folder on OneDrive, so I may soon have the same problem.

traversaro, 2021-09-09:
By the way, on a new laptop that I configure to have Documented in `C:\Users\STraversaro\OneDrive - Fondazione Istituto Italiano
Tecnologia\Documents\`, if I start PowerShell I have the error:
~~~
Windows PowerShell
Copyright (C) Microsoft Corporation. All rights reserved.

Try the new cross-platform PowerShell https://aka.ms/pscore6

. : File C:\Users\STraversaro\OneDrive - Fondazione Istituto Italiano
Tecnologia\Documents\WindowsPowerShell\profile.ps1 cannot be loaded because running scripts is disabled on this
system. For more information, see about_Execution_Policies at https:/go.microsoft.com/fwlink/?LinkID=135170.
At line:1 char:3
+ . 'C:\Users\STraversaro\OneDrive - Fondazione Istituto Italiano Tecno ...
+   ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : SecurityError: (:) [], PSSecurityException
    + FullyQualifiedErrorId : UnauthorizedAccess
PS C:\Users\STraversaro>
~~~
see:
[image]

Indeed, if I inspect the file `C:\Users\STraversaro\OneDrive - Fondazione Istituto Italiano
Tecnologia\Documents\WindowsPowerShell\profile.ps1` it just contains:
~~~

#region conda initialize
# !! Contents within this block are managed by 'conda init' !!
(& "C:\Users\STraversaro\AppData\Local\mambaforge\Scripts\conda.exe" "shell.powershell" "hook") | Out-String | Invoke-Expression
#endregion
~~~
so it is possible that file is not present before, and it is created by the `conda init` command.

traversaro, 2021-09-09:
@ste93 do you have the same error?

traversaro, 2021-09-09:
I briefly checked, and indeed the file `C:\Users\STraversaro\OneDrive - Fondazione Istituto Italiano Tecnologia\Documents\WindowsPowerShell\profile.ps1` is created when running `conda init` in the command prompt, even if it was not there before. 
Interestingly, just running `conda init --reverse` as per uninstall instructions (see https://github.com/robotology/robotology-superbuild/blob/0c4c5ff792c0e2f376091565d27740d0760a28dc/doc/install-mambaforge.md#uninstall-2) does not fix the problem, as it just make  `C:\Users\STraversaro\OneDrive - Fondazione Istituto Italiano Tecnologia\Documents\WindowsPowerShell\profile.ps1` an empty files, but it does not remove it even if before `conda init` the file was not there.

ste93, 2021-09-10:
sorry I haven't used windows in a while. I see the same error, I think maybe is disabled launching scripts from one drive, I can check. I think this solution might work, it seems that you must sign the scripts in some way or you must authorize the system to execute scripts from one drive http://vcloud-lab.com/entries/powershell/different-ways-to-bypass-powershell-execution-policy-ps1-cannot-be-loaded-because-running-scripts-is-disabled

traversaro, 2021-09-10:
> sorry I haven't used windows in a while. I see the same error, I think maybe is disabled launching scripts from one drive, I can check. I think this solution might work, it seems that you must sign [...]

Thanks, so we know that also your issue is indeed that one. For the future, remember to include any strange errors that appear in the bug report, thanks!

ste93, 2021-09-10:
Yes, I haven't noticed it before, my fault. I've noticed it now

traversaro, 2021-09-10:
By the way, as far as I understand it is related to the default execution policy in PowerShell, not wheter the profile is on OneDrive or not. In my system, everything works fine if I run in a Administrator PowerShell shell:
~~~
Set-ExecutionPolicy -ExecutionPolicy RemoteSigned
~~~

`RemoteSigned` seems to be a good compromise, even because it is the default on Windows Server.

Related issues: 
* https://stackoverflow.com/questions/62082873/conda-not-activate-in-power-shell
* https://github.com/conda/conda/issues/8428

traversaro, 2021-09-10:
See also https://docs.microsoft.com/en-us/powershell/module/microsoft.powershell.core/about/about_profiles?view=powershell-7.1#profiles-and-execution-policy .

ste93, 2021-09-10:
ok fantastic, so might be a solution to change only the execution policy.

traversaro, 2021-09-10:
PR proposed in https://github.com/robotology/robotology-superbuild/pull/885 .

closed 2021-09-10 by traversaro