robotology/robotology-superbuild #812: Stop supporting Gazebo 9 and only support Gazebo 11?

traversaro, 2021-07-01:
The `ROBOTOLOGY_USES_GAZEBO` is currently tested against both Gazebo 9 and Gazebo 11. To simplify mantainance, I would like to stop supporting Gazebo 9, and only support Gazebo 11 at some point in the future (2021.11 perhaps?).

I would like to know if there is any blocker for this, or if there are teams that are not using Gazebo 11, but rather are still using Gazebo 9 or 10.

cc @robotology/edpr-developers @robotology/iit-dynamic-interaction-control

arrenglover, 2021-07-01:
@Iaxama what version we are using for EDPR work?

Iaxama, 2021-07-01:
> @Iaxama what version we are using for EDPR work?

We use gazebo 11

traversaro, 2021-07-01:
@Yeshasvitvs @lrapetti not sure if https://github.com/robotology/gazebo-yarp-plugins/issues/494 is still a problem.

Yeshasvitvs, 2021-07-01:
I am not certain about it as I haven't been using Gazebo for a while. May be @HosameldinMohamed can share some details ?

drdanz, 2021-07-01:
Debian testing has gazebo 11, so not a problem for me...

lrapetti, 2021-07-06:
> @Yeshasvitvs @lrapetti not sure if [robotology/gazebo-yarp-plugins#494](https://github.com/robotology/gazebo-yarp-plugins/issues/494) is still a problem.

I am not sure either since I have actually never used it. I saw @paolo-viceconte was involved in the discussion, maybe has some more recent experience?

paolo-viceconte, 2021-07-06:
> I am not sure either since I have actually never used it. I saw @paolo-viceconte was involved in the discussion, maybe has some more recent experience?

Not really. I only used the external wrench plugin for quick tests but then that activity has been dropped, so no recent updates wrt what has been documented in the issue above.

traversaro, 2021-07-12:
Example of issue that emerge with older version of Gazebo: https://github.com/ignitionrobotics/ign-math/issues/214 .

traversaro, 2021-07-16:
A major advantage of switching to Gazebo 11 is that we could ship our models using SDF 1.7, avoiding several problems we had in the past with complex SDF 1.6 models (especially SDF 1.6 models that included SDF 1.7 models) in Gazebo 11, see https://github.com/osrf/gazebo/issues/2728, https://github.com/robotology/icub-models/issues/67 and https://github.com/osrf/gazebo/issues/2739 .

traversaro, 2021-07-19:
Wider announcement done in https://github.com/robotology/community/discussions/534 .

traversaro, 2021-10-05:
Done in https://github.com/robotology/gazebo-yarp-plugins/pull/575 and https://github.com/robotology/robotology-superbuild/pull/873 .

traversaro, 2022-02-15:
Closing as this was announced in https://github.com/robotology/community/discussions/534 .

closed 2022-05-09 by traversaro