robotology/robotology-superbuild #815: icub-main iCub_SIM conda binary problems

traversaro, 2021-07-06:
If one installs on a clean environment:
~~~
conda install -c conda-forge -c robotology icub-main
~~~

and then tries to run:
~~~
iCub_SIM
~~~

the program fails with error:
~~~
(manualenv) straversaro@IITICUBLAP102:~$ iCub_SIM
iCub_SIM: error while loading shared libraries: libode.so.0.16.2: cannot open shared object file: No such file or directory
~~~

If one manually installs `libode` via `conda install libode`, then `iCub_SIM` segfaults:
~~~
(manualenv) straversaro@IITICUBLAP102:~$ iCub_SIM
[DEBUG]
ODE configuration: ODE ODE_EXT_no_debug ODE_EXT_trimesh ODE_EXT_opcode ODE_OPC_new_collider ODE_EXT_threading ODE_THR_builtin_impl ODE_double_precision

[DEBUG] SimConfig::configure
[INFO] default module name:  /icubSim
[INFO] default verbosity level:  0
[INFO] The iCub simulator will start with the following configuration:

 Elevation :  off
 startHomePos :  on
 Legs :  on
 Torso :  on
 Left arm :  on
 Left hand :  on
 Right arm :  on
 Right hand :  on
 Head :  on
 Fixed Hip :  on

 Self-collisions :  off
 Collisions for covers :  off

 Pressure sensors:  off
 Whole body skin emulation:  off
 Cameras : on
 Objects :  off
 Head Cover :  on
 Legs Cover :  on
 Left arm Covers :  on
 Right arm Covers :  on
 Torso Cover :  on
 Screen :  off

[INFO] Creating 3D Model of the icub.......
[DEBUG] Loading mesh data from '/home/straversaro/mambaforge/envs/manualenv/share/iCub/contexts/simConfig/models/leftFoot.x'
[DEBUG] ...
[DEBUG] ...
[DEBUG] ... OK!
[DEBUG] Loading texture coordinate from '/home/straversaro/mambaforge/envs/manualenv/share/iCub/contexts/simConfig/models/leftFoot.x'
[DEBUG] ...
[DEBUG] ...
[DEBUG] ... OK!
[DEBUG] Loading normals from '/home/straversaro/mambaforge/envs/manualenv/share/iCub/contexts/simConfig/models/leftFoot.x'
[DEBUG] ...
[DEBUG] ...
[DEBUG] ... OK!
[DEBUG] Loading mesh data from '/home/straversaro/mambaforge/envs/manualenv/share/iCub/contexts/simConfig/models/leftPalm.x'
[DEBUG] ...
[DEBUG] ...
[DEBUG] ... OK!
[DEBUG] Loading texture coordinate from '/home/straversaro/mambaforge/envs/manualenv/share/iCub/contexts/simConfig/models/leftPalm.x'
[... 575 lines cut ...]
       user config file : /home/straversaro/.condarc
 populated config files : /home/straversaro/mambaforge/.condarc
                          /home/straversaro/.condarc
          conda version : 4.10.3
    conda-build version : not installed
         python version : 3.9.5.final.0
       virtual packages : __linux=4.19.104=0
                          __glibc=2.31=0
                          __unix=0=0
                          __archspec=1=x86_64
       base environment : /home/straversaro/mambaforge  (writable)
      conda av data dir : /home/straversaro/mambaforge/etc/conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/conda-forge/linux-64
                          https://conda.anaconda.org/conda-forge/noarch
          package cache : /home/straversaro/mambaforge/pkgs
                          /home/straversaro/.conda/pkgs
       envs directories : /home/straversaro/mambaforge/envs
                          /home/straversaro/.conda/envs
               platform : linux-64
             user-agent : conda/4.10.3 requests/2.25.1 CPython/3.9.5 Linux/4.19.104-microsoft-standard ubuntu/20.04.2 glibc/2.31
                UID:GID : 1000:1000
             netrc file : None
           offline mode : False
~~~

traversaro, 2021-07-06:
Both issue were reported by @Nicogene and @GrmanRodriguez, but I was able to reproduce both of them on my WSL2 Ubuntu 20.04 system.

traversaro, 2021-07-06:
A quick run under valgrind reveals: 
~~~
[DEBUG] ***** OdeSdlSimulation::simLoop
==18137== Thread 2:
==18137== Invalid read of size 8
==18137==    at 0x487C82A: SDL_WM_SetIcon (in /home/straversaro/mambaforge/envs/manualenv/lib/libSDL-1.2.so.0.11.4)
==18137==    by 0x1A5121: OdeSdlSimulation::simLoop(int, int) (in /home/straversaro/mambaforge/envs/manualenv/bin/iCub_SIM)
==18137==    by 0x146B1A: SimulatorModule::runModule() (in /home/straversaro/mambaforge/envs/manualenv/bin/iCub_SIM)
==18137==    by 0x144D18: SimulationRun::run(SimulationBundle*, int, char**) (in /home/straversaro/mambaforge/envs/manualenv/bin/iCub_SIM)
==18137==    by 0x13A4AD: MainThread::run() (in /home/straversaro/mambaforge/envs/manualenv/bin/iCub_SIM)
==18137==    by 0x4DB8144: theExecutiveBranch(void*) (in /home/straversaro/mambaforge/envs/manualenv/lib/libYARP_os.so.3.4.6)
==18137==    by 0x4EDF066: execute_native_thread_routine (thread.cc:80)
==18137==    by 0x52EF608: start_thread (pthread_create.c:477)
==18137==    by 0x5210292: clone (clone.S:95)
==18137==  Address 0x118 is not stack'd, malloc'd or (recently) free'd
==18137==
==18137==
==18137== Process terminating with default action of signal 11 (SIGSEGV)
==18137==    at 0x5134169: raise (raise.c:46)
==18137==    by 0x513420F: ??? (in /usr/lib/x86_64-linux-gnu/libc-2.31.so)
==18137==    by 0x487C829: SDL_WM_SetIcon (in /home/straversaro/mambaforge/envs/manualenv/lib/libSDL-1.2.so.0.11.4)
==18137==    by 0x1A5121: OdeSdlSimulation::simLoop(int, int) (in /home/straversaro/mambaforge/envs/manualenv/bin/iCub_SIM)
==18137==    by 0x146B1A: SimulatorModule::runModule() (in /home/straversaro/mambaforge/envs/manualenv/bin/iCub_SIM)
==18137==    by 0x144D18: SimulationRun::run(SimulationBundle*, int, char**) (in /home/straversaro/mambaforge/envs/manualenv/bin/iCub_SIM)
==18137==    by 0x13A4AD: MainThread::run() (in /home/straversaro/mambaforge/envs/manualenv/bin/iCub_SIM)
==18137==    by 0x4DB8144: theExecutiveBranch(void*) (in /home/straversaro/mambaforge/envs/manualenv/lib/libYARP_os.so.3.4.6)
==18137==    by 0x4EDF066: execute_native_thread_routine (thread.cc:80)
==18137==    by 0x52EF608: start_thread (pthread_create.c:477)
==18137==    by 0x5210292: clone (clone.S:95)
==18137==
==18137== HEAP SUMMARY:
==18137==     in use at exit: 4,807,254 bytes in 22,596 blocks
==18137==   total heap usage: 584,098 allocs, 561,502 frees, 58,530,062 bytes allocated
==18137==
==18137== LEAK SUMMARY:
==18137==    definitely lost: 0 bytes in 0 blocks
==18137==    indirectly lost: 0 bytes in 0 blocks
==18137==      possibly lost: 944,552 bytes in 74 blocks
==18137==    still reachable: 3,862,702 bytes in 22,522 blocks
==18137==                       of which reachable via heuristic:
==18137==                         newarray           : 255,576 bytes in 54 blocks
==18137==                         multipleinheritance: 2,032 bytes in 12 blocks
==18137==         suppressed: 0 bytes in 0 blocks
==18137== Rerun with --leak-check=full to see details of leaked memory
==18137==
==18137== For lists of detected and suppressed errors, rerun with: -s
==18137== ERROR SUMMARY: 1 errors from 1 contexts (suppressed: 0 from 0)
Segmentation fault
~~~

traversaro, 2021-07-07:
For what regards P1, the root issue is https://github.com/conda-forge/libode-feedstock/issues/12 (and at the moment I do not have a clear idea on how to tackle it).

pattacini, 2021-07-07:
Just to add further info on this.
I've tested `iCub_SIM` within the [`icub-gazebo-grasping-sandbox@v2.0.0`](https://github.com/robotology/icub-gazebo-grasping-sandbox/releases/tag/v2.0.0) that is built w/ `robotology-superbuild@v2021.05` and it starts off smoothly.

Is there any difference with respect to the conda installation?

traversaro, 2021-07-07:
Out of curiosity, I tried to compile icub-main from source in a conda environment, and after commenting the `SDL_WM_SetIcon` call, then the `iCub_SIM` is still segfaulting in:
~~~
[INFO] Adding video texture face
[INFO] |yarp.os.Port| Port /icubSim/texture/face active at tcp://172.29.172.176:10050/
[INFO] Adding video texture screen
[INFO] |yarp.os.Port| Port /icubSim/texture/screen active at tcp://172.29.172.176:10051/
[INFO] Adding video texture sky
[INFO] |yarp.os.Port| Port /icubSim/texture/sky active at tcp://172.29.172.176:10052/
[DEBUG] ***** OdeSdlSimulation::simLoop
logo: /home/straversaro/mambaforge/envs/icubdev/share/iCub/contexts/simConfig/data/robotcub.bmp
Parameter passed to /home/straversaro/mambaforge/envs/icubdev/share/iCub/contexts/simConfig/data/robotcub.bmp
==31655== Thread 2:
==31655== Invalid read of size 1
==31655==    at 0x513BDBB: ____strtod_l_internal (strtod_l.c:611)
==31655==    by 0x4A74967: ??? (in /home/straversaro/mambaforge/envs/icubdev2/lib/libGLU.so.1.3.1)
==31655==    by 0x4A819B0: gluBuild2DMipmaps (in /home/straversaro/mambaforge/envs/icubdev2/lib/libGLU.so.1.3.1)
==31655==    by 0x1AEA0A: LoadTextureRAW(char const*, int) (in /home/straversaro/mambaforge/envs/icubdev2/bin/iCub_SIM)
==31655==    by 0x1AEE36: setup_opengl(yarp::os::ResourceFinder&) (in /home/straversaro/mambaforge/envs/icubdev2/bin/iCub_SIM)
==31655==    by 0x1A5471: OdeSdlSimulation::simLoop(int, int) (in /home/straversaro/mambaforge/envs/icubdev2/bin/iCub_SIM)
==31655==    by 0x146B9A: SimulatorModule::runModule() (in /home/straversaro/mambaforge/envs/icubdev2/bin/iCub_SIM)
==31655==    by 0x144D98: SimulationRun::run(SimulationBundle*, int, char**) (in /home/straversaro/mambaforge/envs/icubdev2/bin/iCub_SIM)
==31655==    by 0x13A52D: MainThread::run() (in /home/straversaro/mambaforge/envs/icubdev2/bin/iCub_SIM)
==31655==    by 0x4DB6144: theExecutiveBranch(void*) (in /home/straversaro/mambaforge/envs/icubdev2/lib/libYARP_os.so.3.4.6)
==31655==    by 0x4EDB066: execute_native_thread_routine (thread.cc:80)
==31655==    by 0x52ED608: start_thread (pthread_create.c:477)
==31655==  Address 0x0 is not stack'd, malloc'd or (recently) free'd
==31655==
==31655==
==31655== Process terminating with default action of signal 11 (SIGSEGV)
==31655==    at 0x5132169: raise (raise.c:46)
==31655==    by 0x513220F: ??? (in /usr/lib/x86_64-linux-gnu/libc-2.31.so)
==31655==    by 0x513BDBA: ____strtod_l_internal (strtod_l.c:611)
==31655==    by 0x4A74967: ??? (in /home/straversaro/mambaforge/envs/icubdev2/lib/libGLU.so.1.3.1)
==31655==    by 0x4A819B0: gluBuild2DMipmaps (in /home/straversaro/mambaforge/envs/icubdev2/lib/libGLU.so.1.3.1)
==31655==    by 0x1AEA0A: LoadTextureRAW(char const*, int) (in /home/straversaro/mambaforge/envs/icubdev2/bin/iCub_SIM)
==31655==    by 0x1AEE36: setup_opengl(yarp::os::ResourceFinder&) (in /home/straversaro/mambaforge/envs/icubdev2/bin/iCub_SIM)
==31655==    by 0x1A5471: OdeSdlSimulation::simLoop(int, int) (in /home/straversaro/mambaforge/envs/icubdev2/bin/iCub_SIM)
==31655==    by 0x146B9A: SimulatorModule::runModule() (in /home/straversaro/mambaforge/envs/icubdev2/bin/iCub_SIM)
==31655==    by 0x144D98: SimulationRun::run(SimulationBundle*, int, char**) (in /home/straversaro/mambaforge/envs/icubdev2/bin/iCub_SIM)
==31655==    by 0x13A52D: MainThread::run() (in /home/straversaro/mambaforge/envs/icubdev2/bin/iCub_SIM)
==31655==    by 0x4DB6144: theExecutiveBranch(void*) (in /home/straversaro/mambaforge/envs/icubdev2/lib/libYARP_os.so.3.4.6)
~~~

While this can be probaly further investigate, another option is just to disable iCub_SIM in conda packages. Just to understand, what did you needed that was not provided by the Gazebo based simulation?

traversaro, 2021-07-07:
> Is there any difference with respect to the conda installation?

Probably there is something different in the libglu and/or sdl1 provided in conda as opposed to the one provided in apt, but this would need to be further investigated.

traversaro, 2021-07-07:
The compilation prints a warning:
~~~
[ 63%] Building CXX object src/simulators/iCubSimulation/CMakeFiles/iCub_SIM.dir/odesdl/rendering.cpp.o
In file included from /home/straversaro/icub-main/src/simulators/iCubSimulation/odesdl/rendering.cpp:32:
/home/straversaro/mambaforge/envs/icubdev2/include/SDL/SDL_opengl.h:116: warning: "GL_GLEXT_VERSION" redefined
  116 | #define GL_GLEXT_VERSION 29
      |
In file included from /home/straversaro/mambaforge/envs/icubdev2/include/GL/gl.h:2055,
                 from /home/straversaro/mambaforge/envs/icubdev2/include/SDL/SDL_opengl.h:46,
                 from /home/straversaro/icub-main/src/simulators/iCubSimulation/odesdl/rendering.cpp:32:
/home/straversaro/mambaforge/envs/icubdev2/include/GL/glext.h:54: note: this is the location of the previous definition
   54 | #define GL_GLEXT_VERSION 20180725
      |
~~~

that could be related to the failure.

traversaro, 2021-07-07:
> The compilation prints a warning:
>
> ```
> [ 63%] Building CXX object src/simulators/iCubSimulation/CMakeFiles/iCub_SIM.dir/odesdl/rendering.cpp.o
> In file included from [...]

Possible related comment: https://github.com/Wargus/stratagus/issues/241#issuecomment-469068787 .

pattacini, 2021-07-07:
>Just to understand, what did you needed that was not provided by the Gazebo based simulation?

Essentially, the skin. Also, there are tests in `icub-tests` that have not been migrated yet to the use of Gazebo (see https://github.com/robotology/icub-basic-demos/pull/32#pullrequestreview-640083769).

cc @vtikha

traversaro, 2021-07-07:
> 
> 
> > Is there any difference with respect to the conda installation?
> 
> Probably there is something different in the libglu and/or sdl1 provided in conda as opposed to the one provided in apt, but this would need to be further investigated.

To elaborate a bit more: sdl1 (differently from sdl2) saw its last release 9 years ago, and since then it did not have any new releases (intentionally, as they wanted people to migrate to sdl1), even if the sdl1.2 branch received ~350 commits since its last release (see https://github.com/libsdl-org/SDL-1.2/compare/release-1.2.15...main). That meant that basically all Linux distribution do not distribute the actual sdl 1.2.15, but rather an heavily patched version with a lot of bugfixes solved (see https://discourse.libsdl.org/t/interest-in-a-sdl-1-2-bugfix-release/24784/11 for an overview of all the patches applied on the top of SDL 1.2.15 by Linux distros, including Ubuntu).

On the other hand, the conda-forge sdl package basically has almost no patch (see https://github.com/conda-forge/sdl-feedstock/tree/master/recipe), so probably is affected by years of bugs that are actually fixed in the sdl1 version distributed by Linux distros. 

Fortunatly, a modern [shim](https://en.wikipedia.org/wiki/Shim_(computing)) of sdl1.2 that actually uses SDL2 underneat is being developed (see https://github.com/libsdl-org/sdl12-compat), and should be soon released (see https://github.com/libsdl-org/sdl12-compat/issues/53). So, before debugging the issue more, it is wiser to wait for  `sdl12-compat`  to be released, and for conda-forge to switch to use it.

pattacini, 2021-07-25:
The [test I mentioned earlier](https://github.com/robotology/robotology-superbuild/issues/815#issuecomment-875409370) has been migrated to the use of Gazebo.
See https://github.com/robotology/icub-tests/pull/58.

traversaro, 2022-03-10:
> So, before debugging the issue more, it is wiser to wait for `sdl12-compat` to be released, and for conda-forge to switch to use it.

I tested `iCub_SIM` with a the new sdl12-compat, and indeeed with that one `iCub_SIM` works fine:
[image]

So I think we can wait for merging https://github.com/conda-forge/sdl-feedstock/pull/6 and then we can close this issue.

~~~
(icubsim) traversaro@IITICUBLAP257:/mnt/c/Users/STraversaro$ mamba list
# packages in environment at /home/traversaro/mambaforge/envs/icubsim:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                       1_gnu    conda-forge
ace                       7.0.6                h9c3ff4c_0    conda-forge
alsa-lib                  1.2.3                h516909a_0    conda-forge
ampl-mp                   3.1.0             h2cc385e_1006    conda-forge
bzip2                     1.0.8                h7f98852_4    conda-forge
c-ares                    1.18.1               h7f98852_0    conda-forge
ca-certificates           2021.10.8            ha878542_0    conda-forge
cairo                     1.16.0            ha12eb4b_1010    conda-forge
dbus                      1.13.6               h5008d03_3    conda-forge
eigen                     3.4.0                h4bd325d_0    conda-forge
expat                     2.4.7                h27087fc_0    conda-forge
ffmpeg                    4.3.2                h37c90e5_3    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
font-ttf-ubuntu           0.83                 hab24e00_0    conda-forge
fontconfig                2.13.96              h8e229c2_1    conda-forge
fonts-conda-ecosystem     1                             0    conda-forge
fonts-conda-forge         1                             0    conda-forge
freeglut                  3.2.2                h9c3ff4c_1    conda-forge
freetype                  2.10.4               h0708190_1    conda-forge
gettext                   0.19.8.1          h73d1719_1008    conda-forge
gmp                       6.2.1                h58526e2_0    conda-forge
gnutls                    3.6.13               h85f3911_1    conda-forge
graphite2                 1.3.13            h58526e2_1001    conda-forge
gsl                       2.7                  he838d99_0    conda-forge
gst-plugins-base          1.18.5               hf529b03_3    conda-forge
gstreamer                 1.18.5               h9f60fe5_3    conda-forge
harfbuzz                  3.4.0                hb4a5f5f_0    conda-forge
hdf5                      1.12.1          nompi_h2386368_104    conda-forge
icu                       69.1                 h9c3ff4c_0    conda-forge
icub-firmware-shared      1.24.0              h27087fc_44    robotology
icub-main                 1.24.0              h0c40f0e_44    robotology
ipopt                     3.14.4               h7ede334_0    conda-forge
jasper                    2.0.33               ha77e612_0    conda-forge
jbig                      2.1               h7f98852_2003    conda-forge
jpeg                      9e                   h7f98852_0    conda-forge
keyutils                  1.6.1                h166bdaf_0    conda-forge
krb5                      1.19.2               h3790be6_4    conda-forge
lame                      3.100             h7f98852_1001    conda-forge
ld_impl_linux-64          2.36.1               hea4e1c9_2    conda-forge
lerc                      3.0                  h9c3ff4c_0    conda-forge
libblas                   3.9.0           13_linux64_openblas    conda-forge
libcblas                  3.9.0           13_linux64_openblas    conda-forge
libclang                  13.0.1          default_hc23dcda_0    conda-forge
libcurl                   7.82.0               h7bff187_0    conda-forge
libdc1394                 2.2.6                he1b5a44_0    conda-forge
libdeflate                1.10                 h7f98852_0    conda-forge
libedit                   3.1.20191231         he28a2e2_2    conda-forge
libev                     4.33                 h516909a_1    conda-forge
libevent                  2.1.10               h9b69904_4    conda-forge
libffi                    3.4.2                h7f98852_5    conda-forge
libgcc-ng                 11.2.0              h1d223b6_13    conda-forge
libgfortran-ng            11.2.0              h69a702a_13    conda-forge
libgfortran5              11.2.0              h5c6108e_13    conda-forge
libglib                   2.70.2               h174f98d_4    conda-forge
libglu                    9.0.0             he1b5a44_1001    conda-forge
[... 56 lines cut ...]
tzdata                    2021e                he74cb21_0    conda-forge
unixodbc                  2.3.9                hb166930_0    conda-forge
wheel                     0.37.1             pyhd8ed1ab_0    conda-forge
x264                      1!161.3030           h7f98852_1    conda-forge
xorg-fixesproto           5.0               h7f98852_1002    conda-forge
xorg-inputproto           2.3.2             h7f98852_1002    conda-forge
xorg-kbproto              1.0.7             h7f98852_1002    conda-forge
xorg-libice               1.0.10               h7f98852_0    conda-forge
[... cut ...]
xorg-xextproto            7.3.0             h7f98852_1002    conda-forge
xorg-xproto               7.0.31            h7f98852_1007    conda-forge
xz                        5.2.5                h516909a_1    conda-forge
yarp                      3.6.0           py39hd74ac5e_44    robotology
ycm-cmake-modules         0.13.0               h9c3ff4c_0    conda-forge
zlib                      1.2.11            h36c2ea0_1013    conda-forge
zstd                      1.5.2                ha95c52a_0    conda-forge
~~~

traversaro, 2022-03-11:
Fixed by https://github.com/conda-forge/sdl-feedstock/pull/6 .

closed 2022-03-11 by traversaro