robotology/robotology-superbuild #819: CMake fails when using YCM master branch

S-Dafarra, 2021-07-08:
Steps to reproduce:

- ``git clone https://github.com/robotology/robotology-superbuild``
- ``cd robotology-superbuild/ && mkdir build && cd build/ && cmake ../``
- ``cd ../src/YCM/``
- ``git checkout master``
- ``cd ../../build``
- ``cmake .``

Then, I get the following error
```
-- YCM not found. Bootstrapping it.
-- Downloading YCMEPHelper.cmake - Attempt 1 of 3
-- Downloading YCMEPHelper.cmake - SUCCESS
-- Downloading StyleGITHUB.cmake - Attempt 1 of 3
-- Downloading StyleGITHUB.cmake - SUCCESS
-- Performing download step (git clone) for 'YCM'
-- Performing update step for 'YCM'
CMake Error at build/CMakeFiles/YCMEPHelper.cmake:1377 (message):
  Cannot update YCM repository
Call Stack (most recent call first):
  cmake/YCMBootstrap.cmake:114 (ycm_bootstrap)
  CMakeLists.txt:56 (include)

-- Configuring incomplete, errors occurred!
See also "/home/sdafarra/Software/temp/robotology-superbuild/build/CMakeFiles/CMakeOutput.log".
See also "/home/sdafarra/Software/temp/robotology-superbuild/build/CMakeFiles/CMakeError.log".
```

traversaro, 2021-08-31:
I can't reproduce the issue, I guess it was some kind of temporary problem on YCM master. I will close this issue, but feel free re-open if the problem persists.

closed 2021-08-31 by traversaro

traversaro, 2022-07-18:
@mebbaid Can you please open a new issue? `Cannot build YCM` is quite the generic error, so it is possible that your error is completly unrelated from the original one of @S-Dafarra, and in general it is tricky to work on closed issues, because you do not know when the "new reply" is addressed or not.

Anyhow, two suggestions:

#### First Suggestion
You can configure cmake with `-DYCM_BOOTSTRAP_VERBOSE:BOOL=ON` to get a more verbose error on what is going on.

#### Second Suggestion

Unfortunatly the YCM bootstrap was intrinsically quite fragile, especially before the improvements in https://github.com/robotology/ycm/pull/403 and https://github.com/robotology/ycm/pull/402 . However, you can almost the same behaviour by manually compiling the exact version of YCM before compiling the superbuild (that is 0.13.0, see https://icub-tech-iit.github.io/documentation/sw_versioning_table/2021.08.1/), so that the superbuild will find the installed YCM:
~~~
git clone -b v0.13.0 https://github.com/robotology/ycm
cd ycm
mkdir -p build && cd build && cmake -GNinja .. & ninja install
~~~
this should either work, or at least provide a more easy error to deal with.

mebbaid, 2022-07-19:
Indeed that was the issue. Thanks for the help !!