robotology/robotology-superbuild #828: Move to YARP 3.5 for Stable branches and LatestReleases

traversaro, 2021-07-15:
We should understand when it will be desirable for people working on the superbuild and on the iCub robot to update the Stable branch (https://github.com/robotology/robotology-superbuild/blob/master/cmake/ProjectsTagsStable.cmake#L17) to yarp-3.5 and the latest release to v3.5.0 . 

Given the amount of changes, a fair question is: do you want to have v3.5 in 2021.08 or stick to v3.4 ? @pattacini @Nicogene

pattacini, 2021-07-15:
>Given the amount of changes, a fair question is: do you want to have v3.5 in 2021.08 or stick to v3.4 ? @pattacini @Nicogene

Holidays are approaching and we still have other tasks underway; I would be thus in favor of postponing the adoption of YARP 3.5.0 since we need to carry out a thorough verification on the robot.

pattacini, 2021-07-15:
>we need to carry out a thorough verification on the robot.

Will get aligned w/ @randaz81 soon on this particular argument.

Nicogene, 2021-07-15:
> Holidays are approaching and we still have other tasks underway; I would be thus in favor of postponing the adoption of YARP 3.5.0 since we need to carry out a thorough verification on the robot.

I agree, the august release is quite difficult to handle for the holidays, I would postpone to the 2021.11 too

traversaro, 2021-07-16:
Just FYI, from the compilation point of view everything works fine: https://github.com/robotology/robotology-superbuild/pull/830 .

Nicogene, 2021-07-19:
> Just FYI, from the compilation point of view everything works fine: #830 .

Great! It is just a matter of handle the deprecation warnings then

pattacini, 2021-07-19:
The situation can be more subtle than this because of https://github.com/robotology/yarp/pull/2491, which induces heavy changes to the XML files of the robots. That's right why @randaz81 wanted to be in contact with us.

We need to get a clear picture of the context.

drdanz, 2021-07-27:
> The situation can be more subtle than this because of [robotology/yarp#2491](https://github.com/robotology/yarp/pull/2491), which induces heavy changes to the XML files of the robots.

Just for reference, the XML files of the robot do not need to be modified, the old devices still work (with just a warning when "open" is called).

traversaro, 2021-09-01:
@pattacini @Nicogene Now that 2021.08 was release, we may want to move forward on this as soon as possible to ensure that we have all the time to detect issues before 2021.11 . Do you prefer to first to a basic run on a robot and after move stable branches  to yarp-3.5, or just move yarp-3.5 to stable and see if issue emerge?

pattacini, 2021-09-01:
I think that yarp-3.5 can be already moved to stable in the superbuild 👍🏻 
For the rest, I'd like to see the behaviors of the new components on the robot, for which @randaz81 is already working.
The idea is that once he'll be done with the test on R1SN003, then we'll agree on test sessions on iCub.

traversaro, 2021-09-01:
> I think that yarp-3.5 can be already moved to stable in the superbuild 👍🏻

Great, thanks!

Nicogene, 2021-09-01:
Yes we can move to yarp-3.5 as stable branch from my side!

traversaro, 2021-09-17:
Done in https://github.com/robotology/robotology-superbuild/pull/870 for stable branches, and in https://github.com/robotology/robotology-superbuild/pull/880 for latest releases. I think we can close know given that we even have the conda binary: https://anaconda.org/robotology/yarp/files .

closed 2021-09-17 by traversaro