robotology/robotology-superbuild #842: Stop supporting Homebrew in robotology-superbuild Continuous Integration and direct macOS users to conda?

traversaro, 2021-07-30:
Homebrew CI is extremely fragile, for several reasons. The main one, is that there is no easy way in Homebrew to be able to install the same package you installed a few days before, basically making really difficult to achieve reproducibility, that is quite critical in scientific research. 

To give an idea, this are just the issue related to Homebrew CI failures just in the last few months: 
* https://github.com/robotology/robotology-superbuild/issues/826
* https://github.com/robotology/robotology-superbuild/issues/789
* https://github.com/robotology/robotology-superbuild/issues/704
* https://github.com/robotology/robotology-superbuild/issues/693
* https://github.com/robotology/robotology-superbuild/issues/571
* https://github.com/robotology/robotology-superbuild/issues/568
* https://github.com/robotology/robotology-superbuild/issues/565
* https://github.com/robotology/robotology-superbuild/issues/490
* https://github.com/robotology/robotology-superbuild/issues/476
* https://github.com/robotology/robotology-superbuild/issues/440

Continuously dealing with these CI failures is quite demanding. At the same time, during the last year I think that `conda-forge` and the related channels (`robostack` and `robotology`) reached a level of maturity for which they are able to provide to macOS users easily installable and reproducible binaries. 

At this point I think a reasonable plan for the future, unless someone step in to volunteer mantainance of the Homebrew CI, is to **stop supporting the use of Homebrew in the robotology-superbuild CI and docs**, and direct macOS users to use conda instead. 

Ideally we would like all regular robotology-superbuild macOS user to be proficient with conda before that day, but I guess that unless there is a reason for doing so, people will not go out of their Homebrew-using comfort zone, and I know this process take time, so better to start off.

As usual for this issue, any kind of feedback is extremely welcome.

pattacini, 2021-07-30:
I'm in favor 👍🏻

lrapetti, 2021-08-04:
As MacOS/homebrew user, I think that if it can make more reproducible the installation robotology dependency w.r.t. homebrew, I have no problem in moving to conda after testing it.

traversaro, 2023-04-07:
After that the regression in GitHub Actions/Homebrew always continue, I decided. The Homebrew CI will be **removed** from the robotology-superbuild on the 15th of May, 2023.

lrapetti, 2023-04-07:
> As MacOS/homebrew user, I think that if it can make more reproducible the installation robotology dependency w.r.t. homebrew, I have no problem in moving to conda after testing it.

As a follow-up of this comment, I switched to `conda` dependencies since a while and I have never encountered issues.

traversaro, 2023-05-12:
> After that the regression in GitHub Actions/Homebrew always continue, I decided. The Homebrew CI will be **removed** from the robotology-superbuild on the 15th of May, 2023.

PR that will be merged on Monday: https://github.com/robotology/robotology-superbuild/pull/1407 .

traversaro, 2023-05-17:
Fixed by https://github.com/robotology/robotology-superbuild/pull/1407 .

closed 2023-05-17 by traversaro