robotology/robotology-superbuild #848: Build failure of Conda\Linux if ROBOTOLOGY_ENABLE_ROBOT_TESTING option is enabled

traversaro, 2021-08-05:
Since this night, the build of blocktest is failing on Conda\Linux with error:
~~~
2021-08-05T02:23:37.1917330Z [36/94] Linking CXX executable bin/blocktestrunner
2021-08-05T02:23:37.1917840Z FAILED: bin/blocktestrunner 
2021-08-05T02:23:37.1922579Z : && /usr/share/miniconda3/envs/test/bin/x86_64-conda-linux-gnu-c++ -fvisibility-inlines-hidden -std=c++17 -fmessage-length=0 -march=nocona -mtune=haswell -ftree-vectorize -fPIC -fstack-protector-strong -fno-plt -O2 -ffunction-sections -pipe -isystem /usr/share/miniconda3/envs/test/include -Wall -Wextra -O3 -DNDEBUG -rdynamic src/blocktestrunner/CMakeFiles/blocktestrunner.dir/blocktestrunner_autogen/mocs_compilation.cpp.o src/blocktestrunner/CMakeFiles/blocktestrunner.dir/main.cpp.o src/blocktestrunner/CMakeFiles/blocktestrunner.dir/blockTestRunner.cpp.o src/blocktestrunner/CMakeFiles/blocktestrunner.dir/__/__/extern/pugixml/pugixml.cpp.o -o bin/blocktestrunner  -Wl,-rpath,/home/runner/work/robotology-superbuild/robotology-superbuild/build/src/blocktestcore/lib:  lib/libblocktestcore.so.2.3.2  /usr/share/miniconda3/envs/test/lib/libboost_system.so.1.74.0  /usr/share/miniconda3/envs/test/lib/libboost_filesystem.so.1.74.0  -ldl  -lpthread && :
2021-08-05T02:23:37.1927349Z /usr/share/miniconda3/envs/test/bin/../lib/gcc/x86_64-conda-linux-gnu/9.4.0/../../../../x86_64-conda-linux-gnu/bin/ld: warning: libstdc++.so.6, needed by lib/libblocktestcore.so.2.3.2, not found (try using -rpath or -rpath-link)
2021-08-05T02:23:37.1929618Z /usr/share/miniconda3/envs/test/bin/../lib/gcc/x86_64-conda-linux-gnu/9.4.0/../../../../x86_64-conda-linux-gnu/bin/ld: lib/libblocktestcore.so.2.3.2: undefined reference to `VTT for std::__cxx11::basic_stringstream<char, std::char_traits<char>, std::allocator<char> >@GLIBCXX_3.4.21'
2021-08-05T02:23:37.1932075Z /usr/share/miniconda3/envs/test/bin/../lib/gcc/x86_64-conda-linux-gnu/9.4.0/../../../../x86_64-conda-linux-gnu/bin/ld: lib/libblocktestcore.so.2.3.2: undefined reference to `vtable for std::basic_streambuf<char, std::char_traits<char> >@GLIBCXX_3.4'
2021-08-05T02:23:37.2024802Z /usr/share/miniconda3/envs/test/bin/../lib/gcc/x86_64-conda-linux-gnu/9.4.0/../../../../x86_64-conda-linux-gnu/bin/ld: lib/libblocktestcore.so.2.3.2: undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::find_first_not_of(char const*, unsigned long, unsigned long) const@GLIBCXX_3.4.21'
2021-08-05T02:23:37.2026938Z /usr/share/miniconda3/envs/test/bin/../lib/gcc/x86_64-conda-linux-gnu/9.4.0/../../../../x86_64-conda-linux-gnu/bin/ld: lib/libblocktestcore.so.2.3.2: undefined reference to `__cxa_throw_bad_array_new_length@CXXABI_1.3.8'
2021-08-05T02:23:37.2029554Z /usr/share/miniconda3/envs/test/bin/../lib/gcc/x86_64-conda-linux-gnu/9.4.0/../../../../x86_64-conda-linux-gnu/bin/ld: lib/libblocktestcore.so.2.3.2: undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::compare(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const@GLIBCXX_3.4.21'
2021-08-05T02:23:37.2032270Z /usr/share/miniconda3/envs/test/bin/../lib/gcc/x86_64-conda-linux-gnu/9.4.0/../../../../x86_64-conda-linux-gnu/bin/ld: lib/libblocktestcore.so.2.3.2: undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_create(unsigned long&, unsigned long)@GLIBCXX_3.4.21'
2021-08-05T02:23:37.2034304Z /usr/share/miniconda3/envs/test/bin/../lib/gcc/x86_64-conda-linux-gnu/9.4.0/../../../../x86_64-conda-linux-gnu/bin/ld: lib/libblocktestcore.so.2.3.2: undefined reference to `__cxa_allocate_exception@CXXABI_1.3'
2021-08-05T02:23:37.2036486Z /usr/share/miniconda3/envs/test/bin/../lib/gcc/x86_64-conda-linux-gnu/9.4.0/../../../../x86_64-conda-linux-gnu/bin/ld: lib/libblocktestcore.so.2.3.2: undefined reference to `std::basic_ios<char, std::char_traits<char> >::init(std::basic_streambuf<char, std::char_traits<char> >*)@GLIBCXX_3.4'
2021-08-05T02:23:37.2038615Z /usr/share/miniconda3/envs/test/bin/../lib/gcc/x86_64-conda-linux-gnu/9.4.0/../../../../x86_64-conda-linux-gnu/bin/ld: lib/libblocktestcore.so.2.3.2: undefined reference to `std::bad_alloc::~bad_alloc()@GLIBCXX_3.4'
2021-08-05T02:23:37.2040771Z /usr/share/miniconda3/envs/test/bin/../lib/gcc/x86_64-conda-linux-gnu/9.4.0/../../../../x86_64-conda-linux-gnu/bin/ld: lib/libblocktestcore.so.2.3.2: undefined reference to `std::ostream::put(char)@GLIBCXX_3.4'
2021-08-05T02:23:37.2043951Z /usr/share/miniconda3/envs/test/bin/../lib/gcc/x86_64-conda-linux-gnu/9.4.0/../../../../x86_64-conda-linux-gnu/bin/ld: lib/libblocktestcore.so.2.3.2: undefined reference to `std::basic_streambuf<char, std::char_traits<char> >::sync()@GLIBCXX_3.4'
2021-08-05T02:23:37.2048908Z /usr/share/miniconda3/envs/test/bin/../lib/gcc/x86_64-conda-linux-gnu/9.4.0/../../../../x86_64-conda-linux-gnu/bin/ld: lib/libblocktestcore.so.2.3.2: undefined reference to `vtable for std::length_error@GLIBCXX_3.4'
2021-08-05T02:23:37.2174315Z /usr/share/miniconda3/envs/test/bin/../lib/gcc/x86_64-conda-linux-gnu/9.4.0/../../../../x86_64-conda-linux-gnu/bin/ld: /usr/share/miniconda3/envs/test/lib/libboost_filesystem.so.1.74.0: undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::find_first_of(char const*, unsigned long, unsigned long) const@GLIBCXX_3.4.21'
2021-08-05T02:23:37.2176863Z /usr/share/miniconda3/envs/test/bin/../lib/gcc/x86_64-conda-linux-gnu/9.4.0/../../../../x86_64-conda-linux-gnu/bin/ld: lib/libblocktestcore.so.2.3.2: undefined reference to `std::basic_streambuf<char, std::char_traits<char> >::uflow()@GLIBCXX_3.4'
2021-08-05T02:23:37.2179361Z /usr/share/miniconda3/envs/test/bin/../lib/gcc/x86_64-conda-linux-gnu/9.4.0/../../../../x86_64-conda-linux-gnu/bin/ld: lib/libblocktestcore.so.2.3.2: undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::operator=(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&&)@GLIBCXX_3.4.21'
2021-08-05T02:23:37.2181794Z /usr/share/miniconda3/envs/test/bin/../lib/gcc/x86_64-conda-linux-gnu/9.4.0/../../../../x86_64-conda-linux-gnu/bin/ld: lib/libblocktestcore.so.2.3.2: undefined reference to `std::__cxx11::basic_stringbuf<char, std::char_traits<char>, std::allocator<char> >::str() const@GLIBCXX_3.4.21'
2021-08-05T02:23:37.2183825Z /usr/share/miniconda3/envs/test/bin/../lib/gcc/x86_64-conda-linux-gnu/9.4.0/../../../../x86_64-conda-linux-gnu/bin/ld: lib/libblocktestcore.so.2.3.2: undefined reference to `std::__throw_system_error(int)@GLIBCXX_3.4.11'
2021-08-05T02:23:37.2185738Z /usr/share/miniconda3/envs/test/bin/../lib/gcc/x86_64-conda-linux-gnu/9.4.0/../../../../x86_64-conda-linux-gnu/bin/ld: lib/libblocktestcore.so.2.3.2: undefined reference to `std::__detail::_List_node_base::_M_unhook()@GLIBCXX_3.4.15'
2021-08-05T02:23:37.2188028Z /usr/share/miniconda3/envs/test/bin/../lib/gcc/x86_64-conda-linux-gnu/9.4.0/../../../../x86_64-conda-linux-gnu/bin/ld: lib/libblocktestcore.so.2.3.2: undefined reference to `std::basic_filebuf<char, std::char_traits<char> >::open(char const*, std::_Ios_Openmode)@GLIBCXX_3.4'
2021-08-05T02:23:37.2190660Z /usr/share/miniconda3/envs/test/bin/../lib/gcc/x86_64-conda-linux-gnu/9.4.0/../../../../x86_64-conda-linux-gnu/bin/ld: lib/libblocktestcore.so.2.3.2: undefined reference to `std::basic_ostream<char, std::char_traits<char> >& std::operator<< <std::char_traits<char> >(std::basic_ostream<char, std::char_traits<char> >&, char const*)@GLIBCXX_3.4'
2021-08-05T02:23:37.2193648Z /usr/share/miniconda3/envs/test/bin/../lib/gcc/x86_64-conda-linux-gnu/9.4.0/../../../../x86_64-conda-linux-gnu/bin/ld: lib/libblocktestcore.so.2.3.2: undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_replace_aux(unsigned long, unsigned long, unsigned long, char)@GLIBCXX_3.4.21'
2021-08-05T02:23:37.2196350Z /usr/share/miniconda3/envs/test/bin/../lib/gcc/x86_64-conda-linux-gnu/9.4.0/../../../../x86_64-conda-linux-gnu/bin/ld: lib/libblocktestcore.so.2.3.2: undefined reference to `std::ostream& std::ostream::_M_insert<bool>(bool)@GLIBCXX_3.4.9'
2021-08-05T02:23:37.2198848Z /usr/share/miniconda3/envs/test/bin/../lib/gcc/x86_64-conda-linux-gnu/9.4.0/../../../../x86_64-conda-linux-gnu/bin/ld: lib/libblocktestcore.so.2.3.2: undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_append(char const*, [... cut ...]
2021-08-05T02:23:37.4188438Z /usr/share/miniconda3/envs/test/bin/../lib/gcc/x86_64-conda-linux-gnu/9.4.0/../../../../x86_64-conda-linux-gnu/bin/ld: lib/libblocktestcore.so.2.3.2: undefined reference to `std::basic_ios<char, std::char_traits<char> >::clear(std::_Ios_Iostate)@GLIBCXX_3.4'
2021-08-05T02:23:37.4190767Z /usr/share/miniconda3/envs/test/bin/../lib/gcc/x86_64-conda-linux-gnu/9.4.0/../../../../x86_64-conda-linux-gnu/bin/ld: lib/libblocktestcore.so.2.3.2: undefined reference to `std::istream::peek()@GLIBCXX_3.4'
2021-08-05T02:23:37.4191807Z collect2: error: ld returned 1 exit status
~~~

traversaro, 2021-08-05:
Hopefully, this should be fixed by:
* https://github.com/robotology/blocktest/pull/51
* https://github.com/robotology/blocktest-custom-plugins/pull/2
* https://github.com/robotology/blocktest-yarp-plugins/pull/10
* https://github.com/robotology/diagnostic-daemon/pull/18

I also spawned a `generate-conda-packages`  to check if the problem also affects the conda binary generation: https://github.com/robotology/robotology-superbuild/actions/runs/1101337237 .

traversaro, 2021-08-05:
The GitHub Action problem prevents us to test everything, but it is possible that the problem emerged only in blocktest for its use of std::filesystem . Given that @Nicogene do you think we could get a patch release of blocktest ? Thanks in advance!

Nicogene, 2021-08-05:
Yes sure!

Nicogene, 2021-08-05:
Here the blocktest (https://github.com/robotology/blocktest/releases/tag/v2.3.3) and blocktest-yarp-plugins(https://github.com/robotology/blocktest-yarp-plugins/releases/tag/v1.1.2) releases

traversaro, 2021-08-06:
Fixed by https://github.com/robotology/blocktest/pull/51  and https://github.com/robotology/robotology-superbuild/pull/851 .

closed 2021-08-06 by traversaro