robotology/robotology-superbuild #859: Compilation failed

triccyx, 2021-08-30:
git status:
```
On branch master
Your branch is up to date with 'origin/master'.
```
Following:
https://github.com/robotology/robotology-superbuild
and
```
cd robotology-superbuild
sudo bash ./scripts/install_apt_dependencies.sh
```

**In ccmake all default**

When
```
make
```
Error:
```
/home/triccyx/Documents/robotology-superbuild/build/src/YARP/bin/yarpidl_rosmsg: symbol lookup error: /home/triccyx/Documents/robotology-superbuild/build/src/YARP/bin/yarpidl_rosmsg: undefined symbol: _ZN4yarp2os7NetType8toStringB5cxx11Ei
make[5]: *** [src/libYARP_rosmsg/src/CMakeFiles/YARP_rosmsg_gen_trajectory_msgs.dir/build.make:73: src/libYARP_rosmsg/src/yarp/rosmsg/trajectory_msgs/JointTrajectory.h] Error 127
make[5]: Leaving directory '/home/triccyx/Documents/robotology-superbuild/build/src/YARP'
make[4]: *** [CMakeFiles/Makefile2:4299: src/libYARP_rosmsg/src/CMakeFiles/YARP_rosmsg_gen_trajectory_msgs.dir/all] Error 2
make[4]: Leaving directory '/home/triccyx/Documents/robotology-superbuild/build/src/YARP'
make[3]: *** [Makefile:152: all] Error 2
make[3]: Leaving directory '/home/triccyx/Documents/robotology-superbuild/build/src/YARP'
make[2]: *** [CMakeFiles/YARP.dir/build.make:107: src/YARP/CMakeFiles/YCMStamp/YARP-build] Error 2
make[2]: Leaving directory '/home/triccyx/Documents/robotology-superbuild/build'
make[1]: *** [CMakeFiles/Makefile2:388: CMakeFiles/YARP.dir/all] Error 2
make[1]: Leaving directory '/home/triccyx/Documents/robotology-superbuild/build'
make: *** [Makefile:95: all] Error 2
```

traversaro, 2021-08-30:
I suspect you have already an older YARP library version in your system and for some reason the build system when linking `yarpidl_rosmsg` tries to link that one, even if it used the header of the new version. 

Can you please post: 
* The output of the `env` command
* The error output by compiling with `make VERBOSE=1` ? 

Thanks!

triccyx, 2021-08-31:
You are right! I find a lost YARP in a tmp subfolder.
Thanks.

triccyx, 2021-08-31:
@traversaro as this is not the first time I had the problem do you think it is possible to add in cmake a check for inconsistency between the robotology folder and yarp folder with some warnings?

closed 2021-08-31 by triccyx

traversaro, 2021-08-31:
> 
> 
> @traversaro as this is not the first time I had the problem do you think it is possible to add in cmake a check for inconsistency between the robotology folder and yarp folder with some warnings?

Can you report the output of `make VERBOSE=1` ? Otherwise it is difficult to understand the specific kind of problem and provide a check/warning for it?

triccyx, 2021-08-31:
No, I'm sorry after I read your suggestion I find immediately the lost YARP folder and rebuild it all.
However, the problem was that in ```.bash_alias```
```
export YARP_DIR=<old YARP folder>
```
also in ccmake, the <old YARP folder> was present, but I haven't take care of it.

I think that the warning can be issued if YARP_DIR, if present, is different from build/src/YARP or something like this.