robotology/robotology-superbuild #861: Update of latest release of human-dynamics-estimation is blocked by walking-teleoperation release

traversaro, 2021-08-31:
See https://github.com/robotology/robotology-superbuild/pull/857#issuecomment-908090181 . 

cc @lrapetti @kouroshD @Yeshasvitvs

kouroshD, 2021-09-12:
@traversaro  @lrapetti  if it is OK for you and does not delay the super-build release, I will do the release for walking teleopertation by the end of next week. Since, I would like to open a PR for a configuration file.
Otherwise, I can do it today without that addition.

traversaro, 2021-09-12:
> s not delay the super-build release, I will do the release for walking teleopertation by the end of next week. Since, I would like to open a PR for a configuration file.
> Otherwise, I can do it today without that addition.

Sure, no problem! I was just pinged you as remainder, but there is no hurry on my side.

traversaro, 2021-09-12:
> Since, I would like to open a PR for a configuration file.

No need to do it manually, there is an automatic job that monitors new releases and automatically creates PRs like https://github.com/robotology/robotology-superbuild/pull/886 that update the configuration files in robotology-superbuild .

kouroshD, 2021-09-12:
> > Since, I would like to open a PR for a configuration file.
> 
> No need to do it manually, there is an automatic job that monitors new releases and automatically creates PRs like #886 that update the configuration files in robotology-superbuild .

The PR I was referring to is related to some configuration files to add on walking teleoperation side.

traversaro, 2021-09-12:
> The PR I was referring to is related to some configuration files to add on walking teleoperation side.

Ah, I see, I misunderstood!