robotology/robotology-superbuild #871: Missing QT_QPA_PLATFORM_PLUGIN_PATH when installing robotology-superbuild from source on Windows with conda

traversaro, 2021-09-01:
Thanks to @sgiraz tests, it seems that when following the steps in https://github.com/robotology/robotology-superbuild/blob/master/doc/conda-forge.md#run-software-installed, to actually run graphical users interfaces such as FirmwareUpdater, the `QT_QPA_PLATFORM_PLUGIN_PATH` env variable should be defined and point to `%CONDA_PREFIX%\Library\plugins\platforms` . We need to understand if this is a bug in https://github.com/conda-forge/qt-feedstock or something that we need to address on our side.

sgiraz, 2021-09-02:
I would notify another potential problem that could be related to the `conda-Qt` installation.
Once the `yarpmotorgui` is open, the graphics of some button does not appear, despite the buttons and their tooltip exist.

[image]

traversaro, 2021-09-03:
Just to double check, I launched `yarpmotorgui` installed from YARP conda binaries (so without compiling from source the superbuild) and indeed there is no need to of setting `QT_QPA_PLATFORM_PLUGIN_PATH`, and the icons can be found:
[image]

sgiraz, 2021-09-03:
<details>
  <summary>Here my `conda-list`:</summary>

``` console
# Name                    Version                   Build  Channel
ace                       7.0.3                h0e60522_0    conda-forge
asio                      1.14.1               h0e60522_1    conda-forge
boost                     1.74.0           py39hefe7e4c_3    conda-forge
boost-cpp                 1.74.0               h5b4e17d_4    conda-forge
bzip2                     1.0.8                h8ffe710_4    conda-forge
c-compiler                1.2.0                h8ffe710_0    conda-forge
ca-certificates           2021.5.30            h5b45459_0    conda-forge
cairo                     1.16.0            hb19e0ff_1008    conda-forge
cfitsio                   3.470                h0af3d06_7    conda-forge
clang_variant             1.0                     default    conda-forge
clangdev                  5.0.0             h5aa907a_1012    conda-forge
cmake                     3.21.2               h39d44d4_0    conda-forge
compilers                 1.2.0                h57928b3_0    conda-forge
console_bridge            1.0.1                h5362a0b_0    conda-forge
cppzmq                    4.7.1                h4324990_2    conda-forge
curl                      7.78.0               h789b8ee_0    conda-forge
cxx-compiler              1.2.0                h2d74725_0    conda-forge
dirent                    1.21                          0    conda-forge
dlfcn-win32               1.3.0                h0e60522_0    conda-forge
eigen                     3.4.0                h2d74725_0    conda-forge
expat                     2.4.1                h39d44d4_0    conda-forge
ffmpeg                    4.3.1                ha925a31_0    conda-forge
flang                     5.0.0           he025d50_20180525    conda-forge
flang_win-64              5.0.0           h13ae965_20180526    conda-forge
fontconfig                2.13.1            h1989441_1005    conda-forge
fortran-compiler          1.2.0                h9655429_0    conda-forge
freeglut                  3.2.1                h0e60522_2    conda-forge
freeimage                 3.18.0               hfcb8cd7_7    conda-forge
freetype                  2.10.4               h546665d_1    conda-forge
freexl                    1.0.6                ha8e266a_0    conda-forge
gazebo                    11.8.1               h2e5bf29_0    conda-forge
geos                      3.9.1                h39d44d4_2    conda-forge
geotiff                   1.6.0                ha8a8a2d_6    conda-forge
gettext                   0.19.8.1          h1a89ca6_1005    conda-forge
glew                      2.1.0                h39d44d4_2    conda-forge
glfw                      3.3.4                h8ffe710_0    conda-forge
gsl                       2.7                  hdfb1a43_0    conda-forge
gts                       0.7.6                h7c369d9_2    conda-forge
hdf4                      4.2.15               h0e5069d_3    conda-forge
hdf5                      1.10.6          nompi_h5268f04_1114    conda-forge
icu                       68.1                 h0e60522_0    conda-forge
ilmbase                   2.5.5                h12d4b20_0    conda-forge
intel-openmp              2021.3.0          h57928b3_3372    conda-forge
ipopt                     3.14.2               hf6be2e5_0    conda-forge
jasper                    2.0.14               h77af90b_2    conda-forge
jbig                      2.1               h8d14728_2003    conda-forge
jpeg                      9d                   h8ffe710_0    conda-forge
jsoncpp                   1.9.4                h2d74725_3    conda-forge
jxrlib                    1.1                  h8ffe710_2    conda-forge
kealib                    1.4.14               h96bfa42_2    conda-forge
krb5                      1.19.2               hbae68bd_0    conda-forge
lcms2                     2.12                 h2a16943_0    conda-forge
lerc                      2.2.1                h0e60522_0    conda-forge
libblas                   3.9.0                     8_mkl    conda-forge
libcblas                  3.9.0                     8_mkl    conda-forge
libclang                  11.1.0          default_h5c34c98_1    conda-forge
libcurl                   7.78.0               h789b8ee_0    conda-forge
libdeflate                1.7                  h8ffe710_5    conda-forge
libffi                    3.3                  h0e60522_2    conda-forge
libflang                  5.0.0           h6538335_20180525    conda-forge
libgdal                   3.3.1                h7e75cf7_2    conda-forge
libglib                   2.68.4               h1e62bf3_0    conda-forge
libiconv                  1.16                 he774522_0    conda-forge
libignition-cmake2        2.8.0                h0e60522_0    conda-forge
libignition-common3       3.13.2               h9bd54ec_1    conda-forge
[... 75 lines cut ...]
tiledb                    2.3.3                h78dabda_0    conda-forge
tiny-process-library      2.0.4                h0e60522_0    conda-forge
tinyxml                   2.6.2                h2d74725_2    conda-forge
tinyxml2                  9.0.0                h39d44d4_1    conda-forge
tk                        8.6.11               h8ffe710_0    conda-forge
tzdata                    2021a                he74cb21_1    conda-forge
ucrt                      10.0.20348.0         h57928b3_0    conda-forge
urdfdom                   2.3.5                h2d74725_0    conda-forge
[... cut ...]
xerces-c                  3.2.3                h0e60522_2    conda-forge
xz                        5.2.5                h62dcd97_1    conda-forge
yaml                      0.2.5                he774522_0    conda-forge
zeromq                    4.3.4                h0e60522_1    conda-forge
zlib                      1.2.11            h62dcd97_1010    conda-forge
zstd                      1.5.0                h6255e5f_0    conda-forge
zziplib                   0.13.69              h1d00b33_1    conda-forge
```
<details>

It doesn't seem to be differences in `Qt` ...

traversaro, 2021-09-03:
> there seems to be no difference about `Qt` ...

Yes, the difference is probably related to where YARP resources and libraries are installed, as long as they are installed in the same install prefix of qt, everything works fine, otherwise not. Probably something that can be fixed either by seeting env variables, or by some fixes in qt-feedstock, let's see.

traversaro, 2021-09-07:
I was able to replicate the issue by just compiling YARP from source, instead of using the complete robotology-superbuild.
Commands: 
~~~
mamba create -n yarpdeps 
mamba activate yarpdeps
mamba install ace vs2019_win-64 eigen qt ycm-cmake-modules cmake ninja
git clone https://github.com/robotology/yarp
cd yarp
md build
cd build
cmake -DCMAKE_INSTALL_PREFIX=<...> -GNinja ..
ninja install 
~~~ 

Interestingly, if instead we set the conda location as install prefix (i.e. `-DCMAKE_INSTALL_PREFIX=%CONDA_PREFIX%\Library`) everything works fine. See the following screenshot:

[image]

The bottom windows is yarp is installed in `%CONDA_PREFIX%\Library`, while the top is installed in an arbitrary library (after setting manually `QT_QPA_PLATFORM_PLUGIN_PATH`.

Installed packages:
~~~
# packages in environment at C:\Users\STraversaro\AppData\Local\mambaforge\envs\yarpdeps:
#
# Name                    Version                   Build  Channel
ace                       7.0.3                h0e60522_0    conda-forge
ca-certificates           2021.5.30            h5b45459_0    conda-forge
cmake                     3.21.2               h39d44d4_0    conda-forge
eigen                     3.4.0                h2d74725_0    conda-forge
icu                       68.1                 h0e60522_0    conda-forge
jpeg                      9d                   h8ffe710_0    conda-forge
libclang                  11.1.0          default_h5c34c98_1    conda-forge
libpng                    1.6.37               h1d00b33_2    conda-forge
ninja                     1.10.2               h5362a0b_0    conda-forge
openssl                   1.1.1l               h8ffe710_0    conda-forge
qt                        5.12.9               h5909a2a_4    conda-forge
sqlite                    3.36.0               h8ffe710_0    conda-forge
ucrt                      10.0.20348.0         h57928b3_0    conda-forge
vc                        14.2                 hb210afc_5    conda-forge
vs2015_runtime            14.29.30037          h902a5da_5    conda-forge
vs2019_win-64             19.29.30037          h774bd60_5    conda-forge
vswhere                   2.8.4                h57928b3_0    conda-forge
ycm-cmake-modules         0.13.0               h0e60522_0    conda-forge
zlib                      1.2.11            h62dcd97_1010    conda-forge
~~~

traversaro, 2021-09-07:
Interestingly, yarpino (the small YARP mascotte on the bottom right) is the only one always working, @Nicogene do you have any idea how this is different from the rest of the icons?

traversaro, 2021-09-07:
Also in the case of yarpmotorgui some icons are there while something disappear.

traversaro, 2021-09-07:
Related issue: https://github.com/microsoft/vcpkg/issues/7461 .

traversaro, 2021-09-07:
I found an alternative workaround (instead of setting `QT_QPA_PLATFORM_PLUGIN_PATH`), that also fixes the icon problem. If you copy the `%CONDA_PREFIX%\qt.conf` file in the `<yarp_cmake_install_prefix>\bin` , i.e. : 
~~~
cd <yarp_cmake_install_prefix>\bin
copy %CONDA_PREFIX%\qt.conf .\
~~~

Then everything works correctly:
[image]

Nicogene, 2021-09-07:
> Interestingly, yarpino (the small YARP mascotte on the bottom right) is the only one always working, @Nicogene do you have any idea how this is different from the rest of the icons?

Maybe because yarpino is set from code ad Pixmap ? 

https://github.com/robotology/yarp/blob/15ff8fdf5c68c7be04494b778bafa0487807cc48/src/yarpmanager/src-manager/mainwindow.cpp#L84-L88

traversaro, 2021-09-07:
> 
> 
> > Interestingly, yarpino (the small YARP mascotte on the bottom right) is the only one always working, @Nicogene do you have any idea how this is different from the rest of the icons?
> 
> Maybe because yarpino is set from code as Pixmap ?
> 
> https://github.com/robotology/yarp/blob/15ff8fdf5c68c7be04494b778bafa0487807cc48/src/yarpmanager/src-manager/mainwindow.cpp#L84-L88

If that the case, for sure this is the reason. The missing icon are a problem of resources not found, but if an image is part of the code that bypass any resource location problem.

traversaro, 2021-09-07:
> Maybe because yarpino is set from code as Pixmap ?
>
> https://github.com/robotology/yarp/blob/15ff8fdf5c68c7be04494b778bafa0487807cc48/src/yarpmanager/src-manager/mainwindow.cpp#L84-L88

I think the reason is probably related to how the `:/yarp-robot-22.png` file is located.

closed 2021-09-09 by traversaro