robotology/robotology-superbuild #872: Unstable branches CI failure September 2021

traversaro, 2021-09-02:
See https://github.com/robotology/robotology-superbuild/runs/3491205115 .

Seems some incompatibility between std-defined byte and iCub_SIM, probably related to https://github.com/robotology/yarp/pull/2700 .

~~~
2021-09-02T02:41:33.3567443Z [334/458] Building CXX object src/simulators/iCubSimulation/CMakeFiles/iCub_SIM.dir/odesdl/MS3D.cpp.o
2021-09-02T02:41:33.3568333Z FAILED: src/simulators/iCubSimulation/CMakeFiles/iCub_SIM.dir/odesdl/MS3D.cpp.o 
2021-09-02T02:41:33.3575273Z /usr/bin/c++ -DICUB_SIM_ENABLE_FAKE -DICUB_SIM_ENABLE_ODESDL -DOMIT_LOGPOLAR -DUSE_GLUT -D_USE_MATH_DEFINES -DdDOUBLE -I/home/runner/work/robotology-superbuild/robotology-superbuild/src/ICUB/src/simulators/iCubSimulation -I/home/runner/work/robotology-superbuild/robotology-superbuild/src/ICUB/src/simulators/iCubSimulation/wrapper -I/home/runner/work/robotology-superbuild/robotology-superbuild/src/ICUB/src/simulators/iCubSimulation/fake -I/usr/include/SDL -I/home/runner/work/robotology-superbuild/robotology-superbuild/src/ICUB/src/simulators/iCubSimulation/odesdl -I/home/runner/work/robotology-superbuild/robotology-superbuild/src/ICUB/src/libraries/iKin/include -I/home/runner/work/robotology-superbuild/robotology-superbuild/src/ICUB/src/libraries/ctrlLib/include -I/home/runner/work/robotology-superbuild/robotology-superbuild/src/ICUB/src/libraries/skinDynLib/include -isystem /home/runner/work/robotology-superbuild/robotology-superbuild/build/install/include -O3 -DNDEBUG -fPIE -std=c++17 -MD -MT src/simulators/iCubSimulation/CMakeFiles/iCub_SIM.dir/odesdl/MS3D.cpp.o -MF src/simulators/iCubSimulation/CMakeFiles/iCub_SIM.dir/odesdl/MS3D.cpp.o.d -o src/simulators/iCubSimulation/CMakeFiles/iCub_SIM.dir/odesdl/MS3D.cpp.o -c /home/runner/work/robotology-superbuild/robotology-superbuild/src/ICUB/src/simulators/iCubSimulation/odesdl/MS3D.cpp
2021-09-02T02:41:33.3581518Z In file included from /home/runner/work/robotology-superbuild/robotology-superbuild/src/ICUB/src/simulators/iCubSimulation/odesdl/MS3D.h:40,
2021-09-02T02:41:33.3583084Z                  from /home/runner/work/robotology-superbuild/robotology-superbuild/src/ICUB/src/simulators/iCubSimulation/odesdl/MS3D.cpp:28:
2021-09-02T02:41:33.3584113Z /usr/include/SDL/SDL_opengl.h:116: warning: "GL_GLEXT_VERSION" redefined
2021-09-02T02:41:33.3584686Z   116 | #define GL_GLEXT_VERSION 29
2021-09-02T02:41:33.3585030Z       | 
2021-09-02T02:41:33.3585432Z In file included from /usr/include/GL/gl.h:2050,
2021-09-02T02:41:33.3586034Z                  from /usr/include/SDL/SDL_opengl.h:46,
2021-09-02T02:41:33.3587168Z                  from /home/runner/work/robotology-superbuild/robotology-superbuild/src/ICUB/src/simulators/iCubSimulation/odesdl/MS3D.h:40,
2021-09-02T02:41:33.3588699Z                  from /home/runner/work/robotology-superbuild/robotology-superbuild/src/ICUB/src/simulators/iCubSimulation/odesdl/MS3D.cpp:28:
2021-09-02T02:41:33.3589741Z /usr/include/GL/glext.h:54: note: this is the location of the previous definition
2021-09-02T02:41:33.3590335Z    54 | #define GL_GLEXT_VERSION 20190805
2021-09-02T02:41:33.3590703Z       | 
2021-09-02T02:41:33.3591747Z In file included from /home/runner/work/robotology-superbuild/robotology-superbuild/src/ICUB/src/simulators/iCubSimulation/odesdl/MS3D.cpp:28:
2021-09-02T02:41:33.3593445Z /home/runner/work/robotology-superbuild/robotology-superbuild/src/ICUB/src/simulators/iCubSimulation/odesdl/MS3D.h:84:5: error: reference to ‘byte’ is ambiguous
2021-09-02T02:41:33.3594377Z    84 |     byte m_flags;
2021-09-02T02:41:33.3594692Z       |     ^~~~
2021-09-02T02:41:33.3595691Z In file included from /home/runner/work/robotology-superbuild/robotology-superbuild/build/install/include/yarp/conf/numeric.h:19,
2021-09-02T02:41:33.3597208Z                  from /home/runner/work/robotology-superbuild/robotology-superbuild/build/install/include/yarp/os/ConnectionReader.h:10,
2021-09-02T02:41:33.3598828Z                  from /home/runner/work/robotology-superbuild/robotology-superbuild/build/install/include/yarp/os/PortReaderCreator.h:10,
2021-09-02T02:41:33.3600306Z                  from /home/runner/work/robotology-superbuild/robotology-superbuild/build/install/include/yarp/os/Port.h:13,
2021-09-02T02:41:33.3601731Z                  from /home/runner/work/robotology-superbuild/robotology-superbuild/build/install/include/yarp/os/BufferedPort.h:11,
2021-09-02T02:41:33.3603276Z                  from /home/runner/work/robotology-superbuild/robotology-superbuild/src/ICUB/src/simulators/iCubSimulation/odesdl/VideoTexture.h:34,
2021-09-02T02:41:33.3604882Z                  from /home/runner/work/robotology-superbuild/robotology-superbuild/src/ICUB/src/simulators/iCubSimulation/odesdl/rendering.h:33,
2021-09-02T02:41:33.3606435Z                  from /home/runner/work/robotology-superbuild/robotology-superbuild/src/ICUB/src/simulators/iCubSimulation/odesdl/MS3D.h:52,
2021-09-02T02:41:33.3607951Z                  from /home/runner/work/robotology-superbuild/robotology-superbuild/src/ICUB/src/simulators/iCubSimulation/odesdl/MS3D.cpp:28:
2021-09-02T02:41:33.3609312Z /usr/include/c++/9/cstddef:69:14: note: candidates are: ‘enum class std::byte’
2021-09-02T02:41:33.3609977Z    69 |   enum class byte : unsigned char {};
2021-09-02T02:41:33.3610355Z       |              ^~~~
2021-09-02T02:41:33.3611444Z In file included from /home/runner/work/robotology-superbuild/robotology-superbuild/src/ICUB/src/simulators/iCubSimulation/odesdl/MS3D.cpp:28:
2021-09-02T02:41:33.3613140Z /home/runner/work/robotology-superbuild/robotology-superbuild/src/ICUB/src/simulators/iCubSimulation/odesdl/MS3D.h:73:23: note:                 ‘typedef unsigned char byte’
2021-09-02T02:41:33.3614131Z    73 | typedef unsigned char byte;
2021-09-02T02:41:33.3614523Z       |                       ^~~~
~~~

traversaro, 2021-09-02:
Opened upstream issue in https://github.com/robotology/icub-main/issues/763 .

traversaro, 2021-09-02:
As you can see in https://github.com/robotology/robotology-superbuild/runs/3494858604, that error was fixed by https://github.com/robotology/icub-main/issues/763 . Now there is another error that is related to https://github.com/robotology/gazebo-yarp-plugins/pull/575 : to speed up CI time, we were installing Gazebo from Ubuntu repositories instead of OpenRobotics one, so the Gazebo version was 9.12 . However in the docs we always suggested to use the OpenRobotics apt repos to get the latest version, so we just need to switch the CI to be coherent with the docs.

traversaro, 2021-09-02:
We actually also have some CI failures on macOS:
~~~
2021-09-02T12:34:08.9333270Z In file included from /Users/runner/work/robotology-superbuild/robotology-superbuild/build/src/speech/svox-speech/yarp_plugin_speech.cpp:10:
2021-09-02T12:34:08.9436220Z In file included from /Users/runner/work/robotology-superbuild/robotology-superbuild/src/speech/svox-speech/speech.h:22:
2021-09-02T12:34:08.9539250Z In file included from /Users/runner/work/robotology-superbuild/robotology-superbuild/src/speech/svox-speech/./svoxpico/picoapid.h:38:
2021-09-02T12:34:08.9642220Z In file included from /Users/runner/work/robotology-superbuild/robotology-superbuild/src/speech/svox-speech/./svoxpico/picorsrc.h:38:
2021-09-02T12:34:08.9745620Z /Users/runner/work/robotology-superbuild/robotology-superbuild/src/speech/svox-speech/./svoxpico/picoknow.h:167:53: error: ISO C++17 does not allow 'register' storage class specifier [-Wregister]
2021-09-02T12:34:08.9848060Z typedef pico_status_t (* picoknow_kbSubDeallocate) (register picoknow_KnowledgeBase that, picoos_MemoryManager mm);
2021-09-02T12:34:08.9949780Z                                                     ^~~~~~~~~
2021-09-02T12:34:09.0052540Z In file included from /Users/runner/work/robotology-superbuild/robotology-superbuild/build/src/speech/svox-speech/yarp_plugin_speech.cpp:10:
2021-09-02T12:34:09.0155370Z In file included from /Users/runner/work/robotology-superbuild/robotology-superbuild/src/speech/svox-speech/speech.h:22:
2021-09-02T12:34:09.0258520Z In file included from /Users/runner/work/robotology-superbuild/robotology-superbuild/src/speech/svox-speech/./svoxpico/picoapid.h:39:
2021-09-02T12:34:09.0361980Z In file included from /Users/runner/work/robotology-superbuild/robotology-superbuild/src/speech/svox-speech/./svoxpico/picoctrl.h:33:
2021-09-02T12:34:09.0464870Z /Users/runner/work/robotology-superbuild/robotology-superbuild/src/speech/svox-speech/./svoxpico/picodata.h:86:32: error: ISO C++17 does not allow 'register' storage class specifier [-Wregister]
2021-09-02T12:34:09.0564400Z pico_status_t picodata_cbPutCh(register picodata_CharBuffer that, picoos_char ch);
2021-09-02T12:34:09.0665880Z                                ^~~~~~~~~
2021-09-02T12:34:09.0768950Z /Users/runner/work/robotology-superbuild/robotology-superbuild/src/speech/svox-speech/./svoxpico/picodata.h:89:31: error: ISO C++17 does not allow 'register' storage class specifier [-Wregister]
2021-09-02T12:34:09.0871140Z picoos_int16 picodata_cbGetCh(register picodata_CharBuffer that);
2021-09-02T12:34:09.0972150Z                               ^~~~~~~~~
2021-09-02T12:34:09.1074520Z /Users/runner/work/robotology-superbuild/robotology-superbuild/src/speech/svox-speech/./svoxpico/picodata.h:92:33: error: ISO C++17 does not allow 'register' storage class specifier [-Wregister]
2021-09-02T12:34:09.1202040Z pico_status_t picodata_cbReset (register picodata_CharBuffer that);
2021-09-02T12:34:09.1309920Z                                 ^~~~~~~~~
2021-09-02T12:34:09.1414480Z /Users/runner/work/robotology-superbuild/robotology-superbuild/src/speech/svox-speech/./svoxpico/picodata.h:301:34: error: ISO C++17 does not allow 'register' storage class specifier [-Wregister]
2021-09-02T12:34:09.1416130Z pico_status_t picodata_cbGetItem(register picodata_CharBuffer that,
2021-09-02T12:34:09.1515880Z                                  ^~~~~~~~~
2021-09-02T12:34:09.1619300Z /Users/runner/work/robotology-superbuild/robotology-superbuild/src/speech/svox-speech/./svoxpico/picodata.h:313:40: error: ISO C++17 does not allow 'register' storage class specifier [-Wregister]
2021-09-02T12:34:09.1722450Z pico_status_t picodata_cbGetSpeechData(register picodata_CharBuffer that,
2021-09-02T12:34:09.1823920Z                                        ^~~~~~~~~
2021-09-02T12:34:09.1926810Z /Users/runner/work/robotology-superbuild/robotology-superbuild/src/speech/svox-speech/./svoxpico/picodata.h:324:34: error: ISO C++17 does not allow 'register' storage class specifier [-Wregister]
2021-09-02T12:34:09.2028980Z pico_status_t picodata_cbPutItem(register picodata_CharBuffer that,
2021-09-02T12:34:09.2130340Z                                  ^~~~~~~~~
2021-09-02T12:34:09.2233340Z /Users/runner/work/robotology-superbuild/robotology-superbuild/src/speech/svox-speech/./svoxpico/picodata.h:329:42: error: ISO C++17 does not allow 'register' storage class specifier [-Wregister]
2021-09-02T12:34:09.2332460Z picoos_uint8 picodata_cbGetFrontItemType(register picodata_CharBuffer that);
2021-09-02T12:34:09.2433900Z                                          ^~~~~~~~~
2021-09-02T12:34:09.2536960Z /Users/runner/work/robotology-superbuild/robotology-superbuild/src/speech/svox-speech/./svoxpico/picodata.h:501:56: error: ISO C++17 does not allow 'register' storage class specifier [-Wregister]
2021-09-02T12:34:09.2639260Z typedef pico_status_t (* picodata_puInitializeMethod) (register picodata_ProcessingUnit that, picoos_int32 mode);
2021-09-02T12:34:09.2740980Z                                                        ^~~~~~~~~
2021-09-02T12:34:09.2843710Z /Users/runner/work/robotology-superbuild/robotology-superbuild/src/speech/svox-speech/./svoxpico/picodata.h:502:55: error: ISO C++17 does not allow 'register' storage class specifier [-Wregister]
2021-09-02T12:34:09.2945960Z typedef pico_status_t (* picodata_puTerminateMethod) (register picodata_ProcessingUnit that);
2021-09-02T12:34:09.3047560Z                                                       ^~~~~~~~~
[... cut ...]
2021-09-02T12:34:09.3466250Z /Users/runner/work/robotology-superbuild/robotology-superbuild/src/speech/svox-speech/./svoxpico/picodata.h:504:59: error: ISO C++17 does not allow 'register' storage class specifier [-Wregister]
2021-09-02T12:34:09.3568640Z typedef pico_status_t (* picodata_puSubDeallocateMethod) (register picodata_ProcessingUnit that, picoos_MemoryManager mm);
2021-09-02T12:34:09.3670530Z                                                           ^~~~~~~~~
2021-09-02T12:34:09.3771960Z 12 errors generated.
2021-09-02T12:34:09.3874360Z [44/45] Building CXX object svox-speech/CMakeFiles/svox_speech.dir/Speech_IDL.cpp.o
~~~

traversaro, 2021-09-02:
Upstream issue: https://github.com/robotology/speech/issues/40 .

pattacini, 2021-09-02:
> Upstream issue: [robotology/speech#40](https://github.com/robotology/speech/issues/40) .

Fix provided in https://github.com/robotology/speech/pull/41.
We should fire up the CI again.

pattacini, 2021-09-02:
>We should fire up the CI again.

Jobs just started over in https://github.com/robotology/robotology-superbuild/pull/870.
@traversaro, I've finally found the button to re-run all checks.

traversaro, 2021-09-03:
Thanks @pattacini ! A lot of failures are related to commit https://github.com/robotology/yarp/commit/6f43d4d91dc4962bc523a207f42e06051d4276b0, that however does not come with a ChangeLog line. @drdanz given this experience, probably it can make sense to add a changelog line on this change before YARP 3.6 release?

traversaro, 2021-09-03:
Fixed by:
* https://github.com/robotology/robotology-superbuild/pull/873
* https://github.com/robotology/speech/pull/41

closed 2021-09-03 by traversaro

drdanz, 2021-09-06:
Actually that was not supposed to be merged yet, I think I messed up somehow...

drdanz, 2021-09-06:
I reverted the commit, it will come (soon) in a pull request.