robotology/robotology-superbuild #892: CMake failure with libdl not found following source installation with conda-forge dependencies

prashanthr05, 2021-09-20:
I am trying to setup `robotology-superbuild` from source using conda-forge dependencies by following the tutorial in https://github.com/robotology/robotology-superbuild/blob/master/doc/conda-forge.md#source-installation. 
The installation is being done on a fresh Ubuntu 20.04 installation. 

The setup fails at CMake configuration step with the following log,

``` 
cmake ..
-- YCM not found. Bootstrapping it.
-- Downloading YCMEPHelper.cmake - Attempt 1 of 3
-- Downloading YCMEPHelper.cmake - SUCCESS
-- Downloading StyleGITHUB.cmake - Attempt 1 of 3
-- Downloading StyleGITHUB.cmake - SUCCESS
-- Performing download step (git clone) for 'YCM'
-- Performing update step for 'YCM'
-- Performing configure step for 'YCM'
-- Performing uninstall step for 'YCM'
-- Performing build step for 'YCM'
-- Performing install step for 'YCM'
-- Found YCM: /home/pramadoss/iit_ws/robotology/robotology-superbuild/build/install/share/cmake/YCM (found version "0.13.0.2-20210705.5+gitbed44ba")
-- Package YARP not found. Will be downloaded and built.
-- Package ICUB not found. Will be downloaded and built.
-- Package ICUBcontrib not found. Will be downloaded and built.
-- Package icub-models not found. Will be downloaded and built.
-- Package robots-configuration not found. Will be downloaded and built.
CMake Warning (dev) at /home/pramadoss/mambaforge/envs/robsub/share/cmake-3.21/Modules/FindPackageHandleStandardArgs.cmake:438 (message):
  The package name passed to `find_package_handle_standard_args` (PkgConfig)
  does not match the name of the calling package (gazebo).  This can lead to
  problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  /home/pramadoss/mambaforge/envs/robsub/share/cmake-3.21/Modules/FindPkgConfig.cmake:70 (find_package_handle_standard_args)
  /home/pramadoss/mambaforge/envs/robsub/lib/cmake/gazebo/gazebo-config.cmake:62 (include)
  cmake/BuildGazeboYARPPlugins.cmake:31 (find_package)
  build/install/share/YCM/modules/FindOrBuildPackage.cmake:204 (include)
  cmake/RobotologySuperbuildLogic.cmake:19 (find_or_build_package)
  CMakeLists.txt:63 (include)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found DART: /home/pramadoss/mambaforge/envs/robsub/include (Required is at least version "6.6") found components: dart 
-- Found Protobuf: /home/pramadoss/mambaforge/envs/robsub/lib/libprotobuf.so;-lpthread (found version "3.16.0") 
-- Looking for ignition-math6 -- found version 6.8.0
-- Searching for dependencies of ignition-math6
-- Found OGRE
--   static     : OFF
--   components : Bites;HLMS;MeshLodGenerator;Overlay;Paging;Property;RTShaderSystem;Terrain;Volume
--   plugins    : Plugin_BSPSceneManager;Plugin_OctreeSceneManager;Plugin_PCZSceneManager;Plugin_ParticleFX;RenderSystem_GL;RenderSystem_GL3Plus
--   media      : /home/pramadoss/mambaforge/envs/robsub/share/OGRE/Media
-- Looking for ignition-math6 -- found version 6.8.0
-- Looking for ignition-transport8 -- found version 8.1.0
-- Searching for dependencies of ignition-transport8
-- Found Protobuf: /home/pramadoss/mambaforge/envs/robsub/lib/libprotobuf.so;-lpthread (found suitable version "3.16.0", minimum required is "3") 
-- Checking for module 'uuid'
--   Found uuid, version 2.32.1
-- Looking for ignition-msgs5 -- found version 5.7.0
-- Searching for dependencies of ignition-msgs5
-- Looking for ignition-math6 -- found version 6.8.0
-- Checking for module 'tinyxml2'
--   Found tinyxml2, version 9.0.0
-- Looking for ignition-msgs5 -- found version 5.7.0
-- Looking for ignition-common3 -- found version 3.13.2
-- Searching for dependencies of ignition-common3
-- Looking for dlfcn.h - not found
-- Looking for libdl - not found
CMake Error at /home/pramadoss/mambaforge/envs/robsub/share/cmake-3.21/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find DL (missing: DL_FOUND)
Call Stack (most recent call first):
  /home/pramadoss/mambaforge/envs/robsub/share/cmake-3.21/Modules/FindPackageHandleStandardArgs.cmake:594 (_FPHSA_FAILURE_MESSAGE)
  /home/pramadoss/mambaforge/envs/robsub/share/cmake/ignition-cmake2/cmake2/FindDL.cmake:101 (find_package_handle_standard_args)
  /home/pramadoss/mambaforge/envs/robsub/lib/cmake/ignition-common3/ignition-common3-config.cmake:92 (find_package)
  /home/pramadoss/mambaforge/envs/robsub/lib/cmake/gazebo/gazebo-config.cmake:240 (find_package)
  cmake/BuildGazeboYARPPlugins.cmake:31 (find_package)
  build/install/share/YCM/modules/FindOrBuildPackage.cmake:204 (include)
  cmake/RobotologySuperbuildLogic.cmake:19 (find_or_build_package)
  CMakeLists.txt:63 (include)

-- Configuring incomplete, errors occurred!
See also "/home/pramadoss/iit_ws/robotology/robotology-superbuild/build/CMakeFiles/CMakeOutput.log".
See also "/home/pramadoss/iit_ws/robotology/robotology-superbuild/build/CMakeFiles/CMakeError.log".
```

On searching for the `libdl.so`, it exists. However `whereis` output is empty, 
``` 
pramadoss@iiticublap264:~$ ls /lib/x86_64-linux-gnu/libdl*
/lib/x86_64-linux-gnu/libdl-2.31.so  /lib/x86_64-linux-gnu/libdl.so.2
pramadoss@iiticublap264:~$ whereis libdl
libdl:
```

Any help would be appreciated. Thanks in advance.

prashanthr05, 2021-09-20:
cc @traversaro

traversaro, 2021-09-20:
Can you report the output of the `conda info`  and `conda list` commands? Thanks!

traversaro, 2021-09-20:
For reference, the steps in CI that seems not to be affected by this problem are https://github.com/robotology/robotology-superbuild/blob/master/.github/workflows/ci.yml#L75 .

prashanthr05, 2021-09-20:
> Can you report the output of the `conda info` and `conda list` commands? Thanks!

<details>
<summary> conda info </summary>

``` sh

     active environment : robsub
    active env location : /home/pramadoss/mambaforge/envs/robsub
            shell level : 1
       user config file : /home/pramadoss/.condarc
 populated config files : /home/pramadoss/mambaforge/.condarc
                          /home/pramadoss/.condarc
          conda version : 4.10.3
    conda-build version : not installed
         python version : 3.9.6.final.0
       virtual packages : __linux=5.11.0=0
                          __glibc=2.31=0
                          __unix=0=0
                          __archspec=1=x86_64
       base environment : /home/pramadoss/mambaforge  (writable)
      conda av data dir : /home/pramadoss/mambaforge/etc/conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/conda-forge/linux-64
                          https://conda.anaconda.org/conda-forge/noarch
          package cache : /home/pramadoss/mambaforge/pkgs
                          /home/pramadoss/.conda/pkgs
       envs directories : /home/pramadoss/mambaforge/envs
                          /home/pramadoss/.conda/envs
               platform : linux-64
             user-agent : conda/4.10.3 requests/2.26.0 CPython/3.9.6 Linux/5.11.0-34-generic ubuntu/20.04.3 glibc/2.31
                UID:GID : 1000:1000
             netrc file : None
           offline mode : False

```
</details>

<details>
<summary> conda list </summary>

```
conda list
# packages in environment at /home/pramadoss/mambaforge/envs/robsub:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                       1_gnu    conda-forge
ace                       7.0.3                h9c3ff4c_0    conda-forge
alsa-lib                  1.2.3                h516909a_0    conda-forge
ampl-mp                   3.1.0             h2cc385e_1005    conda-forge
asio                      1.14.1               h9c3ff4c_1    conda-forge
assimp                    5.0.1                hedfc422_6    conda-forge
atk-1.0                   2.36.0               h3371d22_4    conda-forge
bash-completion           2.11                          0    conda-forge
binutils                  2.36.1               hdd6e379_2    conda-forge
binutils_impl_linux-64    2.36.1               h193b22a_2    conda-forge
binutils_linux-64         2.36                 hf3e587d_1    conda-forge
boost                     1.74.0           py39h5472131_3    conda-forge
boost-cpp                 1.74.0               h312852a_4    conda-forge
bullet-cpp                3.17                 hde0f152_0    conda-forge
bzip2                     1.0.8                h7f98852_4    conda-forge
c-ares                    1.17.2               h7f98852_0    conda-forge
c-compiler                1.3.0                h7f98852_0    conda-forge
ca-certificates           2021.5.30            ha878542_0    conda-forge
cairo                     1.16.0            h6cf1ce9_1008    conda-forge
cfitsio                   3.470                hb418390_7    conda-forge
cmake                     3.21.2               h8897547_0    conda-forge
compilers                 1.3.0                ha770c72_0    conda-forge
console_bridge            1.0.1                h4bd325d_0    conda-forge
[... 227 lines cut ...]
xorg-libxaw               1.0.14               h7f98852_0    conda-forge
xorg-libxcursor           1.2.0                h516909a_0    conda-forge
xorg-libxdmcp             1.1.3                h7f98852_0    conda-forge
xorg-libxext              1.3.4                h516909a_0    conda-forge
xorg-libxfixes            5.0.3             h516909a_1004    conda-forge
xorg-libxi                1.7.10               h516909a_0    conda-forge
xorg-libxinerama          1.1.4             h9c3ff4c_1001    conda-forge
xorg-libxmu               1.1.3                h516909a_0    conda-forge
xorg-libxpm               3.5.13               h516909a_0    conda-forge
xorg-libxrandr            1.5.2                h516909a_1    conda-forge
xorg-libxrender           0.9.10            h516909a_1002    conda-forge
xorg-libxt                1.1.5             h516909a_1003    conda-forge
xorg-randrproto           1.5.0             h7f98852_1001    conda-forge
xorg-renderproto          0.11.1            h7f98852_1002    conda-forge
xorg-xextproto            7.3.0             h7f98852_1002    conda-forge
xorg-xproto               7.0.31            h7f98852_1007    conda-forge
xz                        5.2.5                h516909a_1    conda-forge
yaml                      0.2.5                h516909a_0    conda-forge
zeromq                    4.3.4                h9c3ff4c_1    conda-forge
zlib                      1.2.11            h516909a_1010    conda-forge
zstd                      1.5.0                ha95c52a_0    conda-forge
zziplib                   0.13.69              h27826a3_1    conda-forge 
```

</details>

prashanthr05, 2021-09-20:
> For reference, the steps in CI that seems not to be affected by this problem are
> 
> https://github.com/robotology/robotology-superbuild/blob/a4b8d6730a0ff6b49cb9369ab7b2f3e2172cea97/.github/workflows/ci.yml#L75
> .

Following the dependency installation and then running CMake, I still end up with same error.

traversaro, 2021-09-20:
I just tried with the exact same environment down to the version+build number of all dependencies, and it is working fine:
~~~
- YCM not found. Bootstrapping it.
-- Downloading YCMEPHelper.cmake - Attempt 1 of 3
-- Downloading YCMEPHelper.cmake - SUCCESS
-- Downloading StyleGITHUB.cmake - Attempt 1 of 3
-- Downloading StyleGITHUB.cmake - SUCCESS
-- Performing download step (git clone) for 'YCM'
-- Performing update step for 'YCM'
-- Performing configure step for 'YCM'
-- Performing uninstall step for 'YCM'
-- Performing build step for 'YCM'
-- Performing install step for 'YCM'
-- Found YCM: /home/traversaro/robotology-superbuild/build/install/share/cmake/YCM (found version "0.13.0.2-20210705.5+gitbed44ba")
-- Package YARP not found. Will be downloaded and built.
-- Package ICUB not found. Will be downloaded and built.
-- Package ICUBcontrib not found. Will be downloaded and built.
-- Package icub-models not found. Will be downloaded and built.
-- Package robots-configuration not found. Will be downloaded and built.
CMake Warning (dev) at /home/traversaro/mambaforge/envs/robsub/share/cmake-3.21/Modules/FindPackageHandleStandardArgs.cmake:438 (message):
  The package name passed to `find_package_handle_standard_args` (PkgConfig)
  does not match the name of the calling package (gazebo).  This can lead to
  problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  /home/traversaro/mambaforge/envs/robsub/share/cmake-3.21/Modules/FindPkgConfig.cmake:70 (find_package_handle_standard_args)
  /home/traversaro/mambaforge/envs/robsub/lib/cmake/gazebo/gazebo-config.cmake:62 (include)
  cmake/BuildGazeboYARPPlugins.cmake:31 (find_package)
  build/install/share/YCM/modules/FindOrBuildPackage.cmake:204 (include)
  cmake/RobotologySuperbuildLogic.cmake:19 (find_or_build_package)
  CMakeLists.txt:63 (include)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found DART: /home/traversaro/mambaforge/envs/robsub/include (Required is at least version "6.6") found components: dart
-- Found Protobuf: /home/traversaro/mambaforge/envs/robsub/lib/libprotobuf.so;-lpthread (found version "3.16.0")
-- Looking for ignition-math6 -- found version 6.8.0
-- Searching for dependencies of ignition-math6
-- Found OGRE
--   static     : OFF
--   components : Bites;HLMS;MeshLodGenerator;Overlay;Paging;Property;RTShaderSystem;Terrain;Volume
--   plugins    : Plugin_BSPSceneManager;Plugin_OctreeSceneManager;Plugin_PCZSceneManager;Plugin_ParticleFX;RenderSystem_GL;RenderSystem_GL3Plus
--   media      : /home/traversaro/mambaforge/envs/robsub/share/OGRE/Media
-- Looking for ignition-math6 -- found version 6.8.0
-- Looking for ignition-transport8 -- found version 8.1.0
-- Searching for dependencies of ignition-transport8
-- Found Protobuf: /home/traversaro/mambaforge/envs/robsub/lib/libprotobuf.so;-lpthread (found suitable version "3.16.0", minimum required is "3")
-- Checking for module 'uuid'
--   Found uuid, version 2.32.1
-- Looking for ignition-msgs5 -- found version 5.7.0
-- Searching for dependencies of ignition-msgs5
-- Looking for ignition-math6 -- found version 6.8.0
-- Checking for module 'tinyxml2'
--   Found tinyxml2, version 9.0.0
-- Looking for ignition-msgs5 -- found version 5.7.0
-- Looking for ignition-common3 -- found version 3.13.2
-- Searching for dependencies of ignition-common3
-- Looking for dlfcn.h - found
-- Looking for libdl - found
-- Searching for <ignition-common3> component [graphics]
-- Looking for ignition-common3-graphics -- found version 3.13.2
-- Searching for dependencies of ignition-common3-graphics
-- Looking for ignition-math6 -- found version 6.8.0
-- Looking for ignition-fuel_tools4 -- found version 4.4.0
-- Searching for dependencies of ignition-fuel_tools4
-- Checking for module 'jsoncpp'
--   Found jsoncpp, version
-- Checking for module 'yaml-0.1'
--   Found yaml-0.1, version 0.2.5
-- Checking for module 'libzip'
--   Found libzip, version 1.8.0
[... 1529 lines cut ...]
pkgcfg_lib_JSONCPP_jsoncpp-ADVANCED:INTERNAL=1
//ADVANCED property for variable: pkgcfg_lib_PC_CURL_curl
pkgcfg_lib_PC_CURL_curl-ADVANCED:INTERNAL=1
//ADVANCED property for variable: pkgcfg_lib_PC_FCL_ccd
pkgcfg_lib_PC_FCL_ccd-ADVANCED:INTERNAL=1
//ADVANCED property for variable: pkgcfg_lib_PC_FCL_fcl
pkgcfg_lib_PC_FCL_fcl-ADVANCED:INTERNAL=1
//ADVANCED property for variable: pkgcfg_lib_PC_FCL_m
pkgcfg_lib_PC_FCL_m-ADVANCED:INTERNAL=1
//ADVANCED property for variable: pkgcfg_lib_PC_FCL_octomap
pkgcfg_lib_PC_FCL_octomap-ADVANCED:INTERNAL=1
//ADVANCED property for variable: pkgcfg_lib_PC_FCL_octomath
pkgcfg_lib_PC_FCL_octomath-ADVANCED:INTERNAL=1
//ADVANCED property for variable: pkgcfg_lib_TINYXML2_tinyxml2
pkgcfg_lib_TINYXML2_tinyxml2-ADVANCED:INTERNAL=1
//ADVANCED property for variable: pkgcfg_lib_UUID_uuid
pkgcfg_lib_UUID_uuid-ADVANCED:INTERNAL=1
//ADVANCED property for variable: pkgcfg_lib_YAML_yaml
pkgcfg_lib_YAML_yaml-ADVANCED:INTERNAL=1
//ADVANCED property for variable: pkgcfg_lib_ZIP_zip
pkgcfg_lib_ZIP_zip-ADVANCED:INTERNAL=1
prefix_result:INTERNAL=/home/traversaro/mambaforge/envs/robsub/lib
~~~

</details>

traversaro, 2021-09-20:
Note that the relevant part of `CMakeCache.txt` is: 
~~~
//Path to a file.
DL_INCLUDE_DIRS:PATH=/home/traversaro/mambaforge/envs/robsub/x86_64-conda-linux-gnu/sysroot/usr/include

//Path to a library.
DL_LIBRARIES:FILEPATH=/home/traversaro/mambaforge/envs/robsub/x86_64-conda-linux-gnu/sysroot/usr/lib/libdl.so
~~~

traversaro, 2021-09-20:
Those files are installed by the `sysroot_linux-64` package. So I suggest to check:
* That both `${CONDA_PREFIX}/x86_64-conda-linux-gnu/sysroot/usr/lib/libdl.so` and `${CONDA_PREFIX}/x86_64-conda-linux-gnu/sysroot/usr/include/dlfcn.h` exist.
* To understand if there is something else that could interfere, I would also report the output of the `env` command inside the environment: 
~~~
(robsub) traversaro@IITICUBLAP257:~$ env
SHELL=/bin/bash
DEBUG_FFLAGS=-fopenmp -march=nocona -mtune=haswell -ftree-vectorize -fPIC -fstack-protector-strong -fno-plt -O2 -ffunction-sections -pipe -isystem /home/traversaro/mambaforge/envs/robsub/include -fopenmp -march=nocona -mtune=haswell -ftree-vectorize -fPIC -fstack-protector-all -fno-plt -Og -g -Wall -Wextra -fcheck=all -fbacktrace -fimplicit-none -fvar-tracking-assignments -ffunction-sections -pipe
GAZEBO_MASTER_URI=http://localhost:11345
GCC_RANLIB=/home/traversaro/mambaforge/envs/robsub/bin/x86_64-conda-linux-gnu-gcc-ranlib
CONDA_EXE=/home/traversaro/mambaforge/bin/conda
_CE_M=
WSL_DISTRO_NAME=Ubuntu-20.04
build_alias=x86_64-conda-linux-gnu
CMAKE_ARGS=-DCMAKE_AR=/home/traversaro/mambaforge/envs/robsub/bin/x86_64-conda-linux-gnu-ar -DCMAKE_CXX_COMPILER_AR=/home/traversaro/mambaforge/envs/robsub/bin/x86_64-conda-linux-gnu-gcc-ar -DCMAKE_C_COMPILER_AR=/home/traversaro/mambaforge/envs/robsub/bin/x86_64-conda-linux-gnu-gcc-ar -DCMAKE_RANLIB=/home/traversaro/mambaforge/envs/robsub/bin/x86_64-conda-linux-gnu-ranlib -DCMAKE_CXX_COMPILER_RANLIB=/home/traversaro/mambaforge/envs/robsub/bin/x86_64-conda-linux-gnu-gcc-ranlib -DCMAKE_C_COMPILER_RANLIB=/home/traversaro/mambaforge/envs/robsub/bin/x86_64-conda-linux-gnu-gcc-ranlib -DCMAKE_LINKER=/home/traversaro/mambaforge/envs/robsub/bin/x86_64-conda-linux-gnu-ld -DCMAKE_STRIP=/home/traversaro/mambaforge/envs/robsub/bin/x86_64-conda-linux-gnu-strip
GPROF=/home/traversaro/mambaforge/envs/robsub/bin/x86_64-conda-linux-gnu-gprof
CONDA_TOOLCHAIN_BUILD=x86_64-conda-linux-gnu
_CONDA_PYTHON_SYSCONFIGDATA_NAME=_sysconfigdata_x86_64_conda_cos6_linux_gnu
STRINGS=/home/traversaro/mambaforge/envs/robsub/bin/x86_64-conda-linux-gnu-strings
CPP=/home/traversaro/mambaforge/envs/robsub/bin/x86_64-conda-linux-gnu-cpp
FFLAGS=-fopenmp -march=nocona -mtune=haswell -ftree-vectorize -fPIC -fstack-protector-strong -fno-plt -O2 -ffunction-sections -pipe -isystem /home/traversaro/mambaforge/envs/robsub/include
NAME=IITICUBLAP257
PWD=/home/traversaro
GSETTINGS_SCHEMA_DIR=/home/traversaro/mambaforge/envs/robsub/share/glib-2.0/schemas
LOGNAME=traversaro
CONDA_PREFIX=/home/traversaro/mambaforge/envs/robsub
GSETTINGS_SCHEMA_DIR_CONDA_BACKUP=
CXX=/home/traversaro/mambaforge/envs/robsub/bin/x86_64-conda-linux-gnu-c++
CXXFLAGS=-fvisibility-inlines-hidden -std=c++17 -fmessage-length=0 -march=nocona -mtune=haswell -ftree-vectorize -fPIC -fstack-protector-strong -fno-plt -O2 -ffunction-sections -pipe -isystem /home/traversaro/mambaforge/envs/robsub/include
CONDA_TOOLCHAIN_HOST=x86_64-conda-linux-gnu
DEBUG_CXXFLAGS=-fvisibility-inlines-hidden -std=c++17 -fmessage-length=0 -march=nocona -mtune=haswell -ftree-vectorize -fPIC -fstack-protector-all -fno-plt -Og -g -Wall -Wextra -fvar-tracking-assignments -ffunction-sections -pipe -isystem /home/traversaro/mambaforge/envs/robsub/include
MOTD_SHOWN=update-motd
LDFLAGS=-Wl,-O2 -Wl,--sort-common -Wl,--as-needed -Wl,-z,relro -Wl,-z,now -Wl,--disable-new-dtags -Wl,--gc-sections -Wl,-rpath,/home/traversaro/mambaforge/envs/robsub/lib -Wl,-rpath-link,/home/traversaro/mambaforge/envs/robsub/lib -L/home/traversaro/mambaforge/envs/robsub/lib
HOME=/home/traversaro
GAZEBO_PLUGIN_PATH=/home/traversaro/mambaforge/envs/robsub/lib/gazebo-11/plugins:
LANG=C.UTF-8
WSL_INTEROP=/run/WSL/8_interop
GAZEBO_MODEL_DATABASE_URI=http://models.gazebosim.org
DEBUG_CFLAGS=-march=nocona -mtune=haswell -ftree-vectorize -fPIC -fstack-protector-all -fno-plt -Og -g -Wall -Wextra -fvar-tracking-assignments -ffunction-sections -pipe -isystem /home/traversaro/mambaforge/envs/robsub/include
LS_COLORS=rs=0:di=01;34:ln=01;36:mh=00:pi=40;33:so=01;35:do=01;35:bd=40;33;01:cd=40;33;01:or=40;31;01:mi=00:su=37;41:sg=30;43:ca=30;41:tw=30;42:ow=34;42:st=37;44:ex=01;32:*.tar=01;31:*.tgz=01;31:*.arc=01;31:*.arj=01;31:*.taz=01;31:*.lha=01;31:*.lz4=01;31:*.lzh=01;31:*.lzma=01;31:*.tlz=01;31:*.txz=01;31:*.tzo=01;31:*.t7z=01;31:*.zip=01;31:*.z=01;31:*.dz=01;31:*.gz=01;31:*.lrz=01;31:*.lz=01;31:*.lzo=01;31:*.xz=01;31:*.zst=01;31:*.tzst=01;31:*.bz2=01;31:*.bz=01;31:*.tbz=01;31:*.tbz2=01;31:*.tz=01;31:*.deb=01;31:*.rpm=01;31:*.jar=01;31:*.war=01;31:*.ear=01;31:*.sar=01;31:*.rar=01;31:*.alz=01;31:*.ace=01;31:*.zoo=01;31:*.cpio=01;31:*.7z=01;31:*.rz=01;31:*.cab=01;31:*.wim=01;31:*.swm=01;31:*.dwm=01;31:*.esd=01;31:*.jpg=01;35:*.jpeg=01;35:*.mjpg=01;35:*.mjpeg=01;35:*.gif=01;35:*.bmp=01;35:*.pbm=01;35:*.pgm=01;35:*.ppm=01;35:*.tga=01;35:*.xbm=01;35:*.xpm=01;35:*.tif=01;35:*.tiff=01;35:*.png=01;35:*.svg=01;35:*.svgz=01;35:*.mng=01;35:*.pcx=01;35:*.mov=01;35:*.mpg=01;35:*.mpeg=01;35:*.m2v=01;35:*.mkv=01;35:*.webm=01;35:*.ogm=01;35:*.mp4=01;35:*.m4v=01;35:*.mp4v=01;35:*.vob=01;35:*.qt=01;35:*.nuv=01;35:*.wmv=01;35:*.asf=01;35:*.rm=01;35:*.rmvb=01;35:*.flc=01;35:*.avi=01;35:*.fli=01;35:*.flv=01;35:*.gl=01;35:*.dl=01;35:*.xcf=01;35:*.xwd=01;35:*.yuv=01;35:*.cgm=01;35:*.emf=01;35:*.ogv=01;35:*.ogx=01;35:*.aac=00;36:*.au=00;36:*.flac=00;36:*.m4a=00;36:*.mid=00;36:*.midi=00;36:*.mka=00;36:*.mp3=00;36:*.mpc=00;36:*.ogg=00;36:*.ra=00;36:*.wav=00;36:*.oga=00;36:*.opus=00;36:*.spx=00;36:*.x [... cut ...]
CFLAGS=-march=nocona -mtune=haswell -ftree-vectorize -fPIC -fstack-protector-strong -fno-plt -O2 -ffunction-sections -pipe -isystem /home/traversaro/mambaforge/envs/robsub/include
CXXFILT=/home/traversaro/mambaforge/envs/robsub/bin/x86_64-conda-linux-gnu-c++filt
BUILD=x86_64-conda-linux-gnu
HOSTTYPE=x86_64
RANLIB=/home/traversaro/mambaforge/envs/robsub/bin/x86_64-conda-linux-gnu-ranlib
CONDA_BUILD_SYSROOT=/home/traversaro/mambaforge/envs/robsub/x86_64-conda-linux-gnu/sysroot
PROJ_NETWORK=ON
OGRE_RESOURCE_PATH=/home/traversaro/mambaforge/envs/robsub/lib/OGRE
OLDPWD=/home/traversaro/mambaforge/envs/robsub/conda-meta
_=/usr/bin/env
(robsub) traversaro@IITICUBLAP257:~$
~~~

prashanthr05, 2021-09-21:
I was a bit desperate yesterday. So I took the `apt` package manager way to install the superbuild.

Today, coming back to this issue, in a conda environment, I tried to clone and compile a fresh superbuild. And the configuration is successful. `DL_FOUND` is true.

<details>
<summary> First build configuration of superbuild </summary>

``` sh
pramadoss@iiticublap264:~/iit_ws/conda$ conda activate robsub
(robsub) pramadoss@iiticublap264:~/iit_ws/conda$ 
(robsub) pramadoss@iiticublap264:~/iit_ws/conda$ 
(robsub) pramadoss@iiticublap264:~/iit_ws/conda$ cd robotology-superbuild/
(robsub) pramadoss@iiticublap264:~/iit_ws/conda/robotology-superbuild$ ls
apt.txt       cmake           conda  LGPL.TXT   README.md  scripts
CHANGELOG.md  CMakeLists.txt  doc    packaging  releases
(robsub) pramadoss@iiticublap264:~/iit_ws/conda/robotology-superbuild$ mkdir build
(robsub) pramadoss@iiticublap264:~/iit_ws/conda/robotology-superbuild$ cd build/
(robsub) pramadoss@iiticublap264:~/iit_ws/conda/robotology-superbuild/build$ cmake ..
-- The C compiler identification is GNU 9.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /home/pramadoss/mambaforge/envs/robsub/bin/x86_64-conda-linux-gnu-cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- The CXX compiler identification is GNU 9.4.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /home/pramadoss/mambaforge/envs/robsub/bin/x86_64-conda-linux-gnu-c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- YCM not found. Bootstrapping it.
-- Downloading YCMEPHelper.cmake - Attempt 1 of 3
-- Downloading YCMEPHelper.cmake - SUCCESS
-- Downloading CMakeParseArguments.cmake - Attempt 1 of 3
-- Downloading CMakeParseArguments.cmake - SUCCESS
-- Downloading ExternalProject.cmake - Attempt 1 of 3
-- Downloading ExternalProject.cmake - SUCCESS
-- Downloading StyleGITHUB.cmake - Attempt 1 of 3
-- Downloading StyleGITHUB.cmake - SUCCESS
-- Performing download step (git clone) for 'YCM'
-- Performing update step for 'YCM'
-- Performing configure step for 'YCM'
-- Performing uninstall step for 'YCM'
-- Performing build step for 'YCM'
-- Performing install step for 'YCM'
-- Found YCM: /home/pramadoss/iit_ws/conda/robotology-superbuild/build/install/share/cmake/YCM (found version "0.13.0.2-20210705.5+gitbed44ba")
-- Package YARP not found. Will be downloaded and built.
-- Package ICUB not found. Will be downloaded and built.
-- Package ICUBcontrib not found. Will be downloaded and built.
-- Package icub-models not found. Will be downloaded and built.
-- Package robots-configuration not found. Will be downloaded and built.
CMake Warning (dev) at /home/pramadoss/mambaforge/envs/robsub/share/cmake-3.21/Modules/FindPackageHandleStandardArgs.cmake:438 (message):
  The package name passed to `find_package_handle_standard_args` (PkgConfig)
  does not match the name of the calling package (gazebo).  This can lead to
  problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  /home/pramadoss/mambaforge/envs/robsub/share/cmake-3.21/Modules/FindPkgConfig.cmake:70 (find_package_handle_standard_args)
  /home/pramadoss/mambaforge/envs/robsub/lib/cmake/gazebo/gazebo-config.cmake:62 (include)
  cmake/BuildGazeboYARPPlugins.cmake:31 (find_package)
  build/install/share/YCM/modules/FindOrBuildPackage.cmake:204 (include)
  cmake/RobotologySuperbuildLogic.cmake:19 (find_or_build_package)
  CMakeLists.txt:63 (include)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found PkgConfig: /home/pramadoss/mambaforge/envs/robsub/bin/pkg-config (found version "0.29.2") 
-- Checking for module 'bullet>=2.82'
--   Found bullet, version 3.17
-- Found fcl: /usr/include (found suitable version "0.6.1", minimum required is "0.3.2") 
-- Found assimp: /home/pramadoss/mambaforge/envs/robsub/lib/cmake/assimp-5.0/assimpConfig.cmake (found version "5.0.0") 
[... 260 lines cut ...]
DBUS_SESSION_BUS_ADDRESS=unix:path=/run/user/1000/bus
BUILD=x86_64-conda-linux-gnu
ROS_ROOT=/opt/ros/noetic/share/ros
ROS_DISTRO=noetic
RANLIB=/home/pramadoss/mambaforge/envs/robsub/bin/x86_64-conda-linux-gnu-ranlib
CONDA_BUILD_SYSROOT=/home/pramadoss/mambaforge/envs/robsub/x86_64-conda-linux-gnu/sysroot
LC_NUMERIC=it_IT.UTF-8
PROJ_NETWORK=ON
OGRE_RESOURCE_PATH=/home/pramadoss/mambaforge/envs/robsub/lib/OGRE
OLDPWD=/home/pramadoss/iit_ws/conda/robotology-superbuild
_=/usr/bin/env
```

</details>

CMakeCache.txt variables of interest:
```
//Path to a file.
DL_INCLUDE_DIRS:PATH=/home/pramadoss/mambaforge/envs/robsub/x86_64-conda-linux-gnu/sysroot/usr/include

//Path to a library.
DL_LIBRARIES:FILEPATH=/home/pramadoss/mambaforge/envs/robsub/x86_64-conda-linux-gnu/sysroot/usr/lib/libdl.so
```

Now, I am not sure what solved this problem.

prashanthr05, 2021-09-27:
I will close this issue for the moment. We may reopen it if faced again, since the source of the problem is still unresolved.

closed 2021-09-27 by prashanthr05