robotology/robotology-superbuild #895: ``yarpmotorgui`` does not start on Windows with ``conda`` dependencies

S-Dafarra, 2021-09-29:
I tried to launch ``yarpmotorgui`` but I got the following error
[image]

I also tried to run ``yarpmanager`` but nothing showed up. 

I guess there is some path missing.

@Nicogene any idea?

Nicogene, 2021-09-30:
I tried on my pc(Windows as well), and it seems that it doesn't start without any error messages.
Probably it is somehow related to #871 and #883 , unfortunately I have not a fresh conda installation, then probably I miss those fixes

S-Dafarra, 2021-09-30:
I did not understand, do you have the same issue or not?

Nicogene, 2021-09-30:
> I did not understand, do you have the same issue or not?

I cannot start it, but I don't get any message, but I don't have a fresh setup on windows, so it may be another issue.
`yarpmanager` starts without problem instead.

S-Dafarra, 2021-09-30:
Actually, I applied manually the workaround of https://github.com/robotology/robotology-superbuild/pull/883 by manually copying the file ``qt.conf`` from ``mambaforge\envs\robsub\`` to ``robotology-superbuild\build\install\bin`` and it started working.

We were using a release version of the superbuild and it worked. I guess that patch has not been released yet.

Thanks @Nicogene 

Closing

closed 2021-09-30 by S-Dafarra

traversaro, 2021-10-01:
fyi @sgiraz that could be interested in this as well

sgiraz, 2021-10-01:
Thanks @traversaro, if necessary I'll take a look at it. 🙏🏻