robotology/robotology-superbuild #896: All installation using YCMBootstrap are failing

traversaro, 2021-10-03:
See for example: https://github.com/robotology/robotology-superbuild/runs/3779570285 . 

~~~
CMake Error at build/CMakeFiles/YCMEPHelper.cmake:1411 (message):
-- Configuring incomplete, errors occurred!
  Cannot build YCM
Call Stack (most recent call first):
  cmake/YCMBootstrap.cmake:102 (ycm_bootstrap)
  CMakeLists.txt:56 (include)

See also "/home/runner/work/robotology-superbuild/robotology-superbuild/build/CMakeFiles/CMakeOutput.log".
Error: Process completed with exit code 1.
~~~

The actual problem in YCM configuration is:
~~~
-- Cannot download file https://gitlab.com/libeigen/eigen/raw/a12b8a8c75ebef312509da643424951725519348/cmake/FindEigen3.cmake
CMake Error at /home/runner/work/idyntree/idyntree/ycm/build/3rdparty/CMakeFiles/3rdparty-eigen.dir/ycm_download_cmake_FindEigen3_cmake_real.cmake:9 (file):
  file DOWNLOAD HASH mismatch

    for file: [/home/runner/work/idyntree/idyntree/ycm/build/3rdparty/CMakeFiles/3rdparty-eigen.dir/downloads/cmake/FindEigen3.cmake]
      expected hash: [5ad2b8e1ddbd9f0468b21e9d5343d05eda6b9dd1]
        actual hash: [b858bffabe01e7e4a4ef63d9933b9a96aaf5d02d]
             status: [0;"No error"].
  Retrying.

-- Cannot download file https://gitlab.com/libeigen/eigen/raw/a12b8a8c75ebef312509da643424951725519348/cmake/FindEigen3.cmake
CMake Error at /home/runner/work/idyntree/idyntree/ycm/build/3rdparty/CMakeFiles/3rdparty-eigen.dir/ycm_download_cmake_FindEigen3_cmake_real.cmake:9 (file):
  file DOWNLOAD HASH mismatch

    for file: [/home/runner/work/idyntree/idyntree/ycm/build/3rdparty/CMakeFiles/3rdparty-eigen.dir/downloads/cmake/FindEigen3.cmake]
      expected hash: [5ad2b8e1ddbd9f0468b21e9d5343d05eda6b9dd1]
        actual hash: [8bd964dfea820c9bf0b3b95a86689cd380dd8194]
             status: [0;"No error"].
  Retrying.

CMake Error at /home/runner/work/idyntree/idyntree/ycm/build/3rdparty/CMakeFiles/3rdparty-eigen.dir/ycm_download_cmake_FindEigen3_cmake.cmake:30 (message):
  Cannot download file
  https://gitlab.com/libeigen/eigen/raw/a12b8a8c75ebef312509da643424951725519348/cmake/FindEigen3.cmake

  CMake Error at
  /home/runner/work/idyntree/idyntree/ycm/build/3rdparty/CMakeFiles/3rdparty-eigen.dir/ycm_download_cmake_FindEigen3_cmake_real.cmake:9
  (file):

    file DOWNLOAD HASH mismatch

  

      for file: [/home/runner/work/idyntree/idyntree/ycm/build/3rdparty/CMakeFiles/3rdparty-eigen.dir/downloads/cmake/FindEigen3.cmake]
        expected hash: [5ad2b8e1ddbd9f0468b21e9d5343d05eda6b9dd1]
          actual hash: [ce811617ee556f66964a03e9302f17adfa5a4c43]
               status: [0;"No error"]

make[2]: *** [3rdparty/eigen3/cmake/FindEigen3.cmake] Error 1
make[1]: *** [3rdparty/CMakeFiles/3rdparty-eigen.dir/all] Error 2
make: *** [all] Error 2
3rdparty/CMakeFiles/3rdparty-eigen.dir/build.make:102: recipe for target '3rdparty/eigen3/cmake/FindEigen3.cmake' failed
CMakeFiles/Makefile2:1291: recipe for target '3rdparty/CMakeFiles/3rdparty-eigen.dir/all' failed
Makefile:133: recipe for target 'all' failed
Error: Process completed with exit code 2.
~~~

drdanz, 2021-10-05:
This should be fixed now, the issue was upstream in eigen, see robotology/ycm#387

traversaro, 2021-10-05:
Great, thanks!

closed 2021-10-05 by traversaro