robotology/robotology-superbuild #900: CI failure in Unstable branches

traversaro, 2021-10-13:
On all operating systems, the latest nigthly job revealed the following build failure:
~~~
2021-10-13T02:56:50.2083161Z [319/458] Building CXX object src/tools/depth2kin/CMakeFiles/depth2kin.dir/depth2kin_IDL.cpp.o
2021-10-13T02:56:50.2084038Z FAILED: src/tools/depth2kin/CMakeFiles/depth2kin.dir/depth2kin_IDL.cpp.o 
2021-10-13T02:56:50.2090539Z /usr/bin/c++ -DHAVE_CSTDDEF -D_USE_MATH_DEFINES -I/home/runner/work/robotology-superbuild/robotology-superbuild/build/src/ICUB/src/tools/depth2kin -I/home/runner/work/robotology-superbuild/robotology-superbuild/src/ICUB/src/tools/depth2kin/include -I/usr/include/coin -I/home/runner/work/robotology-superbuild/robotology-superbuild/src/ICUB/src/libraries/ctrlLib/include -I/home/runner/work/robotology-superbuild/robotology-superbuild/src/ICUB/src/libraries/iKin/include -I/home/runner/work/robotology-superbuild/robotology-superbuild/src/ICUB/src/libraries/optimization/include -I/home/runner/work/robotology-superbuild/robotology-superbuild/src/ICUB/src/libraries/learningMachine/include -isystem /home/runner/work/robotology-superbuild/robotology-superbuild/build/install/include -isystem /usr/include/opencv4 -O3 -DNDEBUG -fPIE -std=c++17 -MD -MT src/tools/depth2kin/CMakeFiles/depth2kin.dir/depth2kin_IDL.cpp.o -MF src/tools/depth2kin/CMakeFiles/depth2kin.dir/depth2kin_IDL.cpp.o.d -o src/tools/depth2kin/CMakeFiles/depth2kin.dir/depth2kin_IDL.cpp.o -c /home/runner/work/robotology-superbuild/robotology-superbuild/build/src/ICUB/src/tools/depth2kin/depth2kin_IDL.cpp
2021-10-13T02:56:50.2096441Z /home/runner/work/robotology-superbuild/robotology-superbuild/build/src/ICUB/src/tools/depth2kin/depth2kin_IDL.cpp:324:39: error: stray ‘\’ in program
2021-10-13T02:56:50.2097552Z   324 |         "@param type can be \\"experts\\" or \\"calibrator\\", accounting either\n"
2021-10-13T02:56:50.2098118Z       |                                       ^
2021-10-13T02:56:50.2099258Z /home/runner/work/robotology-superbuild/robotology-superbuild/build/src/ICUB/src/tools/depth2kin/depth2kin_IDL.cpp:324:40: error: stray ‘\’ in program
2021-10-13T02:56:50.2100375Z   324 |         "@param type can be \\"experts\\" or \\"calibrator\\", accounting either\n"
2021-10-13T02:56:50.2100931Z       |                                        ^
2021-10-13T02:56:50.2102067Z /home/runner/work/robotology-superbuild/robotology-superbuild/build/src/ICUB/src/tools/depth2kin/depth2kin_IDL.cpp:324:59: error: stray ‘\’ in program
2021-10-13T02:56:50.2103183Z   324 |         "@param type can be \\"experts\\" or \\"calibrator\\", accounting either\n"
2021-10-13T02:56:50.2103750Z       |                                                           ^
2021-10-13T02:56:50.2104901Z /home/runner/work/robotology-superbuild/robotology-superbuild/build/src/ICUB/src/tools/depth2kin/depth2kin_IDL.cpp:324:60: error: stray ‘\’ in program
2021-10-13T02:56:50.2106007Z   324 |         "@param type can be \\"experts\\" or \\"calibrator\\", accounting either\n"
2021-10-13T02:56:50.2106567Z       |                                                            ^
2021-10-13T02:56:50.2107715Z /home/runner/work/robotology-superbuild/robotology-superbuild/build/src/ICUB/src/tools/depth2kin/depth2kin_IDL.cpp:328:23: error: stray ‘\’ in program
2021-10-13T02:56:50.2189911Z   328 |         "\\"calibrator\\" is selected and calibration has not been performed yet.\n"
2021-10-13T02:56:50.2190496Z       |                       ^
2021-10-13T02:56:50.2191984Z /home/runner/work/robotology-superbuild/robotology-superbuild/build/src/ICUB/src/tools/depth2kin/depth2kin_IDL.cpp:328:24: error: stray ‘\’ in program
2021-10-13T02:56:50.2193146Z   328 |         "\\"calibrator\\" is selected and calibration has not been performed yet.\n"
2021-10-13T02:56:50.2193703Z       |                        ^
2021-10-13T02:56:50.2194839Z /home/runner/work/robotology-superbuild/robotology-superbuild/build/src/ICUB/src/tools/depth2kin/depth2kin_IDL.cpp:1033:31: error: stray ‘\’ in program
2021-10-13T02:56:50.2195817Z  1033 |         "@param arm is \\"left\\" or \\"right\\".\n"
2021-10-13T02:56:50.2196225Z       |                               ^
2021-10-13T02:56:50.2197373Z /home/runner/work/robotology-superbuild/robotology-superbuild/build/src/ICUB/src/tools/depth2kin/depth2kin_IDL.cpp:1033:32: error: stray ‘\’ in program
2021-10-13T02:56:50.2198354Z  1033 |         "@param arm is \\"left\\" or \\"right\\".\n"
2021-10-13T02:56:50.2198774Z       |                                ^
2021-10-13T02:56:50.2199899Z /home/runner/work/robotology-superbuild/robotology-superbuild/build/src/ICUB/src/tools/depth2kin/depth2kin_IDL.cpp:1033:46: error: stray ‘\’ in program
2021-10-13T02:56:50.2200876Z  1033 |         "@param arm is \\"left\\" or \\"right\\".\n"
2021-10-13T02:56:50.2201303Z       |                                              ^
2021-10-13T02:56:50.2202430Z /home/runner/work/robotology-superbuild/robotology-superbuild/build/src/ICUB/src/tools/depth2kin/depth2kin_IDL.cpp:1033:47: error: stray ‘\’ in program
2021-10-13T02:56:50.2203404Z  1033 |         "@param arm is \\"left\\" or \\"right\\".\n"
2021-10-13T02:56:50.2203837Z       |                                               ^
2021-10-13T02:56:50.2204960Z /home/runner/work/robotology-superbuild/robotology-superbuild/build/src/ICUB/src/tools/depth2kin/depth2kin_IDL.cpp:1095:25: error: stray ‘\’ in program
2021-10-13T02:56:50.2206174Z  1095 |         "@return \\"left\\" or \\"right\\"."
2021-10-13T02:56:50.2206576Z       |                         ^
2021-10-13T02:56:50.2207700Z /home/runner/work/robotology-superbuild/robotology-superbuild/build/src/ICUB/src/tools/depth2kin/depth2kin_IDL.cpp:1095:26: error: stray ‘\’ in program
2021-10-13T02:56:50.2208766Z  1095 |         "@return \\"left\\" or \\"right\\"."
2021-10-13T02:56:50.2209161Z       |                          ^
2021-10-13T02:56:50.2210278Z /home/runner/work/robotology-superbuild/robotology-superbuild/build/src/ICUB/src/tools/depth2kin/depth2kin_IDL.cpp:1095:40: error: stray ‘\’ in program
2021-10-13T02:56:50.2211226Z  1095 |         "@return \\"left\\" or \\"right\\"."
2021-10-13T02:56:50.2211637Z       |                                        ^
2021-10-13T02:56:50.2212757Z /home/runner/work/robotology-superbuild/robotology-superbuild/build/src/ICUB/src/tools/depth2kin/depth2kin_IDL.cpp:1095:41: error: stray ‘\’ in program
2021-10-13T02:56:50.2213712Z  1095 |         "@return \\"left\\" or \\"right\\"."
2021-10-13T02:56:50.2214118Z       |                                         ^
2021-10-13T02:56:50.2215242Z /home/runner/work/robotology-superbuild/robotology-superbuild/build/src/ICUB/src/tools/depth2kin/depth2kin_IDL.cpp:1163:16: error: stray ‘\’ in program
2021-10-13T02:56:50.2216253Z  1163 |         "\\"se3\\", \\"se3+scale\\", \\"affine\\", \\"lssvm\\".\n"
2021-10-13T02:56:50.2216693Z       |                ^
2021-10-13T02:56:50.2217800Z /home/runner/work/robotology-superbuild/robotology-superbuild/build/src/ICUB/src/tools/depth2kin/depth2kin_IDL.cpp:1163:17: error: stray ‘\’ in program
2021-10-13T02:56:50.2218798Z  1163 |         "\\"se3\\", \\"se3+scale\\", \\"affine\\", \\"lssvm\\".\n"
2021-10-13T02:56:50.2219230Z       |                 ^
2021-10-13T02:56:50.2220339Z /home/runner/work/robotology-superbuild/robotology-superbuild/build/src/ICUB/src/tools/depth2kin/depth2kin_IDL.cpp:1163:33: error: stray ‘\’ in program
2021-10-13T02:56:50.2221344Z  1163 |         "\\"se3\\", \\"se3+scale\\", \\"affine\\", \\"lssvm\\".\n"
2021-10-13T02:56:50.2221798Z       |                                 ^
2021-10-13T02:56:50.2222914Z /home/runner/work/robotology-superbuild/robotology-superbuild/build/src/ICUB/src/tools/depth2kin/depth2kin_IDL.cpp:1163:34: error: stray ‘\’ in program
2021-10-13T02:56:50.2223912Z  1163 |         "\\"se3\\", \\"se3+scale\\", \\"affine\\", \\"lssvm\\".\n"
2021-10-13T02:56:50.2224361Z       |                                  ^
2021-10-13T02:56:50.2225470Z /home/runner/work/robotology-superbuild/robotology-superbuild/build/src/ICUB/src/tools/depth2kin/depth2kin_IDL.cpp:1163:47: error: stray ‘\’ in program
2021-10-13T02:56:50.2226467Z  1163 |         "\\"se3\\", \\"se3+scale\\", \\"affine\\", \\"lssvm\\".\n"
2021-10-13T02:56:50.2226926Z       |                                               ^
2021-10-13T02:56:50.2228365Z /home/runner/work/robotology-superbuild/robotology-superbuild/build/src/ICUB/src/tools/depth2kin/depth2kin_IDL.cpp:1163:48: error: stray ‘\’ in program
[... cut ...]
2021-10-13T02:56:50.2816037Z  2707 |         "@return true/false on success/failure."
2021-10-13T02:56:50.2816481Z       |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
~~~

See https://github.com/robotology/robotology-superbuild/actions/runs/1335661437 for the latest failed job, while https://github.com/robotology/robotology-superbuild/actions/runs/1331310603 is the latest successful job. Not sure if this is expected failure (because a change in icub-main is eventually necessary due to a change in yarp master) or not. FYI @drdanz @pattacini

pattacini, 2021-10-13:
Using backquotes instead of double-quotes seems to work.
I've built the project locally and now launched a CI: https://github.com/robotology/icub-main/actions/runs/1336857899.

traversaro, 2021-10-13:
Thanks @pattacini ! It seems that also other repos (perhaps `idyntree-yarp-tools`, @S-Dafarra ) are affected (see https://github.com/robotology/robotology-superbuild/actions/runs/1337651777):
~~~
2021-10-13T14:25:47.4046517Z [13/31] Building CXX object src/modules/idyntree-yarp-visualizer/CMakeFiles/idyntree-yarp-visualizer.dir/thrifts/VisualizerCommands.cpp.o
2021-10-13T14:25:47.4049272Z FAILED: src/modules/idyntree-yarp-visualizer/CMakeFiles/idyntree-yarp-visualizer.dir/thrifts/VisualizerCommands.cpp.o 
2021-10-13T14:25:47.4057592Z /usr/bin/c++  -I/home/runner/work/robotology-superbuild/robotology-superbuild/build/src/idyntree-yarp-tools/src/modules/idyntree-yarp-visualizer -I/home/runner/work/robotology-superbuild/robotology-superbuild/src/idyntree-yarp-tools/src/lib/Utilities -isystem /usr/include/eigen3 -isystem /home/runner/work/robotology-superbuild/robotology-superbuild/build/install/include -O3 -DNDEBUG -fPIE -std=c++17 -MD -MT src/modules/idyntree-yarp-visualizer/CMakeFiles/idyntree-yarp-visualizer.dir/thrifts/VisualizerCommands.cpp.o -MF src/modules/idyntree-yarp-visualizer/CMakeFiles/idyntree-yarp-visualizer.dir/thrifts/VisualizerCommands.cpp.o.d -o src/modules/idyntree-yarp-visualizer/CMakeFiles/idyntree-yarp-visualizer.dir/thrifts/VisualizerCommands.cpp.o -c /home/runner/work/robotology-superbuild/robotology-superbuild/build/src/idyntree-yarp-tools/src/modules/idyntree-yarp-visualizer/thrifts/VisualizerCommands.cpp
2021-10-13T14:25:47.4185898Z /home/runner/work/robotology-superbuild/robotology-superbuild/build/src/idyntree-yarp-tools/src/modules/idyntree-yarp-visualizer/thrifts/VisualizerCommands.cpp:283:43: error: stray ‘\’ in program
2021-10-13T14:25:47.4187797Z   283 |         "@return A status message. \\"[ok]\\" in case of success"
2021-10-13T14:25:47.4188419Z       |                                           ^
2021-10-13T14:25:47.4190588Z /home/runner/work/robotology-superbuild/robotology-superbuild/build/src/idyntree-yarp-tools/src/modules/idyntree-yarp-visualizer/thrifts/VisualizerCommands.cpp:283:44: error: stray ‘\’ in program
2021-10-13T14:25:47.4192357Z   283 |         "@return A status message. \\"[ok]\\" in case of success"
2021-10-13T14:25:47.4192959Z       |                                            ^
2021-10-13T14:25:47.4194920Z /home/runner/work/robotology-superbuild/robotology-superbuild/build/src/idyntree-yarp-tools/src/modules/idyntree-yarp-visualizer/thrifts/VisualizerCommands.cpp:283:40: error: ‘ok’ was not declared in this scope
2021-10-13T14:25:47.4196745Z   283 |         "@return A status message. \\"[ok]\\" in case of success"
2021-10-13T14:25:47.4197343Z       |                                        ^~
2021-10-13T14:25:47.4199392Z /home/runner/work/robotology-superbuild/robotology-superbuild/build/src/idyntree-yarp-tools/src/modules/idyntree-yarp-visualizer/thrifts/VisualizerCommands.cpp:283:45: error: expected ‘}’ before string constant
2021-10-13T14:25:47.4201225Z   283 |         "@return A status message. \\"[ok]\\" in case of success"
2021-10-13T14:25:47.4201839Z       |                                             ^~~~~~~~~~~~~~~~~~~~~
2021-10-13T14:25:47.4203717Z /home/runner/work/robotology-superbuild/robotology-superbuild/build/src/idyntree-yarp-tools/src/modules/idyntree-yarp-visualizer/thrifts/VisualizerCommands.cpp:281:40: note: to match this ‘{’
2021-10-13T14:25:47.4205366Z   281 |     static constexpr const char* s_help{
2021-10-13T14:25:47.4205917Z       |                                        ^
2021-10-13T14:25:47.4207964Z /home/runner/work/robotology-superbuild/robotology-superbuild/build/src/idyntree-yarp-tools/src/modules/idyntree-yarp-visualizer/thrifts/VisualizerCommands.cpp:283:42: error: expected ‘;’ at end of member declaration
2021-10-13T14:25:47.4209806Z   283 |         "@return A status message. \\"[ok]\\" in case of success"
2021-10-13T14:25:47.4210402Z       |                                          ^
2021-10-13T14:25:47.4210832Z       |                                           ;
2021-10-13T14:25:47.4212951Z /home/runner/work/robotology-superbuild/robotology-superbuild/build/src/idyntree-yarp-tools/src/modules/idyntree-yarp-visualizer/thrifts/VisualizerCommands.cpp:283:45: error: expected unqualified-id before string constant
2021-10-13T14:25:47.4214830Z   283 |         "@return A status message. \\"[ok]\\" in case of success"
2021-10-13T14:25:47.4215443Z       |                                             ^~~~~~~~~~~~~~~~~~~~~
2021-10-13T14:25:47.4217766Z /home/runner/work/robotology-superbuild/robotology-superbuild/build/src/idyntree-yarp-tools/src/modules/idyntree-yarp-visualizer/thrifts/VisualizerCommands.cpp:285:1: error: expected declaration before ‘}’ token
2021-10-13T14:25:47.4219377Z   285 | };
2021-10-13T14:25:47.4219728Z       | ^
~~~

So I guess we can keep this issue open, even if I guess that at least `ROBOTOLOGY_ENABLE_CORE` is now fixed. We can eventually open an issue for this regression in YARP itself.

traversaro, 2021-10-16:
For the time being, I added a workaround in https://github.com/robotology/robotology-superbuild/pull/902 to use an older YARP commit. @drdanz @randaz81 if you think there is a YARP regression for which it is worth to open a YARP issue please let me know, thanks!

traversaro, 2021-11-23:
This should be fixed by https://github.com/robotology/idyntree-yarp-tools/pull/20 , even because it is necessary to fix the latest CI failure (see P2 in https://github.com/robotology/robotology-superbuild/issues/925).

traversaro, 2021-11-23:
> This should be fixed by [robotology/idyntree-yarp-tools#20](https://github.com/robotology/idyntree-yarp-tools/pull/20) , even because it is necessary to fix the latest CI failure (see P2 in #925).

Apparently also https://github.com/robotology/whole-body-estimators/pull/131 was necessary.

traversaro, 2021-11-23:
As I plan to solve this issue on the robotology-superbuild side, I opened https://github.com/robotology/yarp/issues/2768 to track this problem at the YARP level, at least to document it for users that may encounter it.

closed 2021-11-25 by traversaro