robotology/robotology-superbuild #923: Compiling error with dependencies from conda-forge on Manjaro

HosameldinMohamed, 2021-11-19:
I am trying to install the superbuild on Manjaro. But I get an error:

Following [the documentation](https://github.com/robotology/robotology-superbuild/blob/master/doc/conda-forge.md#source-installation). Here's what I have tried.

```
mamba create -n robTest
conda activate robTest
mamba install -c conda-forge cmake compilers make ninja pkg-config
mamba install -c conda-forge ace asio assimp boost eigen gazebo glew glfw gsl ipopt irrlicht libjpeg-turbo libmatio libode libxml2 nlohmann_json opencv pkg-config portaudio qt sdl sdl2 sqlite tinyxml spdlog
mamba install -c conda-forge bash-completion expat-cos6-x86_64 freeglut libdc1394 libselinux-cos6-x86_64 libxau-cos6-x86_64 libxcb-cos6-x86_64 libxdamage-cos6-x86_64 libxext-cos6-x86_64 libxfixes-cos6-x86_64 libxxf86vm-cos6-x86_64 mesalib mesa-libgl-cos6-x86_64
```

Then cloned and build `robotology-superbuild`. I only disabled the option `ROBOTOLOGY_USES_GAZEBO` in cmake.

And here's the error I am getting:
```
[ 28%] Generating code from ../../../extern/ros/geometry2/tf2_msgs/srv/FrameGraph.srv
/home/hosameldin/Github/robotology-superbuild/src/YARP/src/carriers/portmonitor_carrier/lua/MonitorLua.cpp:9:10: fatal error: lua.hpp: No such file or directory
    9 | #include <lua.hpp>
      |          ^~~~~~~~~
compilation terminated.
[ 28%] Building CXX object src/libYARP_serversql/src/CMakeFiles/YARP_serversql.dir/yarp/serversql/impl/ConnectManager.cpp.o
make[5]: *** [src/carriers/portmonitor_carrier/CMakeFiles/yarp_portmonitor.dir/build.make:151: src/carriers/portmonitor_carrier/CMakeFiles/yarp_portmonitor.dir/lua/MonitorLua.cpp.o] Error 1
make[5]: *** Waiting for unfinished jobs....
[ 28%] Building CXX object src/libYARP_serversql/src/CMakeFiles/YARP_serversql.dir/yarp/serversql/impl/ConnectThread.cpp.o
In file included from /home/hosameldin/Github/robotology-superbuild/src/YARP/src/carriers/portmonitor_carrier/lua/MonitorLua.h:13,
                 from /home/hosameldin/Github/robotology-superbuild/src/YARP/src/carriers/portmonitor_carrier/MonitorBinding.cpp:11:
/home/hosameldin/Github/robotology-superbuild/build/src/YARP/src/carriers/portmonitor_carrier/lua/src_gen/swigluarun.h:736:10: fatal error: lua.h: No such file or directory
  736 | #include "lua.h"
      |          ^~~~~~~
compilation terminated.
```

I found the package `lua` installed in the system with the version `5.4.3-1` 

But also, `lua` is installed with `conda-forge`, here's the entry from `mamba list`:

```
lua                       5.4.3                h78e2651_2    conda-forge
```

<details>
<summary>My complete `conda list` </summary>

```
# packages in environment at /home/hosameldin/mambaforge:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                       1_gnu    conda-forge
brotlipy                  0.7.0           py39h3811e60_1001    conda-forge
bzip2                     1.0.8                h7f98852_4    conda-forge
c-ares                    1.17.1               h7f98852_1    conda-forge
ca-certificates           2021.10.8            ha878542_0    conda-forge
certifi                   2021.10.8        py39hf3d152e_1    conda-forge
cffi                      1.14.6           py39he32792d_0    conda-forge
chardet                   4.0.0            py39hf3d152e_1    conda-forge
charset-normalizer        2.0.0              pyhd8ed1ab_0    conda-forge
colorama                  0.4.4              pyh9f0ad1d_0    conda-forge
conda                     4.10.3           py39hf3d152e_0    conda-forge
conda-package-handling    1.7.3            py39h3811e60_0    conda-forge
cryptography              3.4.7            py39hbca0aa6_0    conda-forge
icu                       68.1                 h58526e2_0    conda-forge
idna                      3.1                pyhd3deb0d_0    conda-forge
krb5                      1.19.2               hcc1bbae_0    conda-forge
ld_impl_linux-64          2.36.1               hea4e1c9_2    conda-forge
libarchive                3.5.1                hccf745f_2    conda-forge
libcurl                   7.78.0               h2574ce0_0    conda-forge
libedit                   3.1.20191231         he28a2e2_2    conda-forge
libev                     4.33                 h516909a_1    conda-forge
libffi                    3.3                  h58526e2_2    conda-forge
libgcc-ng                 11.1.0               hc902ee8_6    conda-forge
libgomp                   11.1.0               hc902ee8_6    conda-forge
libiconv                  1.16                 h516909a_0    conda-forge
libnghttp2                1.43.0               h812cca2_0    conda-forge
libsolv                   0.7.19               h780b84a_5    conda-forge
libssh2                   1.9.0                ha56f1ee_6    conda-forge
libstdcxx-ng              11.1.0               h56837e0_6    conda-forge
libxml2                   2.9.12               h72842e0_0    conda-forge
lua                       5.4.3                h78e2651_2    conda-forge
lz4-c                     1.9.3                h9c3ff4c_1    conda-forge
lzo                       2.10              h516909a_1000    conda-forge
mamba                     0.15.2           py39h951de11_0    conda-forge
ncurses                   6.2                  h58526e2_4    conda-forge
openssl                   1.1.1l               h7f98852_0    conda-forge
pip                       21.2.2             pyhd8ed1ab_0    conda-forge
pycosat                   0.6.3           py39h3811e60_1006    conda-forge
pycparser                 2.20               pyh9f0ad1d_2    conda-forge
pyopenssl                 20.0.1             pyhd8ed1ab_0    conda-forge
pysocks                   1.7.1            py39hf3d152e_3    conda-forge
python                    3.9.6           h49503c6_1_cpython    conda-forge
python_abi                3.9                      2_cp39    conda-forge
readline                  8.1                  h46c0cb4_0    conda-forge
reproc                    14.2.1               h36c2ea0_0    conda-forge
reproc-cpp                14.2.1               h58526e2_0    conda-forge
requests                  2.26.0             pyhd8ed1ab_0    conda-forge
ruamel_yaml               0.15.80         py39h3811e60_1004    conda-forge
setuptools                49.6.0           py39hf3d152e_3    conda-forge
six                       1.16.0             pyh6c4a22f_0    conda-forge
sqlite                    3.36.0               h9cd32fc_0    conda-forge
tk                        8.6.10               h21135ba_1    conda-forge
tqdm                      4.62.0             pyhd8ed1ab_0    conda-forge
tzdata                    2021a                he74cb21_1    conda-forge
urllib3                   1.26.6             pyhd8ed1ab_0    conda-forge
wheel                     0.36.2             pyhd3deb0d_0    conda-forge
xz                        5.2.5                h516909a_1    conda-forge
yaml                      0.2.5                h516909a_0    conda-forge
zlib                      1.2.11            h516909a_1010    conda-forge
zstd                      1.5.0                ha95c52a_0    conda-forge
```

</details>

traversaro, 2021-11-19:
Interesting! Can you attach the `CMakeCache.txt` in `build` and in `build/src/YARP`?

HosameldinMohamed, 2021-11-19:
> Interesting! Can you attach the `CMakeCache.txt` in `build` and in `build/src/YARP`?

In `build`:
[CMakeCache.txt](https://github.com/robotology/robotology-superbuild/files/7572431/CMakeCache.txt)

In `build/src/YARP`:
[CMakeCache.txt](https://github.com/robotology/robotology-superbuild/files/7572433/CMakeCache.txt)

traversaro, 2021-11-19:
From YARP's CMAke cache, I see: 
~~~
//Path to a file.
LUA_INCLUDE_DIR:PATH=/usr/include

//Path to a library.
LUA_LIBRARY:FILEPATH=/usr/lib/liblua5.4.so

//Path to a library.
LUA_MATH_LIBRARY:FILEPATH=/home/hosameldin/mambaforge/envs/robTest/x86_64-conda-linux-gnu/sysroot/usr/lib/libm.so
~~~

So it seems that for some reason YARP's FindLua (or CMake one?) finds the Lua from the system rather then the one in conda. A simple workaround is to use `YARP_USE_Lua` option in YARP build to `OFF`, but this is just a workaround.

traversaro, 2021-11-19:
Question: are you sure that `lua` in conda was not installed **after** the build of the robotology-superbuild? Just in case, can you try to create a new build of robotology-superbuild?

HosameldinMohamed, 2021-11-19:
> Question: are you sure that `lua` in conda was not installed **after** the build of the robotology-superbuild? Just in case, can you try to create a new build of robotology-superbuild?

not sure, I'll start with a new buil to verify.

HosameldinMohamed, 2021-11-19:
I created a new build and it compiled correctly!

But the YARP CMake cache contents you posted above https://github.com/robotology/robotology-superbuild/issues/923#issuecomment-974338410 are a bit different! It seems it decided to take all the files from the system!

```
//Path to a file.
LUA_INCLUDE_DIR:PATH=/usr/include

//Path to a library.
LUA_LIBRARY:FILEPATH=/usr/lib/liblua5.4.so

//Path to a library.
LUA_MATH_LIBRARY:FILEPATH=/usr/lib/libm.so
```

traversaro, 2021-11-19:
Are you sure that you run `cmake` in a terminal where you activated the robTest environment?

HosameldinMohamed, 2021-11-19:
> Are you sure that you run `cmake` in a terminal where you activated the robTest environment?

Actually no! I just double checked! my bad.

If I activate the robTest environment I get the same error and the same [contents of YARP CMake cache file](https://github.com/robotology/robotology-superbuild/issues/923#issuecomment-974352498).

HosameldinMohamed, 2021-11-19:
`lua.hpp` exsists within `robTest` only in the location: `/home/hosameldin/mambaforge/envs/robTest/include/bullet/ThirdPartyLibs/lua-5.2.3/src/`

While `liblua5.4.so` or `liblua.so.5.4.3` doesn't exist within `robTest`

traversaro, 2021-11-19:
That is quite strange, are you sure that you always run `cmake` from a clean build and always from inside the `robTest` environment?

traversaro, 2021-11-19:
In any case, if you have the system at IIT next week I can check in person.

HosameldinMohamed, 2021-11-19:
> That is quite strange, are you sure that you always run `cmake` from a clean build and always from inside the `robTest` environment?

Yes, I made sure of that this time..

> In any case, if you have the system at IIT next week I can check in person.

That's great! thanks.

traversaro, 2021-11-23:
We debuggeded the problem with @HosameldinMohamed and found some interesting details.

## The problem

The problem is that on Manjaro the conda's CMake is able to find the Lua in the system, and so YARP's automatically enables the Lua dependency. However, then the compilation fails as `LUA_INCLUDE_DIR` is  `/usr/include`, that CMake does not pass to the compiler as it is assumed that it is a default include path, but this is not true for the conda compiler. 

## Why it does not apper on Debian/Ubuntu?

In Debian/Ubuntu, the lua library is installed in `/usr/lib/x86_64-linux-gnu` (see https://packages.ubuntu.com/focal/amd64/liblua5.3-0/filelist), while in Manjaro it is installed in `/usr/lib`. The conda's CMake is able to find the Manjaro's Lua as `/usr` is part of the `CMAKE_SYSTEM_PREFIX_PATH`, but is not able to find the apt Lua as the `CMAKE_LIBRARY_ARCHITECTURE` on apt's CMake is  `x86_64-linux-gnu`, while it is empty for both Manjaro and conda.

## Workaround?

Just install lua on conda before the build.

HosameldinMohamed, 2021-11-23:
> Just install lua on conda before the build.

Thanks @traversaro! I can open a PR to update the documentation.

traversaro, 2021-11-23:
> > Just install lua on conda before the build.
> 
> Thanks @traversaro! I can open a PR to update the documentation.

Thanks! Just adding lua to conda dependecies in docs and CI is probably enough.

traversaro, 2022-01-19:
> > > Just install lua on conda before the build.
> > 
> > 
> > Thanks @traversaro! I can open a PR to update the documentation.
> 
> Thanks! Just adding lua to conda dependecies in docs and CI is probably enough.

Ok, let's close it!

closed 2022-01-19 by traversaro