robotology/robotology-superbuild #933: Report on the robotology software installation procedure

DanielePucci, 2021-11-27:
Hi all, today I decided to (re-)install the `robotology-superbuild`. I honestly acknowledge that it is a long time since I last installed it being pretty much busy with other businesses over the last two years. So, thank you very much for all the effort and continuous improvement.  

However, I wanted to report my experience and feelings while trying to install the software. On the one hand, I want to apologise in advance if I missed something trivial during the installation procedure; on the other hand, I believe that I could be a good use-case of users that are not up-to-date with the technical details. For the latter reason, I decided to write this report: I think it is worth the risk of appearing shallow or not updated. 

During the attempt at installing the software, I experienced many feelings that I also wanted to report. Probably, how people feel has an impact on how quick they drop the attempt of installing our software, and we would like to avoid this as much as we can. I categorise these feelings with keywords, but I associated them with emoticons, so during the report you can quickly see how I felt:
1. `I-am-dumb` :anguished:
2. `This-is-super-cool` :partying_face:
3. `Is-this-really-necessary?` :thinking:
4. `I-am-scared` :worried:
5. `I-am-bored` :sleeping:
6. `I-give-up` :tired_face:

When I have, I also wanted to provide suggestions for improvements - clearly, these are suggestions that you can decide to rule out. In these cases, I use 

:yellow_circle:   Suggestion

To provide the reader with a detailed and sincere feedback, I go through the robotology-superbuild README of [this commit](https://github.com/robotology/robotology-superbuild/commit/67d2d79b6a7b5cdab9f2d8815dfe6b1bcd3cb2eb) (the latest to date). So, let's get started.

> # robotology-superbuild
> 
> This is a meta repository (so-called "superbuild") that uses [CMake](https://cmake.org/) and [YCM](https://github.com/robotology/ycm)

At this point, I clicked on [CMake](https://cmake.org/) and [YCM](https://github.com/robotology/ycm). I knew the basics, but I clicked and starting looking at these pages randomly. This is fine, but the user is already loosing time on something that - at the end - he does not want to do: he/she just wants to install software. You may think: it is your fault if you clicked. Yeah, but to help users avoid loosing time and going to the point, we may

:yellow_circle:  Add a `Reference` section at the end of the README, and we can cite sources with something like `[1, 2]`, where 1 and 2 are not hyperlinked there, but below in the `Reference` section. By doing so, we may help the reader keep the focus. 

After thinking better, I am not sure if adding the details that the superbuild uses CMAKE and YCM is necessary here :thinking: At the end, at this point we are not suggesting the reader to do some action (i.e. check if you installed CMAKE) but just saying additional information that it is not really clear to me the purpose of (at this point specifically). 

> to automatically download and compile software developed in the robotology GitHub organization

I thought, this is super cool, three clicks and I am done :partying_face:

> such as the YARP middleware or software used to run the iCub humanoid robot.

If people do not know YARP, or they want to know it better, they may start googling for it :thinking:. Furthermore, since YARP is basically necessary to run iCub, probably

:yellow_circle: we may just leave the sentence `such as the software used to run the iCub humanoid robot.`

> [CMake](https://cmake.org/) is an open-source

The hyperlink (twice in two paragraphs) is really suggesting the reader: "hey, you gotta click if you did not do it before, and check the amazing site of CMAKE out!" 

:yellow_circle: I would remove the hyperlink. Sincerely, however, I would remove the entire paragraph below

> [CMake](https://cmake.org/) is an open-source, cross-platform family of tools designed to build, test and package software.
> A [YCM Superbuild](http://robotology.github.io/ycm/gh-pages/git-master/index.html#superbuild) is a CMake project whose only goal is to download and build several other projects.
> If you are familiar with ROS, it is something similar to [catkin](http://wiki.ros.org/catkin/workspaces) or [colcon workspace](https://colcon.readthedocs.io/en/released/user/quick-start.html), but using pure CMake for portability reasons and for customizing the build via CMake options. Furthermore, the `robotology-superbuild` also contains some infrastructure to build **binaries** of the contained projects for some platforms. 

I felt that the documentation really wanted to inform me about CMake and YCM. But really wanted me. On the one hand, I was not understanding the purpose of these information :thinking: on the other hand, I was not really interested in these details, so I perceived a pedantic will of the documentation. As a consequence, I started interpolating :sleeping:

> You can read more about the superbuild concept in [YCM documentation](http://robotology.github.io/ycm/gh-pages/latest/index.html) or in the [related IRC paper](http://lornat75.github.io/papers/2018/domenichelli-irc.pdf).

Probably this was me, but the will of documentation of telling me about YCM here got a little disturbing. It was like someone telling me repeatedly to read about additional details. So, I started and I opened these pages. In a second, I was scared since the amount of things to read was a lot :worried: And I still did not understand if it was necessary to read all this stuff :anguished: : I just wanted to install the software, as probably most users would like to.

> | System  | Continuous Integration Status |
> |:------:|:------:|
> |  Linux/macOS/Windows  |  [image]     |

 :partying_face: :partying_face: :partying_face:

> Table of Contents
> =================
>   * [Superbuild](#superbuild)
>   * [Binary Installation](#binary-installation)
>   * [Source Installation](#source-installation)
>     * [Clone the repo](#clone-the-repo)
>     * [Debian/Ubuntu Linux with dependencies provided by apt](#linux-from-source-with-dependencies-provided-by-apt)
>     * [Linux, macOS or Windows with dependencies provided by conda-forge](#linux-macos-or-windows-from-source-with-dependencies-provided-by-conda-forge)
>     * [Update](#update)
>   * [FAQs](#faqs)
[... 550 lines cut ...]
#     $ conda deactivate
~~~ 

So, at this point I was just doing stuff quickly and trying to get it as soon as possible. I was starving for a list of actions to execute one after the other without narrative

> Once you created the `robotologyenv` environment, you can "activate" it for the current terminal (i.e. make sure that the  installed packages can be found) by the command:
> ~~~
> mamba activate robotologyenv
> ~~~

I launched the command and I obtained
~~~ 
danielepucci@Danieles-MBP-2 ~ % mamba activate robotologyenv

Run 'mamba init' to be able to run mamba activate/deactivate
and start a new shell session. Or use conda to activate/deactivate.

    $ conda activate robotologyenv
~~~

So it seems that I had to run `mamba init`. I did it, and then I re-launched `mamba activate robotologyenv`, but I was not sure about what was going on. 

:yellow_circle: To improve the awareness of the user, it would be better to add the expected outcomes of the commands

At this point, I gave up with the binary installation :tired_face:.

traversaro, 2021-11-29:
> At this point, I gave up with the binary installation: it was probably three hours that I was on the task and I could no longer dedicate time to move on although I had the feeling of being close. I hope to have more time in the next days to continue the task. I thought, however, that already this preliminary user-expereince report may give preliminary inputs on how to improve the documentation.

First of all, thanks for the detailed report! This is gold. Anyone working too much with robotology-superbuild or the rest of our software stack will be inevitable be affected by the so-called [expert blindness](https://www.warc.com/newsandopinion/opinion/expert_blindness_the_silent_threat_to_success/3082). If for you installing the software takes such as long time, that is essentially a bug exactly like any other software bug that could affect the robotology-superbuild, so it is important to try to fix it! 

> I had the feeling of being close.

Indeed! Just to clarify, at this point you could install any library you are interested (I am mentioning just a few MATLAB-library you could be interested in, see https://anaconda.org/robotology/repo/files for more available packages) in with:
~~~
mamba activate robotologyenv
mamba install idyntree-matlab-bindings wb-toolbox osqp-matlab casadi-matlab-bindings gazebo-yarp-plugins icub-models  matlab-whole-body-simulator 
~~~

Then you can just open MATLAB in a termainal where you called `mamba activate robotologyenv` environment before, and all the MATLAB and Simulink libraries will be available.

I will address the rest of the report point by point in future comments.

traversaro, 2021-11-29:
## P1: Introduction section

> > # robotology-superbuild
> > [..] content of the introduction
>
> Probably this was me, but the will of documentation of telling me about YCM here got a little disturbing. It was like someone telling me repeatedly to read about additional details. So, I started and I opened these pages. In a second, I was scared since the amount of things to read was a lot :worried: And I still did not understand if it was necessary to read all this stuff :anguished: : I just wanted to install the software, as probably most users would like to.

All those details were probably there when the superbuild was only meant to be installed by source and not by binary installation (and then, you **really** needed to know what CMake is!) and to properly acknoledge YCM without which all the machinery of the superbuild would not be possible.

Proposed Action:
* Simplify this section, moving the necessary CMake details in the "building from source" section and the YCM part in a "Credits" or "Acknowledgments" section still in the README, but outside of the installation path.

traversaro, 2021-11-29:
## P2: Superbuild section

Leaving this section before "Binary Installation" was probably an error done in https://github.com/robotology/robotology-superbuild/pull/852, especially until we have an easy way (an documentable in a eay way) to install a specific distro with conda (see https://github.com/robotology/robotology-superbuild/issues/727). For the time being, we can just move this section under "source installation".

traversaro, 2021-11-29:
Ah, I just realized that I am missing a fundemental information here. @DanielePucci  Why did you wanted to install the robotology-superbuild/related software? What did you wanted to do?

DanielePucci, 2021-12-31:
Hi @traversaro, sorry for taking so long to answer: I wanted to reserve some time to this task, and the Christmas break helped in this sense.

> First of all, thanks for the detailed report! This is gold. Anyone working too much with robotology-superbuild or the rest of our software stack will be inevitable be affected by the so-called [...]

Thank you for your understanding, the report wanted to be a different standpoint on something that - for sure - is working out for many of the experts.

> On the other hand, the documentation is meant to target different kind of users with different backgrounds, so we try reach the best trade-off of helping all of them, without scaring any of them too much.

Sure, this is also understandable: finding a trade-off for sure is not an easy task.

> Indeed! Just to clarify, at this point you could install any library you are interested (I am mentioning just a few MATLAB-library you could be interested in, see [...]

This is super cool, I did not realise I was that close! I did it and it seems that it worked out although I am not sure that the text, e.g., 
~~~
- nothing provides requested idyntree-matlab-bindings
~~~
is actually saying that it went fine - see below for details.

<details>
  <summary>Click to see the output details (sorry, I mistakenly cleaned the terminal so the following is the output of the second instance of the commands above)</summary>

~~~

danielepucci@Danieles-MBP-2 ~ % mamba activate robotologyenv

(robotologyenv) danielepucci@Danieles-MBP-2 ~ % mamba install idyntree-matlab-bindings wb-toolbox osqp-matlab casadi-matlab-bindings gazebo-yarp-plugins icub-models  matlab-whole-body-simulator

                  __    __    __    __
                 /  \  /  \  /  \  /  \
                /    \/    \/    \/    \
███████████████/  /██/  /██/  /██/  /████████████████████████
              /  / \   / \   / \   / \  \____
             /  /   \_/   \_/   \_/   \    o \__,
            / _/                       \_____/  `
            |/
        ███╗   ███╗ █████╗ ███╗   ███╗██████╗  █████╗
        ████╗ ████║██╔══██╗████╗ ████║██╔══██╗██╔══██╗
        ██╔████╔██║███████║██╔████╔██║██████╔╝███████║
        ██║╚██╔╝██║██╔══██║██║╚██╔╝██║██╔══██╗██╔══██║
        ██║ ╚═╝ ██║██║  ██║██║ ╚═╝ ██║██████╔╝██║  ██║
        ╚═╝     ╚═╝╚═╝  ╚═╝╚═╝     ╚═╝╚═════╝ ╚═╝  ╚═╝

        mamba (0.16.0) supported by @QuantStack

        GitHub:  https://github.com/mamba-org/mamba
        Twitter: https://twitter.com/QuantStack

█████████████████████████████████████████████████████████████

Looking for: ['idyntree-matlab-bindings', 'wb-toolbox', 'osqp-matlab', 'casadi-matlab-bindings', 'gazebo-yarp-plugins', 'icub-models', 'matlab-whole-body-simulator']

conda-forge/osx-64       Using cache
conda-forge/noarch       Using cache
Encountered problems while solving:
  - nothing provides requested idyntree-matlab-bindings
  - nothing provides requested wb-toolbox
  - nothing provides requested osqp-matlab
  - nothing provides requested casadi-matlab-bindings
  - nothing provides requested gazebo-yarp-plugins
  - nothing provides requested icub-models
  - nothing provides requested matlab-whole-body-simulator
~~~
</details>

> Then you can just open MATLAB in a termainal where you called `mamba activate robotologyenv` environment before, and all the MATLAB and Simulink libraries will be available.

I did this but although I had matlab installed, it seems that I do not have the command `matlab` -probably, I should make an alias. For this reason, I did in the terminal:

~~~
(robotologyenv) danielepucci@Danieles-MBP-2 ~ % open -a MATLAB_R2021a
~~~

Once opened, I opened Simulink by clicking on the button. After that, I opened the Simulink library and I looked for, e.g. Whole-Body Toolbox, but it seems to me that I did not find any. So, probably I am missing something 

[image]
[image]

> I will address the rest of the report point by point in future comments. I will number all the different points/aspects to track them easily.

Great!

DanielePucci, 2021-12-31:
> All those details were probably there when the superbuild was only meant to be installed by source and not by binary installation (and then, you really needed to know what CMake is!) and to properly acknoledge YCM without which all the machinery of the superbuild would not be possible. 

I see, it made sense back in time I would say

> To be perfectly honest, a bit of the focus on the paper there is to make sure that people properly attribute the paper if they use the superbuild in academic works, but probably it is metter to explicitly specify that.

Roger, I also agree that it would be better to add the explicit mention to that even in the beginning, like a short section called *How to cite robotology-superbiuld* or something like that 

> Proposed Action:
> 
> * Transform some of links in "references" to make clear that are not compulsory to read.
> * Simplify this section, moving the necessary CMake details in the "building from source" section and the YCM part in a "Credits" or "Acknowledgments" section still in the README, but outside of the installation path.
> * For the remaining sentence, follow the micro observations in the repo (if they make sense).

These made sense to me

DanielePucci, 2021-12-31:
> Can you check if #932 improves this aspect?

It is much better @traversaro, I just approved the PR. What I still notice - but maybe this is just me - is that nesting several READMEs is not helping the user to know where he is. For instance, what I would suggest  is to 
1. move all the details in https://github.com/robotology/robotology-superbuild/blob/simplify_conda_install/doc/conda-forge.md to the README section https://github.com/robotology/robotology-superbuild#binary-installation
or
2. move the entire README section https://github.com/robotology/robotology-superbuild#binary-installation in in https://github.com/robotology/robotology-superbuild/blob/simplify_conda_install/doc/conda-forge.md if the README https://github.com/robotology/robotology-superbuild/blob/master/README.md risks to become too long

DanielePucci, 2021-12-31:
> ## P2: Superbuild section
> 
> Leaving this section before "Binary Installation" was probably an error done in https://github.com/robotology/robotology-superbuild/pull/852, especially until we have an easy way (an documentable in a eay way) to install a specific distro with conda (see https://github.com/robotology/robotology-superbuild/issues/727). For the time being, we can just move this section under "source installation".
> 
> Action point:
> Move "superbuild section" under "Build from source".

Probably you meant "below" or "after"?

DanielePucci, 2021-12-31:
> Ah, I just realized that I am missing a fundemental information here. @DanielePucci Why did you wanted to install the robotology-superbuild/related software? What did you wanted to do?

My original plan was to install the software for making the [Yoga++](https://youtu.be/9XRI4BeXN78) with iCub 3 - although the other day passing by the lab, I just saw that @lrapetti just did it in a couple of hours 😭. In the hindsight, I could have used the [one line installation](https://github.com/ami-iit/matlab-whole-body-simulator#one-line-installation) made by @nunoguedelha, which I actually tested -- see https://github.com/robotology/robotology-superbuild/issues/707#issuecomment-855757836 and https://github.com/ami-iit/element_software-engineering/issues/32

By thinking about it, I do not know if it would make sense to somehow profile the users, even with a list of questions like you just did in the beginning of the README. And then, if the user somehow finds the question for himself/herself on the list, just click on it and that opens a set of actions to follow to install the software other users like him/her would. 

However, I wanted also to have that old feeling of having the entire `robotology-superbuild` installed on the laptop, so I was pushed to give it a try and provide you with a feedback above.

DanielePucci, 2023-04-22:
Today I followed the installation procedure for the sources of robotology-superbuild, the software package we use extensively in our research line @robotology/iit-artificial-mechanical-intelligence.

I wanted to come back to this since I received a new laptop and I didn't even have the brew nor the git installed, so it seemed to me I could be a good testcase of someone not aligned with the recent development or of someone that just arrived in the lab. 

I have to say that the improvements of the installation procedure was substantial: it is now clear, linear, and straight to the point. So, thank you very much @traversaro for having taken into account the comments above.

By considering the effective time I dealt with the installation, in about 20 minutes I completed the task, from installing brew, git, the robotology-superbuild, and launching Gazebo. 

Being on Mac, the only setback I had was due to https://github.com/robotology/robotology-superbuild/issues/916, both when installing the dependencies and during the compilation. Thanks to the @traversaro quick help, I could quickly overcome the issue during the compilation.

To further simplify the installation procedure for Mac users, I opened https://github.com/robotology/robotology-superbuild/pull/1394 that details the passages that I performed when installing the software.

At this point, I close the issue. Thank you all.

CC
@Gianlucamilani @evalli-iit

closed 2023-04-22 by DanielePucci