robotology/robotology-superbuild #944: CI Failure Mid December 2021 Debian Testing Bookworm apt

traversaro, 2021-12-08:
The Debian Testing (currently Bookworm) jobs are now failing because there is not anymore the `gazebo` package (see https://github.com/robotology/robotology-superbuild/actions/runs/1552185107):
~~~
2021-12-08T02:16:47.6912920Z lsb_dist: debian
2021-12-08T02:16:47.6913310Z dist_version: bookworm
2021-12-08T02:16:48.1761219Z Reading package lists...
2021-12-08T02:16:48.3045048Z Building dependency tree...
2021-12-08T02:16:48.3052917Z Reading state information...
2021-12-08T02:16:48.3108997Z E: Unable to locate package libgazebo-dev
~~~

This is not a big problem, as we only use the Debian Testing CI apt to catch early problems that could eventually appear on Ubuntu, and we could simply disable Gazebo testing there. By looking in https://tracker.debian.org/pkg/gazebo, it seems that the package was removed from testing today: https://tracker.debian.org/news/1284683/gazebo-removed-from-testing/ . fyi @j-rivero (even if I guess you are already aware of this)

j-rivero, 2021-12-08:
> This is not a big problem, as we only use the Debian Testing CI apt to catch early problems that could eventually appear on Ubuntu, and we could simply disable Gazebo testing there. By looking in [...]

I am yes, thanks for the notice anyway. A good number of issues on the dependency chain of Gazebo were affecting it by some time and I tried to [help](https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=984276) a [little](https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=984063) but was pretty late for the dimension of the issues. I'll try to get it back as soon as possible but problems are not strictly related to Gazebo maintenance.

traversaro, 2021-12-08:
Thanks for the explanation!

traversaro, 2023-04-26:
Fixed by https://github.com/robotology/robotology-superbuild/pull/1096 .