robotology/robotology-superbuild #953: MATLAB  Low-Level Graphics Issues

traversaro, 2021-12-14:
I recently discovered that several users of the robotology-superbuild, in particular of the MATLAB components, experienced errors such as:
~~~
com.jogamp.opengl.GLException: X11GLXDrawableFactory - Could not initialize shared resources for X11GraphicsDevice[type .x11, connection :0, unitID 0, handle 0x0, owner false, ResourceToolkitLock[obj 0xc2145e1, isOwner false, <513205f6, 4fc7f75b>[count 0, qsz 0, owner ]]]
at jogamp.opengl.x11.glx.X11GLXDrawableFactory$SharedResourceImplementation.createSharedResource(X11GLXDrawableFactory.java:326)
at jogamp.opengl.SharedResourceRunner.run(SharedResourceRunner.java:297)
at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.NullPointerException
at jogamp.opengl.GLContextImpl.makeCurrent(GLContextImpl.java:688)
at jogamp.opengl.GLContextImpl.makeCurrent(GLContextImpl.java:580)
at jogamp.opengl.x11.glx.X11GLXDrawableFactory$SharedResourceImplementation.createSharedResource(X11GLXDrawableFactory.java:297)
... 2 more
MATLAB has experienced a low-level graphics error, and may not have drawn correctly.
Read about what you can do to prevent this issue at [Resolving Low-Level Graphics Issues then restart MATLAB](https://www.mathworks.com/help/matlab/creating_plots/resolving-low-level-graphics-issues.html).
To share details of this issue with MathWorks technical support,
please include this file with your service request.
~~~

As mentioned in the linked MATLAB documntation, a workaround for this error is to launch matlab with the `-softwareopengl` flag, that permits to avoid this problem. However, this means that the rendering of plots (especially when using heavy 3D plots such as the one created by iDynTree's MATLAB 3D simulator) will be **extremly slow**, and if the rendering is happening synchronously with your simulator, they could be **slowing down** your simulations. A much better solution is tipically to avoid using the `-softwareopengl` or `opengl('save','software')`, and instead **upgrade or install the best supported driver for your graphics card**, that is indeed the first option in MATLAB documentation (see https://www.mathworks.com/help/matlab/creating_plots/resolving-low-level-graphics-issues.html).

traversaro, 2021-12-14:
@robotology/iit-artificial-mechanical-intelligence

gabrielenava, 2022-01-25:
I tried to upgrade my graphic card drivers, but unfortunately they were already up-to-date. So the issue is still not solved for me and I still need to use `softwareopengl` option. 

I have a DELL XPS with NVIDIA graphic drivers on Ubuntu 20.04. I am trying to run Matlab 2020b.

S-Dafarra, 2022-01-27:
I had the same issue. I tried to update the drivers with 
```
sudo ubuntu-drivers autoinstall
```
but nothing changed. Also by forcing the use of ``nvidia`` with
```
sudo prime-select nvidia
```
but nothing changed. Then I found this online: https://it.mathworks.com/matlabcentral/answers/342906-could-not-initialize-shared-resources-for-x11graphicsdevice#answer_425485

By launching it with
```
MESA_LOADER_DRIVER_OVERRIDE=i965 matlab
```
the error disappeared.

traversaro, 2022-01-27:
Note that if you run `MESA_LOADER_DRIVER_OVERRIDE=i965`, you are basically forcing it to run with the old intel driver (see https://www.phoronix.com/scan.php?page=news_item&px=i965-Mesa-Classic-To-2022).

S-Dafarra, 2022-01-27:
> Note that if you run `MESA_LOADER_DRIVER_OVERRIDE=i965`, you are basically forcing it to run with the old intel driver (see https://www.phoronix.com/scan.php?page=news_item&px=i965-Mesa-Classic-To-2022).

Thanks! I tried with 
```
MESA_LOADER_DRIVER_OVERRIDE=crocus matlab
```
and works well too!

S-Dafarra, 2022-02-07:
Today I realized that on my XPS machine running Ubuntu 20.04, MATLAB R2021b was selecting "Software OpenGL" even without me specifying it.

I then run the "Additional Driver" Ubuntu utility and I noticed that I was "Using NVIDIA Server Driver metapackage from nvidia-driver-470-server". I then switched to "Using NVIDIA driver metapackage from nvidia-driver-510" and MATLAB did not use "Software OpenGL" anymore, and I did not have that error.

Note, I didn't even need to specify ``MESA_LOADER_DRIVER_OVERRIDE=crocus``. I then checked the PC where I had the issue I mentioned in https://github.com/robotology/robotology-superbuild/issues/953#issuecomment-1023489695, and even in that case the ``470`` driver was selected, although the ``510`` was not present. This is because this driver has been released 4 days ago https://launchpad.net/~graphics-drivers/+archive/ubuntu/ppa/+sourcepub/13238016/+listing-archive-extra

gabrielenava, 2022-02-22:
another solution that worked for me and the new iCub PC (both on Ubuntu 20.04)

https://it.mathworks.com/matlabcentral/answers/1468426-cannot-enable-hardware-opengl-r2021b-ubuntu-20-04

isorrentino, 2022-02-22:
> I then run the "Additional Driver" Ubuntu utility and I noticed that I was Using NVIDIA Server Driver metapackage from nvidia-driver-470-server. I then switched to Using NVIDIA driver metapackage from nvidia-driver-510 and MATLAB did not use "Software OpenGL" anymore, and I did not have that error.

In order to work, does this solution need to use NVIDIA performance mode as default GPU? I've already switched to nvidia-driver-510 some time ago, but if the selected GPU is the intel one the error is still there. If instead, I select NVIDIA performance mode the error disappears. 

@S-Dafarra

DanielePucci, 2022-03-02:
@robotology/iit-artificial-mechanical-intelligence do not hesitate to provide feedback in the case of issues

S-Dafarra, 2022-03-16:
Do we have a place where we store tips and tricks related to MATLAB installations?

cc @SanderRuijters

traversaro, 2022-11-16:
I am running Ubuntu 22.04 and the error "MATLAB has experienced a low-level graphics error, and may not have drawn correctly." is still there if you use iris (that unfortunately it is still the default for integrated intel cards on dual intel/nvidia gpu systems). 

So in a nutshell the solution is:
* If you want to use the Intel card (less power used, less computational power), just set `MESA_LOADER_DRIVER_OVERRIDE=crocus`
* If you want to use the Nvidia card (more power used, more computational power), either use the nvidia-settings tool for using the Nvidia card for all process, or either configure it to be on demand and launch matlab with the env variables `__NV_PRIME_RENDER_OFFLOAD=1 __GLX_VENDOR_LIBRARY_NAME=nvidia` set (see https://download.nvidia.com/XFree86/Linux-x86_64/510.85.02/README/primerenderoffload.html).

traversaro, 2022-11-16:
To check: https://it.mathworks.com/matlabcentral/answers/1631110-when-launching-matlab-i-get-the-following-error-mesa-loader-failed-to-open-iris .

LoreMoretti, 2022-11-17:
On Ubuntu 22.04, using i965 can lead to error:

gfx12 and newer are not supported on i965

so it is better to start using crocus (if it works)

LoreMoretti, 2022-11-17:
It is interesting to note that MATLAB seems to be using for sw rendering an internal MESA 17 while the system has MESA 22

>> opengl INFO
                          Version: '2.1 Mesa 17.1.3'
                           Vendor: 'Brian Paul'
                         Renderer: 'Mesa X11'
                   MaxTextureSize: 16384
                           Visual: 'Visual 0x72, (RGBA 32 bits (8 8 8 8), Z depth 16 bits, Hardware acceleration, Double buffer, Antialias 0 samples)'
                         Software: 'true'
             HardwareSupportLevel: 'none'
        SupportsGraphicsSmoothing: 0
    SupportsDepthPeelTransparency: 1
       SupportsAlignVertexCenters: 0
                       Extensions: {152×1 cell}
               MaxFrameBufferSize: 16384

lomoretti@IITICUBLAP226:~/isaacws/isaac-wp1-5 (devel)$ apt show mesa-vulkan-drivers 
Package: mesa-vulkan-drivers
Version: 22.0.5-0ubuntu0.1
Priority: optional
Section: libs
Source: mesa
Origin: Ubuntu
Maintainer: Ubuntu Developers <ubuntu-devel-discuss@lists.ubuntu.com>
Original-Maintainer: Debian X Strike Force <debian-x@lists.debian.org>
Bugs: https://bugs.launchpad.net/ubuntu/+filebug
Installed-Size: 23,1 MB
Provides: vulkan-icd
Depends: libvulkan1, libc6 (>= 2.34), libdrm-amdgpu1 (>= 2.4.109), libdrm2 (>= 2.4.99), libelf1 (>= 0.142), libexpat1 (>= 2.0.1), libgcc-s1 (>= 3.4), libllvm13, libstdc++6 (>= 11), libwayland-client0 (>= 1.20.0), libx11-xcb1 (>= 2:1.7.5), libxcb-dri3-0 (>= 1.13), libxcb-present0, libxcb-randr0 (>= 1.13), libxcb-shm0, libxcb-sync1, libxcb1 (>= 1.9.2), libxshmfence1, libzstd1 (>= 1.4.0), zlib1g (>= 1:1.1.4)
Homepage: https://mesa3d.org/
Task: ubuntu-desktop-minimal, ubuntu-desktop, ubuntu-desktop-raspi, kubuntu-desktop, xubuntu-core, xubuntu-desktop, lubuntu-desktop, ubuntustudio-desktop-core, ubuntustudio-desktop, ubuntukylin-desktop, ubuntu-mate-core, ubuntu-mate-desktop, ubuntu-budgie-desktop, ubuntu-budgie-desktop-raspi
Download-Size: 4.973 kB
APT-Manual-Installed: no
APT-Sources: http://it.archive.ubuntu.com/ubuntu jammy-updates/main amd64 Packages
Description: Mesa Vulkan graphics drivers
 Vulkan is a low-overhead 3D graphics and compute API. This package
 includes Vulkan drivers provided by the Mesa project.

N: There is 1 additional record. Please use the '-a' switch to see it

Tests done on IITICUBLAP283

S-Dafarra, 2022-11-17:
> N: There is 1 additional record. Please use the '-a' switch to see it

Worth to try?

traversaro, 2022-11-17:
> > N: There is 1 additional record. Please use the '-a' switch to see it
> 
> Worth to try?

I tried in a different Ubuntu 22.04 machine, but here is the output (still 22):
~~~
traversaro@IITICUBLAP257:~$ apt show -a mesa-vulkan-drivers
Package: mesa-vulkan-drivers
Version: 22.0.5-0ubuntu0.1
Priority: optional
Section: libs
Source: mesa
Origin: Ubuntu
Maintainer: Ubuntu Developers <ubuntu-devel-discuss@lists.ubuntu.com>
Original-Maintainer: Debian X Strike Force <debian-x@lists.debian.org>
Bugs: https://bugs.launchpad.net/ubuntu/+filebug
Installed-Size: 23.1 MB
Provides: vulkan-icd
Depends: libvulkan1, libc6 (>= 2.34), libdrm-amdgpu1 (>= 2.4.109), libdrm2 (>= 2.4.99), libelf1 (>= 0.142), libexpat1 (>= 2.0.1), libgcc-s1 (>= 3.4), libllvm13, libstdc++6 (>= 11), libwayland-client0 (>= 1.20.0), libx11-xcb1 (>= 2:1.7.5), libxcb-dri3-0 (>= 1.13), libxcb-present0, libxcb-randr0 (>= 1.13), libxcb-shm0, libxcb-sync1, libxcb1 (>= 1.9.2), libxshmfence1, libzstd1 (>= 1.4.0), zlib1g (>= 1:1.1.4)
Homepage: https://mesa3d.org/
Task: ubuntu-desktop-minimal, ubuntu-desktop, ubuntu-desktop-raspi, kubuntu-desktop, xubuntu-core, xubuntu-desktop, lubuntu-desktop, ubuntustudio-desktop-core, ubuntustudio-desktop, ubuntukylin-desktop, ubuntu-mate-core, ubuntu-mate-desktop, ubuntu-budgie-desktop, ubuntu-budgie-desktop-raspi
Download-Size: 4,973 kB
APT-Manual-Installed: no
APT-Sources: http://archive.ubuntu.com/ubuntu jammy-updates/main amd64 Packages
Description: Mesa Vulkan graphics drivers
 Vulkan is a low-overhead 3D graphics and compute API. This package
 includes Vulkan drivers provided by the Mesa project.

Package: mesa-vulkan-drivers
Version: 22.0.1-1ubuntu2
Priority: optional
Section: libs
Source: mesa
Origin: Ubuntu
Maintainer: Ubuntu Developers <ubuntu-devel-discuss@lists.ubuntu.com>
Original-Maintainer: Debian X Strike Force <debian-x@lists.debian.org>
Bugs: https://bugs.launchpad.net/ubuntu/+filebug
Installed-Size: 23.0 MB
Provides: vulkan-icd
Depends: libvulkan1, libc6 (>= 2.34), libdrm-amdgpu1 (>= 2.4.109), libdrm2 (>= 2.4.99), libelf1 (>= 0.142), libexpat1 (>= 2.0.1), libgcc-s1 (>= 3.4), libllvm13, libstdc++6 (>= 11), libwayland-client0 (>= 1.20.0), libx11-xcb1 (>= 2:1.7.5), libxcb-dri3-0 (>= 1.13), libxcb-present0, libxcb-randr0 (>= 1.13), libxcb-shm0, libxcb-sync1, libxcb1 (>= 1.9.2), libxshmfence1, libzstd1 (>= 1.4.0), zlib1g (>= 1:1.1.4)
Homepage: https://mesa3d.org/
Task: ubuntu-desktop-minimal, ubuntu-desktop, ubuntu-desktop-raspi, kubuntu-desktop, xubuntu-core, xubuntu-desktop, lubuntu-desktop, ubuntustudio-desktop-core, ubuntustudio-desktop, ubuntukylin-desktop, ubuntu-mate-core, ubuntu-mate-desktop, ubuntu-budgie-desktop, ubuntu-budgie-desktop-raspi
Download-Size: 4,971 kB
APT-Sources: http://archive.ubuntu.com/ubuntu jammy/main amd64 Packages
Description: Mesa Vulkan graphics drivers
 Vulkan is a low-overhead 3D graphics and compute API. This package
 includes Vulkan drivers provided by the Mesa project.
~~~

diegoferigo, 2022-11-28:
Interestingly, running MATLAB inside a docker container does cause problems. My setup is the following:

```
~ 
❯ sudo tlp-stat
[sudo] password for dferigo: 
--- TLP 1.5.0 --------------------------------------------
[...]
+++ System Info
System         = Dell Inc.  XPS 15 9500
BIOS           = 1.19.0
OS Release     = KDE neon 5.26
Kernel         = 6.0.0-1007-oem #7-Ubuntu SMP PREEMPT_DYNAMIC Tue Oct 25 18:05:20 UTC 2022 x86_64
/proc/cmdline  = BOOT_IMAGE=/boot/vmlinuz-6.0.0-1007-oem root=UUID=f2c9a09c-95fd-4bdf-b3f6-f69d5c625392 ro quiet splash vt.handoff=7
Init system    = systemd v249 (249.11-0ubuntu3.6)
Boot mode      = UEFI
```

My system executes X11 on the Intel GPU, and the NVidia GPU is active with the `on-demand` profile (basically, it's unused unless I send specific computations e.g. using PyTorch or JAX).

The docker container has access to the Nvidia GPU but, again, it's not used to render X11 windows.

Matlab returns this data:

```
>> rendererinfo(gca)

ans = 

  struct with fields:

    GraphicsRenderer: 'OpenGL Hardware'
              Vendor: 'Intel Open Source Technology Center'
             Version: '3.0 Mesa 21.3.7 Amber'
      RendererDevice: 'Mesa DRI Intel(R) UHD Graphics (CML GT2)'
             Details: [1×1 struct]
```

diegoferigo, 2022-11-28:
Interestingly, running MATLAB inside a docker container does not cause any problem. My setup is the following:

```
~ 
❯ sudo tlp-stat
[sudo] password for dferigo: 
--- TLP 1.5.0 --------------------------------------------
[...]
+++ System Info
System         = Dell Inc.  XPS 15 9500
BIOS           = 1.19.0
OS Release     = KDE neon 5.26
Kernel         = 6.0.0-1007-oem #7-Ubuntu SMP PREEMPT_DYNAMIC Tue Oct 25 18:05:20 UTC 2022 x86_64
/proc/cmdline  = BOOT_IMAGE=/boot/vmlinuz-6.0.0-1007-oem root=UUID=f2c9a09c-95fd-4bdf-b3f6-f69d5c625392 ro quiet splash vt.handoff=7
Init system    = systemd v249 (249.11-0ubuntu3.6)
Boot mode      = UEFI
```

My system executes X11 on the Intel GPU, and the NVidia GPU is active with the `on-demand` profile (basically, it's unused unless I send specific computations e.g. using PyTorch or JAX).

The docker container has access to the Nvidia GPU but, again, it's not used to render X11 windows.

Matlab returns this data:

```
>> rendererinfo(gca)

ans = 

  struct with fields:

    GraphicsRenderer: 'OpenGL Hardware'
              Vendor: 'Intel Open Source Technology Center'
             Version: '3.0 Mesa 21.3.7 Amber'
      RendererDevice: 'Mesa DRI Intel(R) UHD Graphics (CML GT2)'
             Details: [1×1 struct]
```

traversaro, 2023-02-02:
cc @fabiodinatale

traversaro, 2023-03-14:
@LoreMoretti can you please shared the output of `sudo lshw -C video` in your system?

diegoferigo, 2023-03-14:
> @LoreMoretti can you please shared the output of `sudo lshw -C video` in your system?

I think you can find it in https://github.com/robotology/robotology-superbuild/issues/953#issuecomment-1318300640

traversaro, 2023-03-14:
> > @LoreMoretti can you please shared the output of `sudo lshw -C video` in your system?
> 
> I think you can find it in [#953 (comment)](https://github.com/robotology/robotology-superbuild/issues/953#issuecomment-1318300640)

Thanks, I had missed this.

traversaro, 2023-03-14:
@diegoferigo find a nice workaround in https://github.com/robotology/robotology-superbuild/issues/1362#issuecomment-1468500368 : 

> Here another solution [1](#user-content-fn-1-99f41394be344688dafa99be77106843) that does not use `amber`. It requires launching `matlab` from a folder where the `java.opts` file is stored.
> 
> ```shell
> ❯ unset MESA_LOADER_DRIVER_OVERRIDE
> 
> ❯ cat java.opts 
> -Djogl.disable.openglarbcontext=1
> 
> ❯ matlab -batch 'rendererinfo(gca)'
> 
> ans = 
> 
>   struct with fields:
> 
>     GraphicsRenderer: 'OpenGL Hardware'
>               Vendor: 'Intel'
>              Version: '4.6 (Compatibility Profile) Mesa 22.2.5'
>       RendererDevice: 'Mesa Intel(R) UHD Graphics (CML GT2)'
>              Details: [1x1 struct]
> ```
> 
> As you can see, I can get HW acceleration using the new 4.6 mesa driver instead of the older 3.0 mesa with amber.
> ## Footnotes
> 
>     1. https://www.mathworks.com/matlabcentral/answers/239279-hardware-based-opengl-on-linux#answer_428438 [↩](#user-content-fnref-1-99f41394be344688dafa99be77106843)

(I am reporting this here so that we can keep the new issue just to collect information and populate the table)

Looking online I did not find any document downside to this solution, so unless a downside emerge, we could just document it in the robotology-superbuild documentation, it would be quite a big time saver for everyone that is experiencing this problems. As a way to make this solution easier, I found:
* Reading online, it seems there should be a variable called `JAVA_OPTS`, `JAVA_OPTIONS` or `JAVA_TOOL_OPTIONS` (depending on the specific java implementation) with the same functionality of `java.opts`, if it works with MATLAB it could be a great to set it as we do for other env variables,
* if setting an env variable is not a viable option, following https://it.mathworks.com/help/matlab/matlab_env/java-opts-file.html, if we put the `java.opts` in  `<matlabroot>/bin/<arch>`, users can start matlab in the directory were they prefer.

diegoferigo, 2023-03-14:
On my setup I need to use `JAVA_TOOL_OPTIONS`, can anyone else confirm that this solution works on their system?

```bash
❯ JAVA_OPTS="-Djogl.disable.openglarbcontext=1" matlab -batch 'rendererinfo(gca)'
com.jogamp.opengl.GLException: X11GLXDrawableFactory - Could not initialize shared resources for X11GraphicsDevice[type .x11, connection :0, unitID 0, handle 0x0, owner false, ResourceToolkitLock[obj 0x276439ae, isOwner false, <496200ab, 1a9ce6d3>[count 0, qsz 0, owner <NULL>]]]
        at jogamp.opengl.x11.glx.X11GLXDrawableFactory$SharedResourceImplementation.createSharedResource(X11GLXDrawableFactory.java:326)
        at jogamp.opengl.SharedResourceRunner.run(SharedResourceRunner.java:297)
        at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.NullPointerException
        at jogamp.opengl.GLContextImpl.makeCurrent(GLContextImpl.java:688)
        at jogamp.opengl.GLContextImpl.makeCurrent(GLContextImpl.java:580)
        at jogamp.opengl.x11.glx.X11GLXDrawableFactory$SharedResourceImplementation.createSharedResource(X11GLXDrawableFactory.java:297)
        ... 2 more
MATLAB has experienced a low-level graphics error, and may not have drawn correctly.
Read about what you can do to prevent this issue by running this command: opengl problems,
then restart MATLAB.
To share details of this issue with MathWorks technical support, please

❯ JAVA_TOOL_OPTIONS="-Djogl.disable.openglarbcontext=1" matlab -batch 'rendererinfo(gca)'
Picked up JAVA_TOOL_OPTIONS: -Djogl.disable.openglarbcontext=1

ans = 

  struct with fields:

    GraphicsRenderer: 'OpenGL Hardware'
              Vendor: 'Intel'
             Version: '4.6 (Compatibility Profile) Mesa 22.2.5'
      RendererDevice: 'Mesa Intel(R) UHD Graphics (CML GT2)'
             Details: [1x1 struct]
```

traversaro, 2023-03-14:
Indeed, this is what documented in the official Java documentation: https://docs.oracle.com/en/java/javase/18/troubleshoot/environment-variables-and-system-properties.html#GUID-A91E7E21-2E91-48C4-89A4-836A7C0EE93B .

traversaro, 2023-03-23:
~~~
export JAVA_TOOL_OPTIONS="-Djogl.disable.openglarbcontext=1"
~~~

Seems a nice solution with no apparent drawbacks, we can probably just put it in the robotology-superbuild's setup.sh.

traversaro, 2023-03-23:
Given that (see https://docs.oracle.com/javase/8/docs/platform/jvmti/jvmti.html#tooloptions):
>  This variable will be broken into options at white-space boundaries.

We can run:
~~~
export JAVA_TOOL_OPTIONS="${JAVA_TOOL_OPTIONS} -Djogl.disable.openglarbcontext=1"
~~~

to permit other to set `JAVA_TOOL_OPTIONS` if necessary.

traversaro, 2023-03-23:
I opened a MATLAB answer to collect downside of the option in https://it.mathworks.com/matlabcentral/answers/1934060-what-are-the-downside-of-setting-djogl-disable-openglarbcontext-1-in-linux?s_tid=srchtitle . Anyhow, I am not sure if anyone will ever reply, so better to start using this options and remove it if we have problems.

traversaro, 2023-03-27:
> I opened a MATLAB answer to collect downside of the option in [...]

We received a reply from a MathWorks employee:

>  If you are experiencing issues with OpenGL acceleration and Intel integrated GPUs on Linux, using this option may provide a reliable workaround for the problem. It is possible that the performance and graphics quality decrease resulting from disabling OpenGL ARB context will not be noticeable in your particular use case.
In general, it is recommended to test the performance and graphics quality of your application with and without this option to determine if it has any significant impact on your specific use case.

So, it seems that the downside is a quality degradation, so nothing not working or similar. So I would go for enabling it by default, if someone in the future notice this degradation we can always go back to not set it.