robotology/robotology-superbuild #959: Unstable branches failure late December 2021

traversaro, 2021-12-18:
Since yesterday the unstable branches failed with error:
~~~
2021-12-18T03:09:57.2068428Z /usr/bin/c++ -DBOOST_ALL_NO_LIB -DBOOST_ATOMIC_DYN_LINK -DBOOST_DATE_TIME_DYN_LINK -DBOOST_FILESYSTEM_DYN_LINK -DBOOST_IOSTREAMS_DYN_LINK -DBOOST_PROGRAM_OPTIONS_DYN_LINK -DBOOST_REGEX_DYN_LINK -DBOOST_SYSTEM_DYN_LINK -DBOOST_TEST_DYN_LINK -DBOOST_THREAD_DYN_LINK -DLIBBULLET_VERSION=2.88 -DLIBBULLET_VERSION_GT_282 -Dgazebo_yarp_lasersensor_EXPORTS -I/home/runner/work/robotology-superbuild/robotology-superbuild/src/GazeboYARPPlugins/plugins/lasersensor/include/gazebo -I/home/runner/work/robotology-superbuild/robotology-superbuild/src/GazeboYARPPlugins/plugins/lasersensor/include/yarp/dev -I/home/runner/work/robotology-superbuild/robotology-superbuild/src/GazeboYARPPlugins/plugins/lasersensor/include -I/home/runner/work/robotology-superbuild/robotology-superbuild/src/GazeboYARPPlugins/libraries/common/include -I/home/runner/work/robotology-superbuild/robotology-superbuild/src/GazeboYARPPlugins/libraries/singleton/include -isystem /usr/include/gazebo-11 -isystem /usr/include/bullet -isystem /usr/include/simbody -isystem /usr/include/sdformat-9.7 -isystem /usr/include/ignition/math6 -isystem /usr/include/OGRE -isystem /usr/include/OGRE/Terrain -isystem /usr/include/OGRE/Paging -isystem /usr/include/ignition/transport8 -isystem /usr/include/ignition/msgs5 -isystem /usr/include/ignition/common3 -isystem /usr/include/ignition/fuel_tools4 -isystem /home/runner/work/robotology-superbuild/robotology-superbuild/build/install/include -isystem /usr/include/eigen3 -isystem /usr/include/sdformat-9.7/sdf/.. -isystem /usr/include/ignition/cmake2 -isystem /usr/include/uuid -isystem /usr/include/opencv4 -O3 -DNDEBUG -fPIC -I/usr/include/uuid -std=gnu++17 -MD -MT plugins/lasersensor/CMakeFiles/gazebo_yarp_lasersensor.dir/src/LaserSensor.cc.o -MF plugins/lasersensor/CMakeFiles/gazebo_yarp_lasersensor.dir/src/LaserSensor.cc.o.d -o plugins/lasersensor/CMakeFiles/gazebo_yarp_lasersensor.dir/src/LaserSensor.cc.o -c /home/runner/work/robotology-superbuild/robotology-superbuild/src/GazeboYARPPlugins/plugins/lasersensor/src/LaserSensor.cc
2021-12-18T03:09:57.2079962Z In file included from /home/runner/work/robotology-superbuild/robotology-superbuild/src/GazeboYARPPlugins/plugins/lasersensor/src/LaserSensor.cc:9:
2021-12-18T03:09:57.2082863Z /home/runner/work/robotology-superbuild/robotology-superbuild/src/GazeboYARPPlugins/plugins/lasersensor/include/yarp/dev/LaserSensorDriver.h:68:29: error: ‘virtual yarp::os::Stamp yarp::dev::GazeboYarpLaserSensorDriver::getLastInputStamp()’ marked ‘override’, but does not override
2021-12-18T03:09:57.2084624Z    68 |     virtual yarp::os::Stamp getLastInputStamp() override;
2021-12-18T03:09:57.2085091Z       |                             ^~~~~~~~~~~~~~~~~
~~~

traversaro, 2021-12-18:
Probably related to https://github.com/robotology/yarp/commit/b34040885cba9ad3d154c6cc37ec70996310dc82 .

traversaro, 2021-12-18:
A possible fix is to remove the `override` from `virtual yarp::os::Stamp getLastInputStamp()`.

traversaro, 2021-12-18:
The problem is that there is also a change in the laser interface in https://github.com/robotology/yarp/pull/2778 , so probably also that YARP version will not be compatible with the latest gazebo-yarp-plugins.

pattacini, 2021-12-21:
cc @randaz81

traversaro, 2022-01-21:
Fixed by https://github.com/robotology/gazebo-yarp-plugins/pull/599 .

closed 2022-01-21 by traversaro