robotology/robotology-superbuild #963: provide python code for skin sensor from icub simulator

ArfnZl, 2021-12-19:
hello sir
i tried many times to write code,but my code is not working please provide python code for tectile skin sensor from icub simulator thanks
import yarp
import numpy as np
import matplotlib.pyplot as plt
import sys
from mpl_toolkits.mplot3d import Axes3D
import scipy.interpolate
yarp.Network.init()
 

right_hand = yarp.Port()
prop = yarp.Property();
right_hand.open("/python/right_hand")
yarp.Network.connect('/icubSim/skin/right_hand_comp','/python/right_hand')
read.(/reader /icubSim/skin/right_hand_comp
)

ArfnZl, 2021-12-19:
please sir correct my code ..the last line my code is giving errors..ports is connected but reading part is not working

traversaro, 2021-12-19:
Hi @ArfnZl, I guess you are refererring to the old iCub_SIM simulator, right? Could you report the system you are on, the version of robotology-superbuild you are using and the options with which you compiled it?

> the last line my code is giving errors..

Can you report the error that your code is returning?

traversaro, 2021-12-27:
Hi @ArfnZl, did you solved your problem?

ArfnZl, 2021-12-27:
No my problem is not solving  please  help to solve this problem i am
waiting for your reply thank you

traversaro, 2021-12-27:
Hi @ArfnZl, it would be great if you could answer the question asked in https://github.com/robotology/robotology-superbuild/issues/963#issuecomment-997440101, otherwise it would be difficult to help you.

ArfnZl, 2021-12-27:
Dear i am using the latest

traversaro, 2021-12-27:
@ArfnZl, please answer all questions, thanks!

ArfnZl, 2022-01-09:
[image]

ArfnZl, 2022-01-09:
[image]

ArfnZl, 2022-01-09:
@traversaro 
Please  help

ArfnZl, 2022-01-09:
@traversaro can you provide any example of reading data  from tactile  sensors in icub simulator  through python code

ArfnZl, 2022-01-09:
@traversaro 
The problem is not with port just give you an example of Python code skin sensor reading

traversaro, 2022-01-10:
@ArfnZl you can find some examples for the Python bindings of YARP in https://github.com/robotology/yarp/tree/master/bindings/python/examples and https://github.com/s4hri/yarp-python-tutorials . If you follow them, you should be able to understand how to read data from any YARP port, including the ports that publish skin sensor data on the iCub_SIM simulator.

traversaro, 2022-02-21:
Hi @ArfnZl, feel free to open new issues if you have more doubts.

closed 2022-02-21 by traversaro